espressif/arduino-esp32 #2224: assertion "0 && "fatfs internal error""

shivanshubohara, 2018-12-20:
hadware : node mcu -esp32
software: arduino
used esp32 espressif library
Trying to store data in SD card using SD card module.
Code: SD Test code
No compiling error
Card initialised

But then shows error  
assertion "0 && "fatfs internal error"" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/fatfs/src/vfs_fat.c", line 253, function: fresult_to_errno

Serial Monitor output : 

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9220
load:0x40080400,len:6300
entry 0x400806a4
SD Card Type: SDHC
SD Card Size: 14804MB
Listing directory: /
assertion "0 && "fatfs internal error"" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/fatfs/src/vfs_fat.c", line 253, function: fresult_to_errno
abort() was called at PC 0x400db19b on core 1

Backtrace: 0x40085e88:0x3ffb1ba0 0x400860b5:0x3ffb1bc0 0x400db19b:0x3ffb1be0 0x400d995f:0x3ffb1c10 0x400da393:0x3ffb1c30 0x400dae19:0x3ffb1da0 0x4000bcc5:0x3ffb1dc0 0x400dba11:0x3ffb1de0 0x400d17af:0x3ffb1e00 0x400d1b47:0x3ffb1e30 0x400d1cc2:0x3ffb1e70 0x400d1491:0x3ffb1ee0 0x400d0fe3:0x3ffb1f10 0x400d1375:0x3ffb1f80 0x400edce7:0x3ffb1fb0 0x4008767d:0x3ffb1fd0

Rebooting...

This card and module perfectly works with Uno but error shown with esp32.
Can anyone help?

atanisoft, 2018-12-20:
you can start by decoding the backtrace and providing some code to reproduce.

z-trade, 2019-01-30:
I got this error also. My case is:
esp32 custom board
eMMC BGA153 8Gbyte chip
---
The board  constantly writes and reads data. Also the board gets power off very often. I noticed that at some point I start to get such error when I try to read some created files. and it looks to me as some FAT data corrupted (not eMMC). I can reformat eMMC and the problem goes away as well as I can read and write other files with no a problem.
---
What is the problem to me is that with such error whole board is reseted. So I cannot detect and workaround this issue automatically since I never know what file is corrupted in such way.
---
Is it possible to not reset whole board and just return an error what it can be managed?
---
My error 
02:37:47.483 -> assertion "0 && "fatfs internal error"" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/fatfs/src/vfs_fat.c", line 253, function: fresult_to_errno
02:37:47.517 -> abort() was called at PC 0x400ed087 on core 0
02:37:47.517 -> 
02:37:47.517 -> Backtrace: 0x40092294:0x3ffe8d20 0x400924c5:0x3ffe8d40 0x400ed087:0x3ffe8d60 0x400e0cd3:0x3ffe8d90 0x400e0ec7:0x3ffe8db0 0x400e9ca6:0x3ffe8dd0 0x4000bd9f:0x3ffe8df0 0x40001191:0x3ffe8e10 0x400ed569:0x3ffe8e30 0x400ed316:0x3ffe8ea0 0x400d2dd9:0x3ffe8ec0 0x400d35f1:0x3ffe8ee0 0x400d38b7:0x3ffe8f90 0x400d85ab:0x3ffe8fc0 0x400d8621:0x3ffe8fe0 0x400d866e:0x3ffe9010 0x400d87c6:0x3ffe9060 0x400d3a52:0x3ffe90b0 0x400d27d5:0x3ffe90d0 0x4009498d:0x3ffe9110
02:37:47.554 -> 
02:37:47.554 -> Rebooting...
02:37:47.554 -> ets Jun  8 2016 00:22:57
02:37:47.554 -> 
02:37:47.554 -> rst:0xc (SW_CPU_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)

closed 2019-08-15 by stale[bot]