espressif/arduino-esp32 #2226: PSRAM enabled, but not usable as internal heap memory

cyberman54, 2018-12-21:
How can use PSRAM as internal heap memory?

I saw this already working in previous releases, PSRAM was automatically added to internal heap memory when device started. But since latest updates this is broken or changed in some way.

```
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
[I][main.cpp:109] setup(): Internal Total heap 239056, internal Free Heap 213212
[I][main.cpp:111] setup(): SPIRam Total heap 4194252, SPIRam Free Heap 4194252
```

lbernstone, 2018-12-21:
The behavior of ESP.getHeapSize was incorrect when PSRAM support was first added (~Feb 2018).  PSRAM is not directly part of your heap.  So, the functions were reverted in August to just show the free/total SRAM, and PSRAM got its own functions.  If you want to allocate memory from PSRAM, you will need to use ps_malloc manually (as was always the case).

cyberman54, 2018-12-21:
@ibernstone I need to store a std::set container in PSRAM. Not sure how to do this. I thought the PSRAM would increase the heap where std::set normally stores it's contents, and i was convinced that this worked automatically before on boards which has PSRAM? I could see ESP.getHeapSize counting down while storing elements in a std::set container.

lbernstone, 2018-12-21:
Dynamic objects like a set will need a custom allocator.  I don't have an example for you, though I do think it would be quite a useful thing.  Maybe one of the big C brains out there can make an example.

cyberman54, 2018-12-21:
Allright, thank you for your help. Since it turned out that this is not an issue of this library, i close this issue here.

closed 2018-12-21 by cyberman54

me-no-dev, 2018-12-24:
Arduino is compiled with the option to not add PSRAM to the general heap, because most Arduino users have no idea what is the difference between internal RAM and PSRAM and will allocate everything there, then use it in interrupts and have all kinds of issues and errors. Therefore we have provided the `ps_malloc` and friends to allocate memory there when needed.

sonolive, 2022-02-21:
>>Arduino is compiled with the option to not add PSRAM to the general heap,

hello,
ok, but how to allow this option for a new build of arduino ???
thx in advance
olive

lbernstone, 2022-02-21:
Since v2.0.0, psram has been directly mapped, so allocations > 16k (and 4k in the newer chips) are created in psram.
While I am here, I might as well post code for a custom allocator, which I don't think I have posted before.
```
template <class T>
struct PSallocator {
  typedef T value_type;
  PSallocator() = default;
  template <class U> constexpr PSallocator(const PSallocator<U>&) noexcept {}
  [[nodiscard]] T* allocate(std::size_t n) {
    if(n > std::size_t(-1) / sizeof(T)) throw std::bad_alloc();
    if(auto p = static_cast<T*>(ps_malloc(n*sizeof(T)))) return p;
    throw std::bad_alloc();
  }
  void deallocate(T* p, std::size_t) noexcept { std::free(p); }
};
template <class T, class U>
bool operator==(const PSallocator<T>&, const PSallocator<U>&) { return true; }
template <class T, class U>
bool operator!=(const PSallocator<T>&, const PSallocator<U>&) { return false; }

std::vector<int, PSallocator<int> > v;

void setup() {
  Serial.begin(115200);
  delay(1000);
  Serial.println(ESP.getFreePsram());
  v.reserve(900000);
  for (uint32_t x=0; x<900000; x++) {
    v.push_back(x);
  }
  Serial.println(ESP.getFreePsram());
  v.clear();
  v.shrink_to_fit();
  Serial.println(ESP.getFreePsram());
}
```

sonolive, 2022-02-21:
hello [lbernstone](https://github.com/lbernstone)

many thanks for the answer,
but what i would like to achieve is something like this : 
```

const int size_tablo = 150000;
EXT_RAM_ATTR int test_tablo_int[size_tablo];

void setup() {
Serial.begin(115200);

  Serial.print("Build test_tablo_int - size = ");Serial.println(size_tablo);

  for(int i = 0 ; i < size_tablo ; i++)
  {
  test_tablo_int[i] = i;
  }
  Serial.print("size of test_tablo_int : ");Serial.print(float(sizeof(test_tablo_int))/1000,3);Serial.println(" KB");

 Serial.println(ESP.getFreePsram());
}
```
i tried to compile with 

```
#
# SPI RAM config
#
CONFIG_SPIRAM=y
CONFIG_SPIRAM_BOOT_INIT=y
CONFIG_SPIRAM_USE_MALLOC=y
CONFIG_SPIRAM_MEMTEST=y
CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY=y
CONFIG_SPIRAM_CACHE_WORKAROUND=y
```
in eclipse espressif IDE it works well

then copied/pasted the
 libesp_system.a from eclipse espressif IDE
C:\Espressif\frameworks\esp-idf-v4.4\workspace\blink_2\build\esp-idf\esp_system
to :
...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\tools\sdk\esp32\lib

then compiled with Arduino IDE 

it reboots and the error is :
`[E][esp32-hal-psram.c:76] psramInit(): PSRAM could not be added to the heap!`

in many cases i have tested for few days, yhe line
CONFIG_SPIRAM_BOOT_INIT=y
causes the issues in arduino IDE

lbernstone, 2022-02-21:
psram is not ram- it is not available until the system is initialized.  You cannot create global variables there.  Create a static object when you first need to use the psram variable.

sonolive, 2022-02-22:
hello lbernstone
thank you for the answer,
Maybe my explainations were not clear enough ...

i will try to do better ;-)

> You cannot create global variables there

I CAN : this works very well with eclipse ESP-IDE

with this SPI RAM CONFIG in sdkconfig ->
```
#
# SPI RAM config
#
CONFIG_SPIRAM_TYPE_AUTO=y
# CONFIG_SPIRAM_TYPE_ESPPSRAM16 is not set
# CONFIG_SPIRAM_TYPE_ESPPSRAM32 is not set
# CONFIG_SPIRAM_TYPE_ESPPSRAM64 is not set
CONFIG_SPIRAM_SIZE=-1
CONFIG_SPIRAM_SPEED_40M=y
CONFIG_SPIRAM=y
CONFIG_SPIRAM_BOOT_INIT=y
# CONFIG_SPIRAM_USE_MEMMAP is not set
# CONFIG_SPIRAM_USE_CAPS_ALLOC is not set
CONFIG_SPIRAM_USE_MALLOC=y
CONFIG_SPIRAM_MEMTEST=y
CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=4096
CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP=y
CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL=0
CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY=y
# CONFIG_SPIRAM_ALLOW_NOINIT_SEG_EXTERNAL_MEMORY is not set
CONFIG_SPIRAM_CACHE_WORKAROUND=y
```

this code  ->

```
#include "Arduino.h"

void memories (String str);

int free_CAP_8BIT;
int free_CAP_SPIRAM;
int free_CAP_INTERNAL;

const int size_tablo = 150000;
EXT_RAM_ATTR int test_tablo_int[size_tablo];

void setup(){
    Serial.begin(115200);
		for(int i = 0 ; i < size_tablo ; i++){
			test_tablo_int[i] = i;
			}
    Serial.print("size of test_tablo_int : ");Serial.print(float(sizeof(test_tablo_int))/1000,3);Serial.println(" KB");
    test_tablo_int[size_tablo / 10] = 1;
    test_tablo_int[size_tablo / 5] = 2;
    test_tablo_int[size_tablo / 2] = 3;

		for(int i = 0 ; i < size_tablo ; i+=(size_tablo/10)){
			 Serial.print("tablo[");Serial.print(i);Serial.print("] = ");Serial.println(test_tablo_int[i]);
			}
    memories ("memories @ end setup");
    Serial.println("end setup");
}

void loop(){
//	Serial.println("loop");
/*
	for(int i = 0 ; i < size_tablo ; i+=(size_tablo/10))
[... 207 lines cut ...]
```
it does not compile ... 150000 is too large for the DRAM
the output is ->

```
.../appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\OLIVIE~1\AppData\Local\Temp\arduino_build_803706/sketch_feb15a.ino.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
c:/users/olivier_pailler/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: DRAM segment data does not fit.
c:/users/olivier_pailler/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: DRAM segment data does not fit.
c:/users/olivier_pailler/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: region `dram0_0_seg' overflowed by 491792 bytes
collect2.exe: error: ld returned 1 exit status
exit status 1
Erreur de compilation pour la carte ESP32 Dev Module

```
i hope it's clearer ... mmm not so sure !

maybe it is caused by what said me-no-dev :
"Arduino is compiled with the option to not add PSRAM to the general heap,"

i noticed comparing builds with with WinMerge
that only libesp_system.a and libapp_update.a were different ...
and also libarduino.a ...

ouf ...
olive

me-no-dev, 2022-02-22:
maybe the reason is the fact that Arduino initializes PSRAM a bit later than ESP-IDF. This is because we use prebuilt IDF libs with PSRAM support, but in most cases PSRAM is not available and needs to be enabled by the user in the board menu of Arduino IDE. Since v2.0.0, we also add PSRAM to malloc, but as I said... happens slightly later. You can define the array as pointer and allocate it in `setup()`. This will fix your issue.

sonolive, 2022-02-22:
hello me-no-dev
thanx alot for this answer,

> maybe the reason is the fact that Arduino initializes PSRAM a bit later than ESP-IDF

oh yes i didnot think of that bur for sure it could be the problem ...

> Since v2.0.0, we also add PSRAM to malloc, but as I said... happens slightly later. You can define the array as pointer and allocate it in setup()

Yes , ok i know that, and it could fix this little sketch !
the fact is that i am looking for a solution for a MUCH LARGER app, based on LVGL gfx library and after few seeks of dev i encounter this issue ...
320.000 KB is not enough, i need much more, and they are here ... 4MB but i cannot use them

realloc dynamically everything should be a real pain ...
but maybe i will have to do that !
booooooh

anyway thx you 2 for the job you do for arduino-esp32
regards,
olive

star297, 2022-11-27:
If anyone gets here, I had a similar issue, here's a solution.
Works on PlatformIO, haven't tried on regular Arduino IDE.  

https://community.platformio.org/t/how-to-use-a-global-struct-array-in-psram-esp32-wrover/30696

rtprof, 2022-11-27:
> If anyone gets here, I had a similar issue, here's a solution. Works on PlatformIO, haven't tried on regular Arduino IDE.
> 
> https://community.platformio.org/t/how-to-use-a-global-struct-array-in-psram-esp32-wrover/30696

Good job. Just now I'm searching the solution and here it is!!!
Thank you. I tried on Arduino IDE and it works. 
Now I think how to send all my String variables to PSRAM

davepl, 2024-01-08:
Am I correct, then, that it's not possible to allocate BSS and DATA on PSRAM because CONFIG_SPIRAM is hard-coded in the Arduino build?

Or is there a way to get static allocations into PSRAM on an Arduino project?