espressif/arduino-esp32 #2231: ADC outputs 0

albertoZurini, 2018-12-22:
I've tried building an EMF detector with an ESP32 and a Web GUI. The project is available here: https://github.com/albertoZurini/ESP32-EMF-DETECTOR

Everything works fine except the ADC which is outputting 0-readings almost half the time.
What am I doing wrong?

fah, 2018-12-22:
Hallo Alberto, if I get your project right it is a receiver for low frequency emf *pulses* (I don't see the frequency detected/selected in your code or hardware)? A0 is connected to GND via a greater then 1M resistor? 

So what do you expect without the right antenna (what gain?) connected on A0 *and* GND? Of course your experiment is just powered by battery? Is it? If it *is* free floating and the antenna is correctly build with a shielded cable it should work better. 

If your read just min and max values you have a good antenna. :-) Lower the attenuation then.

What I want to say is: Maybe trust the readings. There is maybe a real reason for a zero reading. -> nothing there *this* antenna can receive.

albertoZurini, 2018-12-22:
Yes
1M pulldown or greater is preferred based on how much current we want to capture (higher resistance lower current), so kind of precision
The antenna shouldn't be much of a problem, if I have a wire 1m long I can calculate the Electric field having the voltage (I will implement this in the future)
Powered by battery, yes

The actual problem is that if I put the antenna close to a light bulb it will output 4095, 4095, 0, 0, 0, 4095... So something is wrong.

fah, 2018-12-22:
If you receive an over amplified sinus and do not get all values in between it looks like what you see. 

So a sinus gets a ... rectangle oscillation.  0 4096 0 4096 

BTW: I miss some electronic components in your project. Diodes, coils, capacitors, z-diodes to protect the poor esp32 ... :-)

closed 2019-08-15 by stale[bot]