espressif/arduino-esp32 #2235: ESP32-Arduino w/WolfSSL

AIWIndustries, 2018-12-23:
Do we know when WolfSSL will be implemented into the ESP32 and also the ESP32-Arduino Core. This platform will open new possibilities especially when it comes to government products that are increasingly looking towards FIPS-140-2 certification. The base WolfSSL library will not have the features for FIPS-140-2 enabled by default, but for a fee WolfSSL will provide the instructions on how to implement those and provide a certificate.

lbernstone, 2018-12-23:
If such a thing were to be considered, it would be part of ESP-IDF, not Arduino.  Since there is already a SSL/TLS library included in the IDF, this would be an uphill battle.  The licensing for WolfSSL does not appear to be unencumbered for that certification, and this is an open source project.

AIWIndustries, 2018-12-23:
I notice WolfSSL has a new method of implementing into the ESP-IDF. It’s in the Espressif IDE Folder

https://github.com/wolfSSL/wolfssl/tree/master/IDE

I know the FIPS cert would never be implemented but the utilization of WolfSSL would allow for it if someone wanted to work with them on an individual basis to gain access to those mechanisms.

I also noticed that espressif looks to have plans to implement WolfSSL by reading through the ESP8266 SDK Readme.

everslick, 2018-12-24:
If the goal is to have a smaller footprint TLS implementation, maybe it would be more practical to port the ESP8266 BearSSL library to ESP32.

copercini, 2018-12-25:
` Commercial licenses for wolfSSL, yaSSL, and wolfCrypt are available for $5,000 USD per end product or SKU` from here: https://www.wolfssl.com/license/

People probably do not even know they will be using wolfSSL and have to pay $5,000 USD if they want to sold something that uses SSL/TLS functions of this repo

AIWIndustries, 2018-12-25:
@copercini — very good point. Would definitely need some type of disclaimer warning if this ever got used or was an option. Since it looks like the ESP-IDF might already be exploring this item I’m going to close this issue. Thanks for the feedback.

copercini, 2018-12-25:
@AIWIndustries take a look if MbedTLS doesn't accomplish your requirements, it's very complete and fully supported

AIWIndustries, 2019-04-11:
mBed works just fine but does not have an option for FIPS certified libraries. I plan to use ESP32 in government project and FIPS certification is a plus. ESP-IDF now has it ......maybe an arduino implementation could be achieved.

AIWIndustries, 2019-04-11:
https://github.com/espressif/esp-wolfssl

closed 2019-08-15 by stale[bot]

SomenKumarDas, 2021-06-12:
Can I use wolf SSL with arduino framework Now?....