espressif/arduino-esp32 #2236: Supports IO Extender PCA9534

DeltaVetal26, 2018-12-23:
Hi!

 I want to connect a keypad that is removed from the ESP32 at a distance of 1 meter. In order not to draw a lot of wires, I plan to connect the buttons to the [PCA9534 IO-extender](http://www.ti.com/lit/ds/symlink/pca9534.pdf) running on the I2C bus. Unfortunately, I did not find libraries for ESP32 in this module.
The question is, can I connect this module to ESP32? As far as I understand, this can be done through the Wire library. Is I2C bus support reliable in ESP32? I heard that many people have problems with I2C on ESP32, is this true?

atanisoft, 2018-12-23:
a quick search on google for "android pca9534" turns up https://github.com/alotaiba/PCA9534 which should work provided you are able to use the default I2C pins, if you need to use a different set of pins you can always modify the library code to initialize the bus yourself.

DeltaVetal26, 2018-12-23:
@atanisoft,

Thank you!
I thought that not all Arduino libraries could work as well on ESP. If not, then great!
What about the reliability of I2C on ESP32? Are there any global problems?

atanisoft, 2018-12-23:
Almost all Arduino libraries will work on the ESP32, some may require modifications though if they depend on the AVR hardware.

I2C is mostly stable on the ESP32, there are a few known issues in the 1.0.0 release of arduino-esp32 but @stickbreaker has fixed many of these for the 1.0.1 release (currently available as an RC).

DeltaVetal26, 2018-12-24:
@atanisoft 

Thank you!!!

closed 2019-08-15 by stale[bot]