espressif/arduino-esp32 #2237: Instead SPIFF Rawbytes

schuppeste, 2018-12-23:
It is possible to read and write Spiff directly as Rawbytes without Overhead?

How can i write and read directly to Flash Partitions, can i use OTA Partition?

lbernstone, 2018-12-23:
The EEPROM library reads and writes raw bytes.  OTA is only available for the app partitions, but it would be possible to write code that streams directly out to "disk".  There is no way to bypass the cpu from lwip AFAIK.

schuppeste, 2018-12-27:
No, i need Space for Graphics..

in my case i use constant Arrays for Pictures, spiff is too slow Performance Spiff has Overhead many milliseconds,
I can write Spiff, but i need siliar Space to write and read easy from Partition, maybe data Partition.

copercini, 2018-12-27:
What about NVS (preferences) blob? https://github.com/espressif/arduino-esp32/blob/master/libraries/Preferences/src/Preferences.cpp#L450

schuppeste, 2018-12-28:
i tried:
` epp = (esp_partition_t *)esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_SPIFFS, NULL);

esp_partition_erase_range(epp,i,4096 );
esp_partition_write(epp, i, pos,4096);

esp_partition_read(epp,0 , buf, sizeof(p0);
`
Fast enough. without spiff overhead.

copercini, 2018-12-30:
@schuppeste nice!
Can the issue be closed?

closed 2018-12-30 by schuppeste