espressif/arduino-esp32 #2239: pm_configure function does not work

cziter15, 2018-12-23:
It looks like pm_configure function does not work. It looks Power Managament was disabled during libraries build pipeline.

repro steps:
simply call pm_configure, it always return error ERR_NOT_SUPPORTED

cziter15, 2018-12-24:
> yes and there is a good reason for that. drivers are not ready to handle
> constantly changing APB frequency. When that integration is finished, pm
> options will come. Arduino is used by many different levels of thinkerers
> :) and some features of ESP32 will cause many issues for everyone that does
> not understand what exactly happens when cpu/apb frequencies are changing.
> As of right now, the ESP32 Dev Board in the IDE menu has the option for you
> to select CPU frequency on boot and drivers will work ok. There is also API
> setCpuFrequency(freq_in_mhz) that you can use to change it on the fly,
> but you will have to reset baudrates/clocks of peripherals that you are
> currently using (like Serial for example). We will also have that "fixed"
> soon, but as of now, this is what it is :)

Okalmy, but there is also light sleep option, which sometimes is required.
I had to switch from Arduino to esp-idf because of this.

me-no-dev, 2018-12-24:
for users like you, we have the "arduino as idf component" level of integration, where you have full control of what you are compiling and having access to. As I said, we are working on bringing that functionality in, just not done yet.

cziter15, 2018-12-24:
Ok, tahnk you ;)
@ little offtopic Is it possible to manually recompile libs to put then in
tools/sdk/lib folder? This will enable use of pm_configure if I enable
power management in sdkconfig. I can't see any option, there are only
precompiled libs out there on github.

me-no-dev, 2018-12-24:
follow the arduino as component instructions and use the sdkconfig inside tools/sdk folder as starting point. Then compile empty sketch or whatever with pm enabled. Once all is compiled, in `[proj_root]/build/[name_of_lib]/` you will find the `.a` file to put in your IDE arduino install sdk folder. You will need to also replace `[esp32-arduino]/tools/sdk/include/config/sdkconfig.h` in your IDE arduino with `[proj_root]/build/include/sdkconfig.h` so the PM options are active in the headers.