espressif/arduino-esp32 #2240: CONFIG_FREERTOS_USE_TICKLESS_IDLE not enabled

cziter15, 2018-12-23:
Flag CONFIG_FREERTOS_USE_TICKLESS_IDLE is not enabled during sdk lib build process. It results in power drain, because every delay() call is busy-wait.

igrr, 2018-12-25:
`delay` maps to vTaskDelay, which blocks the task until given number of ticks have expired. The difference in behavior w.r.t. tickless idle option will be for the case when no other tasks are ready to run. If that option is disabled, then the IDLE task will put the CPU into so-called waiti (wait for interrupt) state. This can happen on both CPUs. Since tick interrupt is still enabled, FreeRTOS tick will happen at the defined period, but it will not do anything until delay has expired (since no tasks are ready to run) and will quickly return back to the IDLE task each time. If tickless idle is enabled, tick interrupt will be suspended for the expected duration that the task is idle. In practice the difference between the two options in terms of power consumption is nearly impossible to observe (on the ESP32). Tickless idle option becomes really useful when it is coupled with the automatic light sleep mode, provided by esp_pm APIs.
The downside of enabling tickless idle option is that tick period becomes less predictable, especially if peripheral interrupts happen during the idle period.

lbernstone, 2018-12-25:
As a non-technical follow up, it should be pointed out that arduino-esp32 is intended to be a compatibility layer for existing arduino code on esp32.  So, the configuration is chosen to provide maximum compatibility and ease of integration for hobbyists who may have very low technical skills. For more powerful users, [arduino as an IDF component](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md) or [Platform.io](https://platformio.org/platformio-ide) offer you control of the sdkconfig parameters.

me-no-dev, 2018-12-26:
I guess since answer is given and options are listed, it is ok to close this issue.

closed 2018-12-26 by me-no-dev