espressif/arduino-esp32 #2242: Client cannot connect ESP32 AP in AP-STA Mode

diazaki, 2018-12-25:
### Hardware:
Board: DOIT ESP32 DevKit v1
Core Installation/update date: -
IDE name: Arduino IDE
Flash Frequency:80MHz
PSRAM enabled:-
Upload Speed:115200
Computer OS:Windows 10

### Description:
Hi, i'm sorry before for my bad english language.
My name is Diaz, i wanna make smart plug with ESP32. I've done with my ESP8266 Before, but i need more GPIO, so i've change to ESP32.

My problem is when i've change my program to from ESP8266 to ESP32, my phone cannot connect to ESP32 Access Point when my ESP32 is connected to other Access Point (In AP-STA Mode). My phone can connect to ESP32 only in AP Mode. From three device below, only my phone who is cannot connect to my ESP32.
1. My Phone (Cannot Connected to my ESP32)
2. My Mother Phone (Can Connected to my ESP32)
3. My Laptop (Can Connected to my ESP32)

Can anybody help me? i've been googling and still stuck in this problem. why just my phone who cannot connect to ESP32 in AP-STA Mode,

### Sketch: I'm cut my sketch because it's too long :)
  
#include <Arduino.h>
#include "WiFi.h"
void setup() 
{
WiFi.mode(WIFI_AP_STA);
WiFi.softAP("TEST-DSP", "12345678");
WiFi.begin("spiderman", "spiderboy");
}

void loop() {
}

### Debug Messages:
When my phone trying to connect my ESP32, i got authentication problem and my phone always ask to input password again.

lbernstone, 2018-12-25:
What type of phone?  Turn on logging in Arduino IDE (board-> esp32 dev module, tools -> Core debug level -> Verbose) and post the log.

diazaki, 2018-12-25:
Hi, thank you for the response, my phone is Xiaomi Redmi 4A. How i can get the log? i've choose the Core debug level Verbose

lbernstone, 2018-12-25:
Start the serial monitor and upload your code.  You will need to add `Serial.begin(115200);` to your sketch.

diazaki, 2018-12-25:
owh, yes, in my code has serial begin :D , but i can't get any log except the serial message in my script.

> SSID_AP Sudah Ada
> PASS_AP Sudah Ada
> SSID_AP : TEST-DSP
> PASS_AP : 12345678
> SSID_WIFI : spiderman
> PASS_WIFI : spiderboy
> IP Address : 0.0.0.0
> Soft AP Ip Address : 192.168.19.10
> MAC Address : 3C:71:BF:4D:67:28
> Wifi Not Connected
> Wifi Not Connected
> Wifi Not Connected
> Wifi Not Connected

when i stop get message "Wifi Not Connected", thats mean ESP32 has connected to some access point.

lbernstone, 2018-12-25:
Upload the exact code above with a Serial.begin.  This is clearly not what is on your device.

diazaki, 2018-12-25:
Here my void setup code :)

>   WiFi.mode(WIFI_AP_STA);
>   
>   pinMode(ch1, OUTPUT);
>   pinMode(ch2, OUTPUT);
>   pinMode(ch3, OUTPUT);
>   pinMode(ch4, OUTPUT);
>   
>   Serial.begin(115200);
>   EEPROM.begin(512);
>   
>   esp_read_mac(mac_addr, ESP_MAC_WIFI_STA); str_mac_addr5 = String(mac_addr[4], HEX); str_mac_addr6 = String(mac_addr[5], HEX);
>     
>   if(EEPROM.read(1) != 0){Serial.println("SSID_AP Sudah Ada");}
>   else
>   {
>   str_mac_addr = "DSP-" + str_mac_addr5 + str_mac_addr6;
>   str_mac_addr.toCharArray(ssid_ap,15);
>   Serial.println("Menulis SSID_AP Baru");
>   for(int x=1; x<16; x++){int ssid_ap_addr = x - 1; EEPROM.write(x, ssid_ap[ssid_ap_addr]);} EEPROM.commit();
>   }
>   
>   if(EEPROM.read(21) != 0){Serial.println("PASS_AP Sudah Ada");}
>   else{Serial.println("Menulis PASS_AP Baru"); for(int x=21; x<36; x++){int pass_ap_val = x + 28; if(x < 29) {EEPROM.write(x, pass_ap_val);} else{EEPROM.write(x, 0);} EEPROM.commit();}}
> 
>   read_eeprom();
> 
>   WiFi.softAPConfig(AP_local_IP, AP_gateway, AP_subnet);
>   WiFi.softAP(ssid_ap);//, pass_ap);
>   WiFi.begin(ssid_wifi, pass_wifi);
>   server.begin();
> 
>   get_ip();
>   print_info();

Maybe you need to know function get_ip and print_info, i upload too :)
> void get_ip()
> {
>   str_ip_addr = WiFi.localIP().toString();
>   str_ip_addr.toCharArray(ip_addr,15);
>   for(int x=81; x<96; x++){int ip_addr_addr = x - 81; EEPROM.write(x, ip_addr[ip_addr_addr]); EEPROM.commit();}
>   for(int x=81; x<96; x++){int ip_addr_addr = x - 81; int val_eeprom = EEPROM.read(x); /*if(val_eeprom == 0){val_eeprom = 32;}*/ip_addr[ip_addr_addr] = (char(val_eeprom));}
> }

> void print_info()
> {
>   Serial.print("SSID_AP : ");
>   Serial.println(ssid_ap);
>   Serial.print("PASS_AP : ");
>   Serial.println(pass_ap);
>   Serial.print("SSID_WIFI : ");
>   Serial.println(ssid_wifi);
>   Serial.print("PASS_WIFI : ");
>   Serial.println(pass_wifi);
>   Serial.print("IP Address : ");
>   Serial.println(ip_addr);
>   Serial.print("Soft AP Ip Address : ");
>   Serial.println(WiFi.softAPIP());
>   Serial.print("MAC Address : ");
>   Serial.println(WiFi.macAddress());
> }

diazaki, 2018-12-25:
Hi, this is my log

> [D][WiFiGeneric.cpp:345] _eventCallback(): Event: 13 - AP_START
> [D][WiFiGeneric.cpp:345] _eventCallback(): Event: 13 - AP_START
> [D][WiFiGeneric.cpp:345] _eventCallback(): Event: 0 - WIFI_READY
> [D][WiFiGeneric.cpp:345] _eventCallback(): Event: 4 - STA_CONNECTED
> [D][WiFiGeneric.cpp:345] _eventCallback(): Event: 7 - STA_GOT_IP
> [D][WiFiGeneric.cpp:389] _eventCallback(): STA IP: 192.168.0.2, MASK: 255.255.255.0, GW: 192.168.0.1

When i try to connect my phone, i got this log
> [D][WiFiGeneric.cpp:345] _eventCallback(): Event: 15 - AP_STACONNECTED

But, when i try to connect my laptop and connected, i got log like this
> [D][WiFiGeneric.cpp:345] _eventCallback(): Event: 15 - AP_STACONNECTED
> [D][WiFiGeneric.cpp:345] _eventCallback(): Event: 17 - AP_STAIPASSIGNED

me-no-dev, 2018-12-26:
the fact that when the phone connects you do not see `AP_STAIPASSIGNED` means that dhcp between the two is not working correctly. You could try and use WireShark to look at the network traffic and obtain a log. If you feel capable, install wireshark on your computer and start capture once it is connected to the ESP. Then try to connect the phone and stop capture once it fails. You can post the capture here and we can have a look deeper into what could be happening.

copercini, 2018-12-26:
Try to do an `esptool.py erase_flash` and load the firmware again, just to be sure before go deeper

diazaki, 2018-12-26:
> Try to do an `esptool.py erase_flash` and load the firmware again, just to be sure before go deeper

Hi, i've tried but with "flash download tool", but still stuck :(. maybe tomorrow i'll try with esptool.py :D

diazaki, 2018-12-26:
> the fact that when the phone connects you do not see `AP_STAIPASSIGNED` means that dhcp between the two is not working correctly. You could try and use WireShark to look at the network traffic and [...]

Okee, maybe tomorrow i'll try info about wireshark :D thank you for the suggestion

diazaki, 2018-12-26:
> > Try to do an `esptool.py erase_flash` and load the firmware again, just to be sure before go deeper
> 
> Hi, i've tried but with "flash download tool", but still stuck :(. maybe tomorrow i'll try with esptool.py :D

Today i've try to erase flash with esptool but still stuck in same problem :(

diazaki, 2018-12-27:
> the fact that when the phone connects you do not see `AP_STAIPASSIGNED` means that dhcp between the two is not working correctly. You could try and use WireShark to look at the network traffic and [...]

Hi, today i've try to use wireshark and got this log.
[image]

And this is my description
- 192.168.19.11 is my laptop IP Address
- 192.168.19.10 is ESP AP IP Address
- 192.168.19.1 is ESP AP Gateway

May be you can help me again? :)

lbernstone, 2018-12-27:
Google "troubleshoot dhcp with wireshark"
https://www.lovemytool.com/blog/2013/09/troubleshooting-dhcp-with-wireshark-by-tony-fortunato.html

mouridis, 2019-01-02:
Since other devices connect with no problem and only your Xiaomi phone has problem to connect, it seems we are debugging your phone and not arduino-esp32 or your sketch.

Make sure there is nothing setup or running on your phone that could possibly interfere with DHCP (no tethering of any kind like wifi hotspot or bluetooth tethering, no VPN setup or VPN app active, no network security apps etc). Also make sure that "DHCP / obtain IP automatically" or whatever the option is called in Xiaomi's MIUI is enabled for your ESP32 SSID.

Also consider restoring factory defaults on your phone and retrying... (make sure you backup your phone first)

diazaki, 2019-01-02:
> Since other devices connect with no problem and only your Xiaomi phone has problem to connect, it seems we are debugging your phone and not arduino-esp32 or your sketch.
>
> Make sure there is nothing [...]

Yes, i think i wanna try to restore factory setting my phone, who make me confuse is why my phone cannot connect only when ESP32 in AP STA Mode, but when ESP32 only AP Mode, my phone can connect to ESP32 :D :D. Today i've buy new ESP32, and got same problem.

I think the problem is in my phone, maybe wifi hardware or something not compatible with ESP32.

me-no-dev, 2019-01-02:
you have taken a screenshot of the wireshark window :) and you can actualy take a .pcap file out of it and post that file here so we can look deeply into the packets. That screenshot up there shows little to no info

diazaki, 2019-01-02:
> you have taken a screenshot of the wireshark window :) and you can actualy take a .pcap file out of it and post that file here so we can look deeply into the packets. That screenshot up there shows little to no info

Hi, i'll try to do that, and i'll upload the result :)

diazaki, 2019-01-03:
Hi, this is my log from .pcap file
[Can Connect ESP32 (Only AP Mode).txt](https://github.com/espressif/arduino-esp32/files/2725435/Can.Connect.ESP32.Only.AP.Mode.txt)
[Cannot Connect ESP32 (AP STA Mode).txt](https://github.com/espressif/arduino-esp32/files/2725436/Cannot.Connect.ESP32.AP.STA.Mode.txt)

I think because the problem is in my phone, so i'm using app tPacketCapture in my phone and trying to get data packet when i'm connected and cannot connected to my ESP32. I'm open that .pcap file in wireshark but i still dont understand to trace, i still must studying about that :D :D

me21, 2019-01-14:
Does your phone connect to ESP32 if you set "AP only" mode?

diazaki, 2019-01-14:
> Does your phone connect to ESP32 if you set "AP only" mode?

Yes, my phone can connect to ESP32 if ESP32 AP Only. even if my ESP32 is set to AP_STA Mode, if that ESP32 not connect to any wifi (just as AP Only), my phone still can connect to ESP32.

i'm still try to find someting configuration in my phone :D

me-no-dev, 2019-01-14:
@diazaki this is not a pcap file :)
Here is how to specify it:

[image]

copercini, 2019-01-14:
you can just zip it and drop here on github ;)

diazaki, 2019-01-14:
Oh, yes, this is my pcap file, i have 4 file, 2 file when i can connect, and 2 file when i cannot connect :)

[ESP32 pcap.zip](https://github.com/espressif/arduino-esp32/files/2757344/ESP32.pcap.zip)

me-no-dev, 2019-01-23:
funny enough I do not see any DHCP messages on either log... how did you connect and obtain IP?

davidbuzz, 2019-01-24:
"an esptool.py erase_flash and load the firmware again" - fixed my problem, which was my perfectly good password being refused by my esp32 in AP mode.

diazaki, 2019-01-24:
> funny enough I do not see any DHCP messages on either log... how did you connect and obtain IP?

i just connect to ESP32 AP like connect to other wifi :)

> "an esptool.py erase_flash and load the firmware again" - fixed my problem, which was my perfectly good password being refused by my esp32 in AP mode.
i've try this step with 2 ESP32 board and my xiaomi phone still can't connect :(

today i've got other mobile phone and i'll try to connect with my ESP32 :D

diazaki, 2019-01-28:
Hi, i've try with my lenovo s890 phone, and that's phone is not get any problem to connect my ESP32. Maybe the problem is actually in my xiaomi's phone

akshar001, 2019-01-30:
@diazaki there is no problem it's with library i am trying it also and having same problem

akshar001, 2019-02-02:
I am trying to connect even laptop from it but it doesn't, there is a simple question to ask if your phone has a problem or not. If your phone has a problem then it can not connect to any other wifi. If your phone is connecting to other router also then it has no problem esp32 has only problem.

pmezavergara, 2019-04-20:
Hola, puedes subir el código completo, estoy e lo mismo, gracias

sonastyle, 2019-05-07:
Use ESP32-Arduino Core build 1.0.2. I faced the same issue on build 1.0.1.

GeorgeFlorian, 2019-07-09:
Any luck with this ?
I am experiencing the same issue !
https://github.com/espressif/arduino-esp32/issues/2910

FrenchLab47, 2019-07-11:
Hello all,
I come to share my experience:
I just noticed a problem connecting my phone on my ESP32 in AP mode. It marks "Wifi connected to the ESP32 access point", but the browser can not display the pages on the correct IP address (I use ESPAsyncWebServer, but it does not seem to be a server problem).
I do not have a problem with a tablet, or another phone without a SIM card. I tested different settings, and I noticed that my problem only occurred on the phone with SIM card when it is connected to Internet (mobile data). I disconnect the mobile data and everything works again. This is not really annoying if it is specified to the users, but I do not have the explanation to this blocking.

Regarding the connection with a computer, I have problems when I use the original IP address of the ESP32 (192.168.4.1) because it is the same address as the management page of my Internet box. This address is used by many Internet boxes. To avoid this connection problem, I gave another IP address to my ESP32 (like 1.2.3.4 or whatever) and it works.

When connection problems are all fixed, there may still be slowness problems with the display of pages. I improved this by indicating the size of the files before sending them:
`response-> addHeader ("Content-Length", (String) file.size ());`

If that can help you...

GeorgeFlorian, 2019-07-15:
> Hello all,
> I come to share my experience:
> I just noticed a problem connecting my phone on my ESP32 in AP mode. It marks "Wifi connected to the ESP32 access point", but the browser can not display [...]

Reading through this repository I found that there are some users that have problems connecting to the ESP32's AP Mode. And there isn't a fix yet for it.

I don't know how to add a header because I haven't done this before. How should that work with the following handler :
```
  server.on("/register", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(SPIFFS, "/indexAP.html", "text/html", false, processor);
  });
```

akshar001, 2019-07-16:
@FrenchLab47, Yes you are right the problem that you mention but i want to have to say something for all of them who are having same problem, the problem is not on esp32 or esp8266 problem is, the smart phones are coded in a way that if they don't found internet connectivity to the wifi they automatically shift all the requests to Mobile Data if it's on. So in our case though the phone is showing you a connected to wifi, the requests are route through mobile data instead of wifi,So all you have to do it is just turn off the mobile data. I wasted my time for this purpose like 2-3 days in the end i found this. Hope it will help you.

joaorabelo1121, 2019-07-21:
> > Hello all,
> > I come to share my experience:
> > I just noticed a problem connecting my phone on my ESP32 in AP mode. It marks "Wifi connected to the ESP32 access point", but the browser can not display the pages on the correct IP address (I use ESPAsyncWebServer, but it does not seem to be a server problem).
> > I do not have a problem with a tablet, or another phone without a SIM card. I tested different settings, and I noticed that my problem only occurred on the phone with SIM card when it is connected to Internet (mobile data). I disconnect the mobile data and everything works again. This is not really annoying if it is specified to the users, but I do not have the explanation to this blocking.
> > Regarding the connection with a computer, I have problems when I use the original IP address of the ESP32 (192.168.4.1) because it is the same address as the management page of my Internet box. This address is used by many Internet boxes. To avoid this connection problem, I gave another IP address to my ESP32 (like 1.2.3.4 or whatever) and it works.
> > When connection problems are all fixed, there may still be slowness problems with the display of pages. I improved this by indicating the size of the files before sending them:
> > `response-> addHeader ("Content-Length", (String) file.size ());`
> > If that can help you...
> 
> Reading through this repository I found that there are some users that have problems connecting to the ESP32's AP Mode. And there isn't a fix yet for it.
> 
> #2910
> 
> I don't know how to add a header because I haven't done this before. How should that work with the following handler :
> 
> ```
>   server.on("/register", HTTP_GET, [](AsyncWebServerRequest *request){
>     request->send(SPIFFS, "/indexAP.html", "text/html", false, processor);
>   });
> ```
> 
> Like this maybe ?
> 
> ```
>   server.on("/register", HTTP_GET, [](AsyncWebServerRequest *request){
>     AsyncWebServerResponse *response = request->beginResponse(SPIFFS, "/indexAP.html", "text/html", false, processor);
>     File file = SPIFFS.open("/indexAP.hmtl");
>     response->addHeader("Content-Length", (String) file.size ());
>     request->send(response);
>     file.close();
>   });
> ```

I have a One Plus6, and I got the same problem related to the Xiaomi. Then I just disabled the mobile data and it worked. Maybe it has to do with the manufacturers. I did a test with other devices such as Samsumg it got connected immediately.

closed 2019-10-04 by stale[bot]