espressif/arduino-esp32 #2243: How can I read mpu6050 via Wire library？

AJo07krate, 2018-12-25:
I cannot read MPU6050 data，I do not know how to deal with this problem

#include <Wire.h>
void setup()
{
	Serial.begin(115200);	
	Wire.begin();
	Wire.beginTransmission(0x68); 
	Wire.write(0x6B); 			 //the power Register, arouse the 6050
	Wire.write(0); 					 
	Wire.endTransmission(true); 		 

}

void loop()
{
	//0x68 is WHO_AM_I Register
		Wire.beginTransmission(0x68);				
		Wire.write(0x3B); 							
		Wire.requestFrom(0x68, 2, true);		
		int val =  Wire.read() << 8 | Wire.read(); 				
		Wire.endTransmission(true); 						

		Serial.println(val);			//the val is 0, I can not know why 

}

AJo07krate, 2018-12-26:
thank you so much! I guess that the mpu6050 may not be aroused ,because I could write zero into the power management register,  I will try again to deal with this problems .

closed 2018-12-31 by me-no-dev

chunibyo-wly, 2021-03-28:
Hello, thanks for your answer, when I get data from, it get -1.
Is mpu6050 has been destroyed?
[image]
[image]

AJo07krate, 2021-03-28:
you maybe need to check the GPIO map, the nodeMCU pin number is different from the mark on the board. -------- 原始邮件 --------发件人： Chunibyo ***@***.***>日期： 2021年3月28日周日 下午4:43收件人： espressif/arduino-esp32 ***@***.***>抄送： AJo07krate ***@***.***>, Author ***@***.***>主    题： Re: [espressif/arduino-esp32] How can I read mpu6050 via Wire library？ (#2243)
Hello, thanks for your answer, when I get data from, it get -1.
Is mpu6050 has been destroyed?

chunibyo-wly, 2021-03-29:
thx, I' ll check it

ashi2967, 2021-03-29:
unsubscribe

Windows 10 版のメ`ル<https://go.microsoft.com/fwlink/?LinkId=550986>から送信

差出人: ***@***.***>
送信日r: 2021年3月29日 13:43
宛先: ***@***.***>
CC: ***@***.***>
件名: Re: [espressif/arduino-esp32] How can I read mpu6050 via Wire library？ (#2243)

thx, I' ll check it