espressif/arduino-esp32 #2251: Expired SSL Certificate

ChrisRomer, 2018-12-29:
SSL certificate expired for the suggested URL, on 12/28/2018:
https://dl.espressif.com/dl/package_esp32_index.json

I also tried the following URL, but am unable to find/select "ESP32 Dev Mode" in Arduino IDE any longer.
https://raw.githubusercontent.com/espressif/arduino-esp32/master/package/package_esp32_index.template.json

mathstudi, 2018-12-30:
Problem fixed :-)).

closed 2018-12-30 by me-no-dev

danuker, 2019-03-29:
This is happening again!

HubertWagnerVE, 2019-09-24:
certs for https://dl.espressif.com/ are expired as of ‎Tuesday, ‎24 ‎September ‎2019 7:47:32 PM

NidasioAlberto, 2019-09-24:
Having the same issue here. Hopefully tomorrow they will fix the problem.

perca23, 2019-09-24:
Having the same issue here.

jamesguang, 2019-09-24:
Same issue here.  They may fix it tomorrow but only valid for 90 days.
It looks that espressif uses some DIY certificate agency ("**Let's Encrypt**" is their name) which is only valid for 90 days. So it will happen again on Dec 23, 2019.
Too bad it will be Christmas time.

Hope espressif can use a good long lasting certificate.
[image]

jamesguang, 2019-09-24:
Work around:

Change the date of your computer to Sept 23 or before. Then go to Arduino IDE board manager. The download should be OK. Then you can install ESP32 now.
DO not forget to change the date back to correct date.

me-no-dev, 2019-09-25:
certificate is fixed. This is a private server that is just linked to espressif domain.