espressif/arduino-esp32 #2258: Feature Request : custom partitions.csv in sketch folder

luc-github, 2019-01-01:
Hi, 
Sorry if this has been discussed before but I did not saw it

Now there are a lot boards and lot of features that affect partition definition: WIFI/BT/SPIFFS/FAT/OTA and of course flash size

So it is hard to list all possible partitions in arduino menu , sorry I do not use platformIO.

For example currently some board like 'esp32-gateway' has even no choice to select the partition kind - so no way to get it working with BT and WiFi and BT easily, 
On my side I use different boards for different purpose and change manually default.csv each time I open a project is little bit hard

Would it be possible to put a custom partition.csv file in sketch directory in a `partition` subdirectory` ?
this one would be taken if `custom partition` is selected in menu, so it add only one menu on all boards 
this would make the partition selection really flexible if default one do not fit the needs

I see 3 impacts (hope not missing some) :
1 - the max file upload size 
this should be taken from app partition size (not sure how)

2 - the path in platform.txt
`recipe.objcopy.eep.pattern={tools.gen_esp32part.cmd} -q "{runtime.platform.path}/tools/partitions/{build.partitions}.csv" "{build.path}/{build.project_name}.partitions.bin"` to always use the csv present in `{build.path}/{build.project_name}/partition/partition.csv` so before need to copy the right one

3 - Impact on upload tool like mkspiffs and flashfatfs
no idea as I do not use them

Not sure it is easy to implement but gain would be big IMHO, including maintenance as no need to  maintain lot of entries

me-no-dev, 2019-01-02:
yeah... I have been thinking of this some time back as well :) but did not find a way to do it. I think we need a different way to handle all those boards... I have not found a fool-proof way for that either (yet)

luc-github, 2019-01-02:
Unfortunatly do do not know well the process build but if you have some direction to give I would be happy to help
I am guessing modifications involve `gen_esp32part` to be smarter like always copy the csv file defined in menu to build directory, if file name is custom, then use the one in sketch directory, if none then use default one. 
For maximum storage (upload.maximum_size) I cannot see if variable could be modified on the fly to fit the app partition size but the check could be done by esptool 

for uploadtools I think adding a parameter to using path to partition.csv to get start address would be the way to do

just my though - would be happy if this happen as board number is growing and already hard to maintain

me-no-dev, 2019-01-02:
it will not really be able to work this way. The IDE build process goes through a few steps and I do not really know if we can plug in between to add a csv in the sketch folder and that then be in the build. `gen_esp32part` is not to be modified ;) it's an IDF tool that should stay as is :) Maybe we can raise this to the official Arduino folks and see if they have an idea how we can do this. I would also like to have a small UI tool for generating proper partition schemes, else users will make mistakes :)

luc-github, 2019-08-01:
Stale do not close me - Thanks ^_^

luc-github, 2019-09-30:
no no, please keep open, thank you

me-no-dev, 2019-10-06:
in master :) name the file `partitions.csv`

closed 2019-10-06 by me-no-dev

luc-github, 2019-10-06:
thank you ^_^

CelliesProjects, 2021-03-04:
It seems like it does not work or I don't get it.

Included in project is `partitions.csv`.

```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x200000,
ffat,     data, fat,     0x210000,0x1F0000,
```

I see the file being copied and it really arrives at build path.

[image]

Then in Arduino IDE after a compile I see 
```
/home/cellie/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-size -A /dev/shm/build/arduino/smartMeterLogger-esp32.ino.elf
Sketch uses 964898 bytes (73%) of program storage space. Maximum is 1310720 bytes.
```
Which indicates that the sketch was compiled using the partition table selected in the Arduino IDE instead of `partitions.csv` from the sketch folder.

me-no-dev, 2021-03-04:
there is no way that we can pass the partition size to Arduino IDE to calculate the size properly ;) just disregard the size(percentage) reported. Your customer partitions are being used

CelliesProjects, 2021-03-04:
But how to solve this one?

Exact same partition file in different sketch folder.
Different sketch, a lot bigger.
```
Sketch uses 1383966 bytes (105%) of program storage space. Maximum is 1310720 bytes.
text section exceeds available space in boardGlobal variables use 57888 bytes (17%) of dynamic memory, leaving 269792 bytes for local variables. Maximum is 327680 bytes.

Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
Error compiling for board ESP32 Dev Module.
```

This won't flash.

With this particular partition table I can solve it by selecting the dev board and the wanted table. ( because it is just a copy)
But what if you have a real custom table?

me-no-dev, 2021-03-04:
Just select the "huge app" partition scheme in the menu so that it does not bother you with the size. Again, there is no way that the build process can inform Arduino of your customer partitions.csv, therefore it can not be properly used in the size calculation.

CelliesProjects, 2021-03-04:
Got it. It does not really work. Bummer.