espressif/arduino-esp32 #2261: Promlem I2C SDA

arnold-b, 2019-01-02:
I2C SDA suddenly always High and sometimes it works

Hello I have a strange Problem. My I2C worked always fine after the change to rc2. but suddenly it works only sometimes. i already changed to a diffrent cpu.

Version 1.0.1 RC2
Hardware ESP-Wroom or ESP-Wrover

SDA: 21 , SCL 22

Simple I2C Scanner find the MCP23017 sometimes.

When he dont find it. I see SDA is only High. SCL works normaly

WHo have one Idear . Thanks Arnold

stickbreaker, 2019-01-02:
@arnold-b are you getting any debug messages on the serial monitor?

set the CORE DEBUG LEVEL  to DEBUG

Then post the output here so that i have something to diagnose.

Chuck.

arnold-b, 2019-01-02:
Hi,

thats the Debug.

Only change is that SDA = 18 and scl = 19

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10088
load:0x40080400,len:6380
entry 0x400806a4
[W][esp32-hal-psram.c:30] psramInit(): PSRAM init failed!
[D][esp32-hal-i2c.c:1297] i2cProcQueue():  Busy Timeout start=0x1c, end=0x4e, =50, max=50 error=0
[E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbdbb8
[I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffb81cc
[I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=0
[I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
[I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
[I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=0
[I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffb8250 bits=0
[I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffb8280
[I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffb822c
[I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=1
[I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=0
[I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=0
[I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=0
[I][esp32-hal-i2c.c:334] i2cDumpI2c(): debugFlags=0x00000000
[I][esp32-hal-i2c.c:311] i2cDumpDqData(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:354] i2cDumpInts(): Debug Buffer not Enabled
[D][esp32-hal-i2c.c:1306] i2cProcQueue():  Gross Timeout Dead start=0x9f, end=0xd1, =50, max=50 error=0
[E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbdbb8
[I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffb81cc
[I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=0
[I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
[I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
[I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=0
[I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffb8250 bits=0
[I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffb8280
[I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffb822c
[I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=1
[I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=0
[I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=0
[I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=1
[I][esp32-hal-i2c.c:334] i2cDumpI2c(): debugFlags=0x00000000
[I][esp32-hal-i2c.c:311] i2cDumpDqData(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:354] i2cDumpInts(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:1097] i2cProcQueue(): Bus busy, reinit
[D][esp32-hal-i2c.c:1297] i2cProcQueue():  Busy Timeout start=0x127, end=0x159, =50, max=50 error=0
[E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbdbb8
[I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffb81cc
[I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=0
[I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
[I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
[I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=0
[I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffb8250 bits=0
[I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffb8280
[I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffb822c
[I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=1
[I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=0
[I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=0
[I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=0
[... 732 lines cut ...]
[I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
[I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=0
[I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffb8250 bits=0
[I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffb8280
[I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffb85b4
[I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=1
[I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=0
[I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=0
[I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=0
[I][esp32-hal-i2c.c:334] i2cDumpI2c(): debugFlags=0x00000000
[I][esp32-hal-i2c.c:311] i2cDumpDqData(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:354] i2cDumpInts(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:1097] i2cProcQueue(): Bus busy, reinit
[D][esp32-hal-i2c.c:1297] i2cProcQueue():  Busy Timeout start=0x166d, end=0x169f, =50, max=50 error=0
[E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbdbb8
[I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffb81cc
[I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=0
[I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
[I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
[I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=0
[I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffb8250 bits=0
[I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffb8280
[I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffb85b4
[I][esp32-hal-i2c.c:330] i2cDumpI2c(⸮

arnold-b, 2019-01-02:
void Adafruit_MCP23017::begin(uint8_t addr) {
	if (addr > 7) {
		addr = 7;
	}
	i2caddr = addr;

//	Wire.begin();
//begin(int sda, int scl){
	//Wire.begin(21,22);
	Wire.begin(18,19,100000);

	// set defaults!
	// all inputs on port A and B
	writeRegister(MCP23017_IODIRA,0xff);
	writeRegister(MCP23017_IODIRB,0xff);
}
Adafruit_MCP23017 mcp;
word value = 1;  
void setup() {  
  mcp.begin();      // use default address 0
  Serial.begin(115200);
  for (int i = 0 ; i < 16; i++)
  {
    mcp.pinMode(i, OUTPUT);
  }
  
}
void loop() {
  delay(400);
  mcp.writeGPIOAB(value);
  value = value << 1;
  if (value > 65536) 
  { value = 1;
  }

}

stickbreaker, 2019-01-02:
What value are your pullup resistors?

arnold-b, 2019-01-02:
10K

arnold-b, 2019-01-02:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10088
load:0x40080400,len:6380
entry 0x400806a4
[W][esp32-hal-psram.c:30] psramInit(): PSRAM init failed!
[V][esp32-hal-i2c.c:1442] i2cInit(): num=0 sda=18 scl=19 freq=100000
[V][esp32-hal-i2c.c:1643] i2cSetFrequency(): Fifo threshold=3
[D][esp32-hal-i2c.c:1297] i2cProcQueue():  Busy Timeout start=0x1d, end=0x4f, =50, max=50 error=0
[E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbdbb8
[I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffb83c0
[I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=0
[I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
[I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
[I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=0
[I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffb8444 bits=0
[I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffb8474
[I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffb8420
[I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=1
[I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=0
[I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=0
[I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=0
[I][esp32-hal-i2c.c:334] i2cDumpI2c(): debugFlags=0x00000000
[I][esp32-hal-i2c.c:311] i2cDumpDqData(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:354] i2cDumpInts(): Debug Buffer not Enabled
[V][esp32-hal-i2c.c:1442] i2cInit(): num=0 sda=18 scl=19 freq=0
[V][esp32-hal-i2c.c:1643] i2cSetFrequency(): Fifo threshold=3
[D][esp32-hal-i2c.c:1306] i2cProcQueue():  Gross Timeout Dead start=0xab, end=0xdd, =50, max=50 error=0
[E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbdbb8
[I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffb83c0
[I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=0
[I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
[I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
[I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=0
[I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffb8444 bits=0
[I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffb8474
[I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffb8420
[I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=1
[I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=0
[I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=0
[I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=1
[I][esp32-hal-i2c.c:334] i2cDumpI2c(): debugFlags=0x00000000
[I][esp32-hal-i2c.c:311] i2cDumpDqData(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:354] i2cDumpInts(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:1097] i2cProcQueue(): Bus busy, reinit
[V][esp32-hal-i2c.c:1442] i2cInit(): num=0 sda=18 scl=19 freq=0
[V][esp32-hal-i2c.c:1643] i2cSetFrequency(): Fifo threshold=3
[D][esp32-hal-i2c.c:1297] i2cProcQueue():  Busy Timeout start=0x13e, end=0x170, =50, max=50 error=0
[E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbdbb8
[I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffb83c0
[I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=0
[I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
[I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
[I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=0
[I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffb8444 bits=0
[I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffb8474
[I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffb8420
[I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=1
[I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=0
[I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=0
[I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=0
[... 66 lines cut ...]
[I][esp32-hal-i2c.c:334] i2cDumpI2c(): debugFlags=0x00000000
[I][esp32-hal-i2c.c:311] i2cDumpDqData(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:354] i2cDumpInts(): Debug Buffer not Enabled
32
[I][esp32-hal-i2c.c:1097] i2cProcQueue(): Bus busy, reinit
[V][esp32-hal-i2c.c:1442] i2cInit(): num=0 sda=18 scl=19 freq=0
[V][esp32-hal-i2c.c:1643] i2cSetFrequency(): Fifo threshold=3
[D][esp32-hal-i2c.c:1297] i2cProcQueue():  Busy Timeout start=0xcd9, end=0xd0b, =50, max=50 error=3
[E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbdbb8
[I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffb83c0
[I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=0
[I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
[I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
[I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=3
[I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffb8444 bits=200
[I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffb8474
[I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffb8420
[I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=1
[I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=0
[I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=0
[I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=0
[I][esp32-hal-i2c.c:334] i2cDumpI2c(): debugFlags=0x00000000
[I][esp32-hal-i2c.c:311] i2cDumpDqData(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:354] i2cDumpInts(): Debug Buffer not Enabled

stickbreaker, 2019-01-02:
Add `if (!Wire.begin(18,19,100000)) Serial.printf("fail=%d/n",Wire.lastError());` after `Serial.begin(115200);`

arnold-b, 2019-01-02:
void setup() {  
  Serial.begin(115200);
  if (!Wire.begin(18,19,100000)) Serial.printf("fail=%d/n",Wire.lastError()); 
  mcp.begin();      // use default address 0
  
  for (int i = 0 ; i < 16; i++)

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10088
load:0x40080400,len:6380
entry 0x400806a4
[W][esp32-hal-psram.c:30] psramInit(): PSRAM init failed!
⸮[V][esp32-hal-i2c.c:1442] i2cInit(): num=0 sda=18 scl=19 freq=100000
[V][esp32-hal-i2c.c:1643] i2cSetFrequency(): Fifo threshold=3
[V][esp32-hal-i2c.c:1442] i2cInit(): num=0 sda=18 scl=19 freq=100000
[V][esp32-hal-i2c.c:1643] i2cSetFrequency(): Fifo threshold=3
[D][esp32-hal-i2c.c:1306] i2cProcQueue():  Gross Timeout Dead start=0x28, end=0x5a, =50, max=50 error=4
[E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbdbb8
[I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffb83c0
[I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=0
[I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
[I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
[I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=4
[I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffb8444 bits=200
[I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffb8474
[I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffb8420
[I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=1
[I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=0
[I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=1
[I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=0
[I][esp32-hal-i2c.c:334] i2cDumpI2c(): debugFlags=0x00000000
[I][esp32-hal-i2c.c:311] i2cDumpDqData(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:354] i2cDumpInts(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:1097] i2cProcQueue(): Bus busy, reinit
[V][esp32-hal-i2c.c:1442] i2cInit(): num=0 sda=18 scl=19 freq=0
[V][esp32-hal-i2c.c:1643] i2cSetFrequency(): Fifo threshold=3
[D][esp32-hal-i2c.c:1306] i2cProcQueue():  Gross Timeout Dead start=0xc2, end=0xf4, =50, max=50 error=4
[E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbdbb8
[I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffb83c0
[I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=0
[I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
[I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
[I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=4
[I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffb8444 bits=200
[I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffb8474
[I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffb8420
[I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=1
[I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=0
[I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=-1
[I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=0
[I][esp32-hal-i2c.c:334] i2cDumpI2c(): debugFlags=0x00000000
[I][esp32-hal-i2c.c:311] i2cDumpDqData(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:354] i2cDumpInts(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:1097] i2cProcQueue(): Bus busy, reinit
[V][esp32-hal-i2c.c:1442] i2cInit(): num=0 sda=18 scl=19 freq=0
[V][esp32-hal-i2c.c:1643] i2cSetFrequency(): Fifo threshold=3
[I][esp32-hal-i2c.c:1097] i2cProcQueue(): Bus busy, reinit
[V][esp32-hal-i2c.c:1442] i2cInit(): num=0 sda=18 scl=19 freq=0
[V][esp32-hal-i2c.c:1643] i2cSetFrequency(): Fifo threshold=3
[D][esp32-hal-i2c.c:1306] i2cProcQueue():  Gross Timeout Dead start=0x166, end=0x198, =50, max=50 error=1
[E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbdbb8
[I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffb83c0
[I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=0
[I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
[I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
[I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=1
[I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffb8444 bits=200
[I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffb8474
[I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffb8420
[I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=1
[I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=0
[I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=1
[I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=0
[I][esp32-hal-i2c.c:334] i2cDumpI2c(): debugFlags=0x00000000
[I][esp32-hal-i2c.c:311] i2cDumpDqData(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:354] i2cDumpInts(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:1097] i2cProcQueue(): Bus busy, reinit
[V][esp32-hal-i2c.c:1442] i2cInit(): num=0 sda=18 scl=19 freq=0
[V][esp32-hal-i2c.c:1643] i2cSetFrequency(): Fifo threshold=3
2
4
8
16
32
64
128
256
512
1024
[I][esp32-hal-i2c.c:1097] i2cProcQueue(): Bus busy, reinit
[V][esp32-hal-i2c.c:1442] i2cInit(): num=0 sda=18 scl=19 freq=0
[V][esp32-hal-i2c.c:1643] i2cSetFrequency(): Fifo threshold=3
2048
4096
8192
16384
32768

stickbreaker, 2019-01-02:
The busy error is saying there is activity on i2c bus not caused by esp32
>[D][esp32-hal-i2c.c:1297] i2cProcQueue(): Busy Timeout start=0x1d, end=0x4f, =50, max=50 error=0

Activity detected between `Wire.begin()` and first use of bus, 30ms(0x1d) after boot.

stickbreaker, 2019-01-02:
I've got circuit with MCP23017 and MCP23008.

Post link for adafruit library.

I'll try with my hardware.

arnold-b, 2019-01-02:
Strange. now i could see a few seconds values on the bus. but then 255 again

stickbreaker, 2019-01-02:
Power? 

Busy will persist from START until a STOP is seen.

START = sda high and scl high then sda low
STOP = sda low and scl high then sda high

stickbreaker, 2019-01-02:
Glitching of SDA (high,low,high while SCL high) will cause the esp32 to set BUSY.

arnold-b, 2019-01-02:
I have realy no Idear. Power is stable on 3V3. Im using external powersupply. 

 

This small thing worked 2 days ago perfect and suddenly I don´t. 

I even changed CPU and MCP23017. No result. 

 

If I remove the mcp and check only the signals. It still always 1 on SDA. Only sometimes after restart it works normal for a few minutes.

stickbreaker, 2019-01-02:
I haven't seen any bus recovery msgs, so SDA or SCL are not stuck low, just BUSY, I think there are glitches on SDA.

stickbreaker, 2019-01-02:
Post your  link to the adafruit library you are using. I'll try to duplicate the problem.

arnold-b, 2019-01-02:
Hi,

 

thats the link: https://github.com/adafruit/Adafruit-MCP23017-Arduino-Library

 

but its not only this part. Even I use I2CScanner I get always SDA = 1. What ever I try. 

On I2C Scanner sometimes I find the parts sometimes not. Even I just connect it with wires. 

 

I think I have somewhere a general problem. But no idea where

stickbreaker, 2019-01-02:
Does Scl=0 show up in the error message.

SDA=1 Is correct
SCL=0 is a problem.

arnold-b, 2019-01-02:
This i could see on the logic analyser. But it comes from the clock, I think. 

The timing of the clock is not stable

arnold-b, 2019-01-02:
In the errormessage from ESP ?. I didn’t see. I will check again

arnold-b, 2019-01-02:
Now i changed ESP32 to ESP8266 and it works normal on 12/14

arnold-b, 2019-01-02:
I used now a totaly new Wroom. Same Problem but sometimes i get data. 
In the Log is no where SCL=0

arnold-b, 2019-01-02:
suddenly it works again. just switched it off for 5 minutes. strange, But this happend already a few times.

arnold-b, 2019-01-02:
Some packet are lost. because the bits in mcp are jumping and i see nack

arnold-b, 2019-01-02:
now i changed the pins again and its working.

beginning i uses 21,22  then 18,19 and now 22,23

stickbreaker, 2019-01-02:
What exact board are you using?

stickbreaker, 2019-01-02:
Vroom and Vrover are just the module

stickbreaker, 2019-01-02:
can you post a oscope image of SDA and SCL?

stickbreaker, 2019-01-02:
I have WeMos bluetooth + battery, Heltec Lora 32.  I haven't had any problems as long as I don't use 6..11(FLASH) 1,3(UART0) >34(input only) LED_BUILTIN(25 on Heltec) led draw doesn't allow pin to go high enough.

arnold-b, 2019-01-03:
Hi Chuck,

 

i tried 4 different boards one is self-made. I start to think 2 MCU are broken.

Now the last test is running already 8 hours without any problem. 

 

Later I will post oscope image of SDA and SCL?, in around 3 4 hours

stickbreaker, 2019-02-19:
@arnold-b Looks like your problem is solved, please close this issue.

Chuck.

closed 2019-08-15 by stale[bot]

hpsaturn, 2021-12-11:
I have the same behavior documented in the next issue:

https://github.com/kike-canaries/canairio_firmware/issues/165

It is a device with 2 years or more working, but in the last release it has the same behavior of this thread here. Maybe the bus is damaged? on the sensor or in the main board? or maybe it is a new issue on the Arduino framework SDK?