espressif/arduino-esp32 #2262: I2c crashes when controlling relay

harsh267, 2019-01-02:
### Hardware:
Board:							ESP32 Dev Module wroom
IDE name:							Arduino IDE? 
Flash Frequency:					40Mh
PSRAM enabled:                                          no
Upload Speed:						115200
Computer OS:                                            Windows 10

### Description:
Hi I have Bme280 ,Bh170 and rtc connected to my I2c on esp32 .
I also have 4 channel 5v coil relay connected using uln2003. 
The code works perfect . When i turn on/off the relay at high frequency or occasionally the i2c sensors start sending garbage values  BH170 will send zero and BMe 280 humidity will go to 100% . Analog sensors will keep on sending data and relays will keep on responding and occasionally while flipping the relays on/off the I2c might recover and start sending proper data. 
Do not understand why it is happening . Trying to put mosfet irfz44n to reset the i2c. 

### Sketch:  
#include <MQ135.h>

/*
  @ harshit --> AIR-GROW
*/
#include <FS.h>
#include <PubSubClient.h>
#if defined(ESP8266)
#include <ESP8266WiFi.h>
#else
#include <WiFi.h>
#include <SPIFFS.h>
#endif
#include <DNSServer.h>
#if defined(ESP8266)
#include <ESP8266WebServer.h>
#else
#include <WebServer.h>
#endif

#include <WiFiManager.h>
#include <ArduinoJson.h>
#include <BH1750.h>
#include <OneWire.h>
#include <DallasTemperature.h>
#include <Wire.h>
#include <Adafruit_Sensor.h>
#include <Adafruit_BME280.h>

#include <WiFiClient.h>
#include <ESPmDNS.h>
#include <Update.h>
//#include <SparkFunCCS811.h>
#include <ArduinoOTA.h>
#include <EEPROM.h>

BH1750 lightMeter;
WiFiClient wifiClient;
PubSubClient client(wifiClient);
WebServer server(81);//81 port handles OTA

char mqtt_server[] = "159.89.168.123";
//char mqtt_server[40] = "demo.thingsboard.io";
char mqtt_port[6] = "1883";
char token[33] = "harshit1";
const char* host = "airgrow"; //for OTA airgrow.local:81
/////////////////////////////////////////////////////////////////

/// The load resistance on the board
#define RLOAD 10.0
/// Calibration resistance at atmospheric CO2 level
#define RZERO 76.63
/// Parameters for calculating ppm of CO2 from sensor resistance
#define PARA 116.6020682
#define PARB 2.769034857
[... 1018 lines cut ...]
  for (int x = 0; x < 10; x++) {                //add samples together
    zzz = zzz + co2now[x];

  }
  co2raw = zzz / 10;                          //divide samples by 10
  co2comp = (co2raw - co2Zero);                 //get compensated value
  if ( co2comp > 1024) {
    Serial.println("Co2 jumping");
  }
  else {
    Serial.println("CO 2 comp " + String(co2comp));
    co2ppm = map(co2comp, 0, 1023, 498, 9000);
    co = String(co2ppm * 10);
  }

}

/*
   harshit: Timer interrupt function
*/
void fnTimerSensorData() {
  set_action = true;
}
//c
//bme on

stickbreaker, 2019-01-02:
edit your issue to make it more readable:
* add three back ticks \`\`\`c++ on a separate line before your code,
* add three back ticks \`\`\`  on a separate line after your code.

Relays are really hard on the electrical environment of a microprocessor, you are seeing an EMP event. You need to filter the power line between your relay driver and the esp32. Add some big caps to your 5v rail probably 470uf to 1000uf, perhaps an inductor between 5V and the ULN2003
.
Software wise, with the current V1.0.1Rc2 you can use `Wire.begin()` to recycle the i2c bus if it is hung.  If one of the slave i2c devices are locked up you will have to powercycle them or use their built in hardware reset features if they have them.  
```c++
// if i2c error detected try
if( ! Wire.begin() ) { // failed to recycle i2c bus with existing parameters.
  Serial.printf(" i2c bus recycle failed error =%d(%s)\n",Wire.lastError(),Wire.getErrorText(Wire.lastError()));
}
```
Chuck.

TopiRinkinen, 2019-01-11:
One often needed component is a diode from relay pin to GND/VCC.

If the relay has one pin fixed on GND and the other pin is driven by ULN2003, then you need diode from non-GND pin to GND ( GND ->|- ULN2003,  "->|-" is diode).

If the relay has one pin fixed on VCC, the diode is from relay GND pin to VCC (ULN2003 ->|- VCC), and the datasheet of ULN2003 reveals that the diode is already there. And a several µF capacitor between ULN2003.COM and GND.

Or transient suppressor diode (TVS, TRANSZORB) between relay coil pins (correct polarity or bipolar TVS, and correct voltage of TVS).

stickbreaker, 2019-05-22:
closed due to inactivity.

closed 2019-05-22 by stickbreaker