espressif/arduino-esp32 #2265: BLE use a lot of memory

atlask1, 2019-01-02:
### Hardware:
Board:							ESP32 DOIT DevKit V1 
Core Installation/update date:			sept 2018
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                        ?
Upload Speed:					921600
Computer OS:                                          Ubuntu

Hi, I'm testing the example BLE uart of the BLE example and I see that the sketch use 1193381 byte, 91% of memory. It's a lot!
There isn't space for other things except BLE, there's some way to reduce memory usage?
thanks in advance

me-no-dev, 2019-01-02:
there are other partition schemes available. Switch to ESP32 Dev Board in the menu and then you can select another partitions that will allow for bigger sketches

atlask1, 2019-01-02:
Thanks a lot! Solved! Thanks

closed 2019-01-02 by atlask1