espressif/arduino-esp32 #2267: ESP stops answering network requests

slompf18, 2019-01-03:
### Hardware:
Board:							esp32doit-devkit-v1
IDE name:						Platform.io
Flash Frequency:					80Mhz
Upload Speed:					115200
Computer OS:                                           Mac OSX

### Description:
I'm trying to deliver a website for configuration of ssid/password. The site contains several js/css/ajax calls. Everything works flawlessly when using it in Chrome. When using it in Safari (ios or macos) the board stops answering when reloading the site (second or third time, not deterministic).

To serve the site I was trying ESPAsyncWebServer and my own little webserver (essentially a pool of WiFiClients). Every solution shows the same results.

So my question is, who to debug this problem?

The sketch I'm providing shows the symptoms. I know I could scan the networks async and then it would work in this little sketch. But doing it sync was the only way I found to demonstrate the problem. In the real application the scan is done async, but the symptoms remain.

### Sketch: 
```cpp
#include <Arduino.h>
#include <ESPAsyncWebServer.h>
#include <SPIFFS.h>

AsyncWebServer server(80);

const char* ssid = "YOUR_SSID";
const char* password = "YOUR_PASSWORD";

void setup() {
    Serial.begin(115200);
    WiFi.begin(ssid, password);
    SPIFFS.begin();
    if (WiFi.waitForConnectResult() != WL_CONNECTED) {
        Serial.printf("WiFi Failed!\n");
        return;
    }

    Serial.print("IP Address: "); Serial.println(WiFi.localIP());

    server.serveStatic("/", SPIFFS, "/").setDefaultFile("index.htm");

    server.on("/getNetworkDetails", HTTP_GET, [](AsyncWebServerRequest *request){
        String result = "Network details: ";
        result += WiFi.SSID() + "/" + WiFi.getHostname();
        request->send(200, "plain/text", result);
    });

    server.on("/getNetworks", HTTP_GET, [](AsyncWebServerRequest *request){
        String result = "Found networks: ";
        result += WiFi.scanNetworks();
        request->send(200, "plain/text", result);
    });

    server.begin();
}

void loop() {
}
```
index.htm:
```html
<html>
<body>
    <div>
        ESP TEST
    </div>
    <div id="networkDetails">loading network details ...</div>
    <div id="networks">loading number of networks ...</div>

    <script>
        document.addEventListener("DOMContentLoaded", function(event) { 
            var xhr1 = new XMLHttpRequest();
            xhr1.open('GET', 'getNetworkDetails');
            xhr1.onload = function() {
                document.getElementById("networkDetails").innerHTML = xhr1.responseText;
            };
            xhr1.send();

            var xhr2 = new XMLHttpRequest();
            xhr2.open('GET', 'getNetworks');
            xhr2.onload = function() {
                document.getElementById("networks").innerHTML = xhr2.responseText;
            };
            xhr2.send();
        });
    </script>
</body>
</html>
```
### Debug Messages:
There are no error messages, even when setting debug level on verbose.

me-no-dev, 2019-01-03:
I have been catching the browsers to open extra connections when requesting the site. I do not guarantee that this is what is going on, but could be the case of having some connection hanging. Generally Async can deal with that and timeout, but who knows what exactly happens... :) maybe with more clues we will come up to a conclusion and fix.

slompf18, 2019-01-03:
There are no extra connections in networks tab of the browsers and I did not see any extra connections when debugging my own test web server. And the sample is simple enough, to not animate the browser to do so. 

Do you have any idea who to find the other clues. ;)

me-no-dev, 2019-01-03:
can you please enable debug and set it to at least INFO. You can do that if you select the ESP32 Dev Board. With Async you should see some logs that mention "timeout".

me-no-dev, 2019-01-03:
Ahh you have even debug enabled... hmmm... ESP8266 users have also complained about this since they switched to newer LwIP (same as on esp32)... question is to trace it down to what is causing it.

luc-github, 2019-01-04:
@me-no-dev looks like similar behavior I reported with telnet connection stopping answering after few exchange, no ?

plewka, 2019-01-18:
Hello, my first post...I'm in trouble with this Problem, too. 
Even a ping to the ESP stops, nothing visible on core debug level _verbose_.
Events from Ethernet disconnect (plug cable) etc. are getting dead.
No idea how to detect the failure within the system itself to do a reboot...

I isolated the problem to a simple webserver using ESPAsyncWebServer, SPIFF over wired Ethernet.

I tried with different Olimex ESP32 EVBs. 
(Arduino 1.8.8 and additional stuff in release versions of this week):

`//#include <Arduino.h>
#include <SPIFFS.h>
#include <ESPAsyncWebServer.h>
#include <ETH.h>

AsyncWebServer server(80);

void setup()
{
  Serial.begin(230400);   
  ETH.begin();
  ETH.config(0xc805a8c0, 0x0105a8c0, 0x00ffffff); // 192.168.5.200 / 192.168.5.1 / 255.255.255.0
  if (!SPIFFS.begin(true)) {
    Serial.println("An Error has occurred while mounting SPIFFS");
    return;
  }
   server.serveStatic("/", SPIFFS, "/");
  server.begin();
}
`

This is enough to fail. 
I now doubt it is caused by the wired ethernet. If I access via a WLAN-based client the propability for a problem is much lower, than with 1:1 direct cable connection without switch etc. but it still fails after an hour or more.
Propability to fail seems to increases with the time it takes for the response.
If I increase CPU clock the propability is getting lower, too. I don't have to do massive traffic.
It's enough to do a simple html which makes the browser load a few pictures with simultanous requests.
I even tried to put files to RAM. This only decreases propability, too. If there is only one (big) transfer
per time it seems to be stable forever...

---

plewka, 2019-01-19:
**WLAN** == Ethernet == Hanging
I just tried via WLAN-Client and ESP via WLAN, too:

It is more difficult to force the bug, but it is there. Some reloads with browser cache deactivated
and it hangs. I used watch -n 0.1 wget <url> to load a "big" file of 300kB and and Firefox'
reload.
In Firefox (web developer-> Network analysis) requests show up to be are less simultanous
than with wired ethernet. For sure I deactivated the cache inside web developer to force the browser
to get the small files on any reload.

slompf18, 2019-01-20:
I did not geht this bug fixed, because nobody was able to tell me how to analyse it. Now looking into RTOS. Seems to run more stable.

plewka, 2019-01-20:
If FreeRTOS is stable doing something equivalent here this sounds like an issue at the interfacing of AsyncTCP and lwIP. AsyncTCP is full of great features you won't implement by your own, though. But something fully freezes the lwip stack. I tried to put load to my SOnOffs, but they limit traffic/connections quite soon but continue to answer even if it takes a minute or more. 

Maybe good idea to disable one Core to prevent SMT? I don't need the speed. Is there anybody having access to one of the single core ESP32s or knowing how to manipulate FreeRTOS options?!

slompf18, 2019-01-25:
When working with RTOS I experienced random panics that lead to a reboot. The error looks like the one I experienced when working with Arduino. 

The problem had its root in the Watchdogs running in the background (described [here](
https://www.linkedin.com/pulse/watchdog-processing-esp-idf-neil-kolban) for example). The default time out in RTOS is 20 seconds. But that doesn’t seem to matter. If the system is in idle for 19 seconds and then starts a job that takes 3 seconds, the system seems to reboot. I made this work by calling esp_task_wdt_reset() before certain operations. 

Maybe we do have the same problem here?

plewka, 2019-01-25:
No, this is fully different. The system is basically fine and fully responsible, no reboot and no (verbose) message! 
It simply doesn't respond to network requests anymore including PING. 
It even doesn't detect a cable disconnect when used over cable (not related to cable though).
Up to my limited knowledge I tend to say the stack fully hangs.
I recognized a delay(250) ms in main loop() strongly increases probability, too.
There only has been an if which doesn't trigger and the delay in the main loop.
Anything which takes some time anywhere seems to be harmful.

Is Arduino really using both cores?!

allex1978, 2019-01-31:
The same issue on latest (30/01/19) Core and AsyncWebServer . Esp32 randomly stops answer to network including a ping...but CPU works fine. i see it on display.

malbrook, 2019-03-27:
I have a similar problem using AsyncWebServer on a couple of different ESP32 projects. The web browser on a PC is connected to the ESP32 via a WiFi network displaying the html pages which use regular javascript ajax calls to the server to update sections of the screen without updating the whole screen. These calls happen every 2 seconds on one system and around 30 seconds an the second project. On both projects the web browser loses connection after an indeterminate time period after which we can no longer detect the ESP32 on the network using a network scanner . 

Both projects also have an access point running on the ESP32 and this is also disappears as well. Using verbose debug I find a message rx timeout and ack timeouts just before the WiFi packs up. I can see the ESP32 is still running as IO signals are still working changing LEDs in response to inputs, I also run a second process on the second core and I can see that this is also running even outputting via a serial port so the ESP32 itself is still operating.

Looking for the source of the messages I found they are generated in AsyncTCP and I noticed that when these messages are generated there is a call to _close() which I suspect is closing down the WiFi , so as a test I added a global variable in AsyncTCP that I could monitor this in the rest of the process, and if the message is triggered this is used to stop and then restart the WiFi. This seems to have reduced the problems by up to 50%, however I am now getting system crashes after some of the restarts, mainly relating to heap poisoning, so it clearly needs more work to solve the problem.

The problem can be regularly caused by opening a web page that contains configuration information, make a change and click the update button which causes the page to send a POST to the server which in turn will cause a write to the preferences, which are stored in SPIFFS, and then immediately select a new page on the browser causing the browser to request a new page from the server which in turn is served from the SPIFFS area. Does this imply that there is a problem when writing to the preferences area of the flash and at the same time reading from a different area of the flash causing an issue with the WiFi.

System is coded using Arduino IDE 1.8.9 with ESP32 1.0.2rc1 and SDK V3.3, AsyncWebServer and AsyncTCP are latest versions from github.

BlackBird77, 2019-05-24:
Me too...
Arduino 1.8.7 with ESP32 1.0.2 freezes the "Network Stack" completely. CPU is fine! Ping no chance!
Same with PlatformIO!

Arduino 1.8.7 and ESP32 1.0.1 is better, but some pings are lost. The latency goes higher and higher up 
to 600ms and than fall back to 1ms.

Here my minimal code, to reproduce. Take the IP from Serial and Ping the ESP! 

```
#include "WiFi.h"

void setup()
{
  Serial.begin(115200);
  Serial.setDebugOutput(true);

  Serial.println("Start Wifi..");
  WiFi.begin("***", "***");

  Serial.println("Started.. Wait for IP...");

  while (WiFi.status() != WL_CONNECTED) {
     delay(500);
     Serial.print(".");
  }  
  Serial.println();
  Serial.print("Got Ip: ");
  Serial.println(WiFi.localIP());
}

void loop()
{
  
}
```

BlackBird77, 2019-06-03:
Ok it looks to me, that the newest platform with ESP-IDF3.2.0 has the freeze problem!

The slow pings come from WiFi power saveing,  which can be turned off. Then the pings are okay with ESP-IDF 3.1.3

closed 2019-08-16 by stale[bot]

DhrubojyotiDey, 2019-11-30:
So can someone please guide on how to close this network freezing issue. It really does pose a problem. And is there any micropython code for this?

zekageri, 2021-01-15:
Still an issue.

dheydeck, 2021-05-11:
Adding this comment because the symptoms described in @slompf18's original post were almost identical to mine- using Safari (on iPhone in my case) to an ESP32 Devkit project web server responded (intermittently) OK on first access but timed out on further attempts to access it. Access from Chrome on Windows was fine although Chrome on my iPhone behaved just as Safari. In addition, running the "Net Analyzer" iPhone app only intermittently showed my project, sometimes showed it as running Bonjour services and rarely, showing Bonjour/Pingable/Web interface. Using Net Analyzer's link to Safari almost invariably resulted in a timeout from the project. The project itself is @Edzelf's outstanding ESP32-Radio project which uses WiFiServer as its embedded webserver.
Particularly irksome for me was that, up to a point, my project had been working flawlessly with my iPhone.
From past experience, if there are not lots of search results for an issue that I'm having, it's probably because I've done something unusual.
Having checked that I had the latest Arduino IDE (1.8.13) and that all my libraries were up-to-date, I thought about anything else that I might have changed, and looked at the settings in the IDE's "Tools" section. The only thing that I saw that I knew that I'd changed from the default for the "ESP32 Dev Module" board at some point was the PSRAM setting- I had disabled this for another project.
I set PSRAM to "Enabled" and reloaded my project. Presto! Perfect response from the project's web server! Net Analyser also responded consistently with the full set of Bonjour/Pingable/Web interface.
I acknowledge that I have not provided the volume of information and diagnostics that others have contributed to this thread but I hope that this entry helps anyone searching, as I did, on the terms esp32 and safari, which resulted in this thread being the first hit.
Your mileage may, of course, vary. Good luck.

aleuarore, 2021-05-12:
Still?

phil31, 2021-06-23:
i can confirme this trouble too .. no update/news ?
thanks

chegewara, 2021-06-23:
I tested with example code from this issue and here is result:
```
64 bytes from 192.168.0.103: icmp_seq=153 ttl=255 time=94.0 ms
64 bytes from 192.168.0.103: icmp_seq=154 ttl=255 time=15.1 ms
64 bytes from 192.168.0.103: icmp_seq=155 ttl=255 time=38.0 ms
^C
--- 192.168.0.103 ping statistics ---
155 packets transmitted, 155 received, 0% packet loss, time 154227ms
rtt min/avg/max/mdev = 12.645/63.517/133.061/30.096 ms
```

How many pings do i have to send to confirm the issue or to confirm it works fine?

phil31, 2021-06-25:
u right, my problem is maybe not similar, PING continue to works, as you, but webserver hang time to time, for 10/20 seconds, then restart to work !

chegewara, 2021-06-25:
What i mean is that maybe 150 ping requests is not long enough to reproduce issue and i should have wait a bit longer.

@phil31 if your problem is different, please open new issue with minimal code to reproduce and informations about version/branch etc

Darktemp, 2021-07-10:
Hi, it seems, that I have a similar issue trying several approaches to avoid it, but it always ends in the situation that I can still ping the ESP (like mentioned with increasing latency) but I cannot connect to it (TCP/UDP) and it cannot connect to mqtt.
Is there any hint what I could do to narrow down the reason?
Annoyingly it only happens very randomly after around 5-7 days; the longest it took was 41 days until it froze.

Darktemp, 2021-08-15:
If someone finds this via google, I found two reasons which should improve the situation:
https://github.com/espressif/arduino-esp32/pull/5487 which should be part of the next release (hopefully) and
issue #4736 .
Still takes ~ 41 days until I know if these were the last causes of the problem 😆

Darktemp, 2021-09-28:
ok, I can confirm, that it never froze up anymore since 16.August!

szerwi, 2023-05-22:
Any updates regarding this issue? Is there any fix planned to be implemented?

VojtechBartoska, 2023-05-23:
@szerwi Do you still face this on latest Arduino Core version 2.0.9?

zekageri, 2023-05-23:
@szerwi Everyone has a problem with ESPAsyncWebServer. Unfortunatelly it is buggy.

My best shot was these forks

https://github.com/yubox-node-org/AsyncTCPSock and https://github.com/yubox-node-org/ESPAsyncWebServer

These are really stable for me but they inherit the same buggy design from the original library.

Websocket clients can stuck in there since there is a possibility that the client does not close the socket cleanly.

Edzelf, 2023-05-23:
Check the result of `heap_caps_get_largest_free_block ( MALLOC_CAP_8BIT )`.  It should be well over 20k for a stable WiFi connection at any time.