espressif/arduino-esp32 #2268: WiFiClient client and WiFiClientSecure client, setTimeout doesn't work

robertgregor, 2019-01-03:
Hello,
I am using 1.0.1rc1. However, seems to me, that the setTimeout() doesn't work as it should. I have this code:
WiFiClient client;
client.connect(preferences.getString(PSTR("smtpserver")).c_str(), preferences.getInt(PSTR("smtpport")));
client.setTimeout(2000);
String serverResponse = client.readStringUntil('\r');
serverResponse.trim();
logSketch('D', PSTR("%s"), serverResponse.c_str());

So on client.readStringUntil('\r'); the code is blocked forever, if the \r character is not received in the input data. Based on the arduino, it should either wait for \r or it should return after the timeout and continue.

https://www.arduino.cc/reference/en/language/functions/communication/stream/streamreadstringuntil/

readStringUntil() reads characters from a stream into a String. The function terminates if the terminator character is detected or it times out (see setTimeout()).

Jeroen88, 2019-01-03:
WiFiClient::setTimeout is in seconds, so 2000 seconds is _almost_ forever ;)

robertgregor, 2019-01-03:
Hmmm, OK, the arduino is saying, that it is ms:

https://www.arduino.cc/reference/en/language/functions/communication/stream/streamsettimeout/

setTimeout() sets the maximum milliseconds to wait for stream data, it defaults to 1000 milliseconds. This function is part of the Stream class, and is called by any class that inherits from it (Wire, Serial, etc). See the LANGUAGE Stream class main page for more information.

But yes the code looks strange:

int WiFiClient::setTimeout(uint32_t seconds)
{
    Client::setTimeout(seconds * 1000);
    struct timeval tv;
    tv.tv_sec = seconds;
    tv.tv_usec = 0;
    if(setSocketOption(SO_RCVTIMEO, (char *)&tv, sizeof(struct timeval)) < 0) {
        return -1;
    }
    return setSocketOption(SO_SNDTIMEO, (char *)&tv, sizeof(struct timeval));
} 

This looks, that it is really seconds:

    Client::setTimeout(seconds * 1000);

But I have a feeling, that it is a bug and that the value should be ms...

Jeroen88, 2019-01-03:
> Hmmm, OK, the arduino is saying, that it is ms:
>
> https://www.arduino.cc/reference/en/language/functions/communication/stream/streamsettimeout/

This is referring to the Stream that indeed has a timeout in ms on the ESP32, like on the Arduino. However you are calling the _WiFiClient_ timeout(). I do not know what time unit is used for the WiFiClient on the Arduino.

closed 2019-08-15 by stale[bot]