espressif/arduino-esp32 #2269: WiFiMulti cannot reconnect to AP

TopiRinkinen, 2019-01-03:
Hi,

Just jumped from ESP8266 world to ESP32. Chinese ESP32 DEVKITV1 (Doit clone, I guess). Arduino 1.8.8 on Win10. All libraries up-to-date.

Downloaded the board from 
https://dl.espressif.com/dl/package_esp32_index.json
(Did anyone else notice that their SSL certificate was expired at the end of December, now they have revoked it)

The sketch below, connects to the AP if the AP is up while booting the board.

When connected, I shut down the AP, and boot it up again => ESP32 cannot connect to it anymore.

Based on discussions on this forum, I added the following line to the WiFiMulti.cpp:

```
if(bestNetwork.ssid) {
    DEBUG_WIFI_MULTI("[WIFI] Connecting BSSID: %02X:%02X:%02X:%02X:%02X:%02X SSID: %s Channal: %d (%d)\n", bestBSSID[0], bestBSSID[1], bestBSSID[2], bestBSSID[3], bestBSSID[4], bestBSSID[5], bestNetwork.ssid, bestChannel, bestNetworkDb);

    WiFi.disconnect(true); // <= Added this line
    WiFi.begin(bestNetwork.ssid, bestNetwork.passphrase, bestChannel, bestBSSID);
```

and then it started working as expected.

The question being: Is the arduino_esp32_core or whatever it is called I am using upto date when the board is loaded from dl.espressif.com link stated above?
If not, what clicks I should do on Arduino IDE?

Br, -Topi

```
#include <Time.h>
#include <TimeLib.h>

#include <NTPClient.h>
//#include <ESP32WiFi.h>

//#include <ESP8266WiFiMulti.h>

#include <WiFi.h>
#include <WiFiMulti.h>

#include <WiFiUdp.h>

#include <OneWire.h>
#include <DallasTemperature.h>

#include <PubSubClient.h>

const char *SSID_1 = "SSID";
const char *SSID_1_PW = "password";
const char *SSID_2     = "second SSID"; 
const char *SSID_2_PW = "and it's password";

WiFiMulti wm;

void setup() {
  Serial.begin(115200);

  wm.addAP(SSID_1, SSID_1_PW);
  //wm.addAP(SSID_2, SSID_2_PW);
}

void loop() {
  static int n = 0;

  if(wm.run() == WL_CONNECTED)
  {
    Serial.printf("WiFiMulti connected\n");
  }
  
  Serial.printf("%d\n", n++);

  delay(1000);
}
```

lbernstone, 2019-01-03:
A pre-release version 1.0.1-rc2 is available. The very latest development code is only available through github.  The are [installation instructions](https://github.com/espressif/arduino-esp32#installation-instructions) on the main page.

meltdown03, 2019-01-05:
The newest git version still has this problem on my Espressif ESP32 DevKit-V4. Running the WiFiMulti.ino example results in an endless loop of "WiFi not connected" printed to the console after the SCAN_DONE status. Adding the line like you said allows it to reconnect, but I thought it was supposed to rescan and select a different AP if necessary. That's how the ESP8266 works. It only seems to reconnect to the same AP it first connected to. I tried WiFi.disconnect(true, true), which erases AP, but it still doesn't rescan. I've gotten it to work with the default WiFiMulti.cpp code by fiddling with my program code a bit. Do you have the link to page of the the fix you are using?

TopiRinkinen, 2019-01-05:
I tried the my example code with latest stable version (from dl.espressif.com) added with the disconnect(true) line, with two APs (two mobile phones). I switched APs on and off and tried probably all combinations, and it seems to work in all occasions as expected (when disconnected, it automatically finds the first available AP in the list. If the behavior in ESP8266 is different to this, I didn't realize it earlier).

I couldn't find again the page which directed to the disconnect(true), sorry.

Br, -Topi

meltdown03, 2019-01-05:
Yes, that is exactly how the ESP8266 works, I tried the ESP32 again with that line added and now it is not working again, freezes with:
`[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED`
`[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 200 - BEACON_TIMEOUT`
`[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY`
`[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START`
`[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START`
`[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED`
`[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND`
`[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY`
`[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START`
`[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START`
`[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED`
`[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND`
`[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY`
`[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START`
`[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START`
`[D][WiFiMulti.cpp:144] run(): [WIFI] delete old wifi config...`
`[D][WiFiMulti.cpp:147] run(): [WIFI] start scan`

It works about 1 out of about 20 times, maybe when I tried it before it was just a fluke.

closed 2019-08-15 by stale[bot]

LeoDJ, 2022-12-20:
I'm running into the same issue.
The ESP32 running WiFiMulti won't reconnect to the AP when it was down for a short time.

Didn't try your proposed fix yet, as I don't like messing around with core libraries, as those changes are not portable and need to be remembered.

PS: I really dislike stale bots.