espressif/arduino-esp32 #2273: Lolin D32 pro - issue with built-in SD Card slot when TFT is connected

merdok, 2019-01-04:
Hi,

I have a problem with the built-in SD Card slot on my Lolin D32 Pro. When there is no TFT connected to the TFT port the SD Card slot is working fine and the card is recognized, but when i want to use both TFT and SD Card the mounting fails with the following error:

`[W][sd_diskio.cpp:517] ff_sd_initialize(): READ_OCR failed: 1FE0003
[E][sd_diskio.cpp:739] sdcard_mount(): f_mount failed 0x(3)
Card Mount Failed`

My test code is pretty simple:

`#include "FS.h"
#include "SD.h"
#include "SPI.h"

void setup() {
  Serial.begin(115200);
  if (!SD.begin(4)) {
    Serial.println("Card Mount Failed");
  }

}

void loop() {
  delay(500);
  Serial.println("test");
}`

Interesting is, when i first boot the D32 Pro without a TFT connected the SD Card works, after it boots and i connect the TFT to the TFT port and do a soft reboot using the onboard reset button, both the TFT and SD Card work in that case. But as soon as i disconnect the micro USB and connect it back again the SD Card fails again.

Is there something i am missing or is this a bug?

merdok, 2019-01-04:
Could you tell me how to do this? I am pretty new in this topic...
This code doesn't work:

`
 digitalWrite(12, HIGH);
 digitalWrite(14, HIGH);
`

fabiuz7, 2019-01-05:
Have you added pinMode(12, OUTPUT) before digitalWrite? You need to configure the pins before use them

Fabiano Riccardi

merdok, 2019-01-05:
Just tried that, still the same issue.

merdok, 2019-01-05:
Do you have an sd card plugged in? The tft works perfectly, just the sd card fails to mount when the tft is connected.
I am using ESP32 for Arduino version 1.0.1 rc2...

michaelkleinhenz, 2019-01-21:
I have a similar problem, but with a RFID reader connected to the SPI instead of a TFT. It looks like the SD driver of the esp32 only runs correctly when connected as the only SPI device to the bus. See this thread here: https://esp32.com/viewtopic.php?t=7107. 

Semaphores are not an option for me, the RFID module is sending data. I get the issue even when just connecting (and not initializing) the module. The only way to get the SD to work is to disconnect the power to the RFID module.

Since this is an issue in the espressif sd driver, I would also assume this is also an issue when using a different SD card module externally. 

So question to the community: is it possible to use two distinct SPI busses on one ESP32? Or is there any other way around this issue? This whole think looks rather awkward to me.

michaelkleinhenz, 2019-01-21:
Looking at the schematics at https://wiki.wemos.cc/_media/products:d32:sch_d32_pro_v2.0.0.pdf, it looks to me that the Lolin TFT gets both the chip select lines for the TFT *and* the SD card (TFT_CS and TF_CS). I always asked me why. Perhaps this is a hack to make the Lolin TFT work with the SD card. This looks to me that the TFT then "knows" when the SD card has bus activity and is standing down for that time - but I am not an expert on this level there.

michaelkleinhenz, 2019-01-21:
So, to sum up: what kind of TFT are you using? The Lolin 1.44" one? Maybe you and @fabiuz7 are using different TFTs.

michaelkleinhenz, 2019-01-22:
For my application, I need to have an NFC reader, an I2S audio amp and the SD card connected to the D32. I can't get this to work as either NFC(i2c) conflicts with the amp(i2s) or NFC(SPI) conflicts with SD(SPI). This is just annoying. I cant get all devices to work at the same time.

michaelkleinhenz, 2019-01-22:
I sort of give up. I suspected that it would be fixed with a newer version of the esp32 devtools as the above note of the limitation was removed from the manual. But it still does not work. Even worse, the SD card now initializes properly, but as soon as I use it, the NFC reader becomes unresponsive and effectively diabled. This is just embarrassing and seriously limits the use of SD card modules with the Lolin D32.

merdok, 2019-01-22:
Someone already had a similar issue a few months ago and this is still not fixed.: https://github.com/espressif/arduino-esp32/issues/272#issuecomment-438857471
I also think that this is a really huge issue since you cannot use multiple devices even they have they own dedicated ports (SD card reader and TFT connector) so I wonder why this issue still exists...
@egzplicit did you manage to fix your issue from that post?

jabadabado, 2019-01-22:
Of course the SD card and the TFT display can not have the same chip select. Then it is a design fault on the board.

merdok, 2019-01-22:
@jabadabado 
How come that it works when i disconnect the TFT, boot the ESP32 with an SD Card and then hotplug the TFT back in? I can then reboot the Lolin D32 pro with the reset button and both of them work, until i cut the power completely. Then the issue is there again...

jabadabado, 2019-01-23:
How can it work to begin with, if the ESP32 is talking to both the SD and the TFT when it is sending on the SPI, it might work because the one does not understand the others commands, but that is bad to begin with. In Chinese and English there are some words that sounds the same, but they don't mean the same. Mayo, means No in Chinese, but Mayonnaise in English, look here we had an SPI problem...
I don't know why it works, but I saw a 2nd CS pin to the TFT, so that might differetiate them, but the SD still gets everything the TFT gets when that CS is active and something coming on the SPI.

Maybe the init should be different when all registers has lost its power compared to when all registers still have power when you press reset. Print out all SPI registers etc att boot, then you can see if any difference between a soft or hard boot.

merdok, 2019-01-23:
I could try that. How can i print all the SPI registers?

jabadabado, 2019-01-23:
Find out which the SPI registers are, convert the values to ASCII text so you can see it on your serial monitor. Add some text before each register for more clarity. Add this before your init();
Not sure how this works in Arduino, normally the init is in the beginning of main(), but Arduino seems to have split it up to init() and loop(), I asume init starts first, so add it there before any other init statements.

merdok, 2019-01-23:
Ok, but what are the commands to read out the SPI registers?

jabadabado, 2019-01-23:
Come on, read the technical reference to find the registers.

michaelkleinhenz, 2019-01-24:
The issue is not related to the same chip select lines being used. It is an issue with the core esp32 SD driver. The Espressif documentation states the following:

> Currently SD SPI driver assumes that it can use the SPI controller exclusively, so applications which need to share SPI bus between SD cards and other peripherals need to make sure that SD card and other devices are not used at the same time from different tasks.

On the Lolin, the SD and the TFT *do* use different CS lines. @merdok @jabadabado

merdok, 2019-01-24:
@michaelkleinhenz thanks for the explanation. Then i guess this is a bug in the driver, but i am also wondering why it happens on the Lolin boards?
Maybe on other boards it is also the case just that there it is not that obvious. The Lolin D32 pro has a dedicated SD card reader and TFT port so it is relatively easy to connect two devices to the SPI at them same time...

michaelkleinhenz, 2019-01-24:
Ok, I tried the same setup with an Adafruit ESP32 Feather and the *exact same issue* appears. It really looks like the SD driver in the ESP32 SDK requires no other devices on the bus.

michaelkleinhenz, 2019-01-24:
@merdok there is an open issue on this in the Espressif repo and some background info: https://github.com/espressif/esp-idf/issues/1597#issuecomment-365364458. This looks like there will be no fix as there is a physical requirement for the SD protocol to control the CS line. I switched over to MMC for accessing SD cards. The modules are even cheaper because you just directly access the SD card's pins. The downside is that you'll need more pins.

merdok, 2019-01-24:
But what i don't understand, everything works if i boot the esp32 without the tft connected and afterwards connect the tft. In that case both sd card and TFT work without any issues, so this has to be just some software issue. This is obviously a bug and should be fixed...

michaelkleinhenz, 2019-01-24:
I wouldn't bet on it. This could just be a coincidence that the TFT just ignores the jitter on the bus post-reset. I try to get an NFC module to work with the SD and it doesn't work reliably. Sometimes stuff gets through, but most of the time one of the modules just stops working or is outputting jitter.

michaelkleinhenz, 2019-01-24:
There is also a secondary SPI bus usable on the ESP32. I am currently trying to get it working. There is an example in the esp examples.

merdok, 2019-01-24:
Any solution for me would be good so if the secondary SPI bus is a solution then i would be glad. My dream is to use both sd card and tft at the same time :p

merdok, 2019-02-04:
I'm still wondering why would Lolin add those stuff to their board knowing that they cannot work together at the same time....

carolwi, 2019-02-05:
I had the same problem when using an ESP-32F and the ESP LOLIN D32 PRO.
The project has three devices using the SPI-bus with dedicated ports (CS).
The solution/workaround was not to use the declaration of the tft object 
as given in the examples of the library, but the one WITHOUT the parameters 
for the SPI channel and then define the SPI separately.

Below there is a summery of the main actions:

#include "WiFi.h"
#include "SPI.h"
// SD (device 1)
#include "SD.h"
#include "FS.h"
// TFT (device 2)
#include <Adafruit_GFX.h>    // Core graphics library
#include <Adafruit_ST7735.h> // Hardware-specific library for ST7735
// VS1053 (device 3)
#include "vs1053_ext.h" // source: https://github.com/schreibfaul1/ESP32-vs1053_ext

// General (fixed): used by SD, TFT and VS1053
#define SPI_MISO      19
#define SPI_MOSI      23
#define SPI_SCK       18

// The free pins I used for the ESP LOLIN D32 PRO
// to connect various units.
// SD
#define SD_CS          4
// TFT
#define TFT_CS        14
#define TFT_RST       33  // Or set to -1 and connect to Arduino RESET pin
#define TFT_DC        27
//#define TFT_LED       32   
//VS1053
#define VS1053_CS     15
#define VS1053_DCS     2
#define VS1053_DREQ   13
#define VS1053_RST     0

// IMPORTANT
// don't use this -> Adafruit_ST7735 tft = Adafruit_ST7735(TFT_CS, TFT_DC, SPI_MOSI, SPI_SCK, TFT_RST);
Adafruit_ST7735 tft = Adafruit_ST7735(TFT_CS, TFT_DC, TFT_RST);  // <- use this  IMPORTANT
VS1053 mp3(VS1053_CS, VS1053_DCS, VS1053_DREQ);

....

//in setup:

void setup() {
   
   // General, set all CS neutral at start
   pinMode(TFT_CS, OUTPUT);      
   digitalWrite(TFT_CS, HIGH);
   pinMode(SD_CS, OUTPUT);
   digitalWrite(SD_CS, HIGH);       // IMPORTANT
   pinMode(VS1053_CS, OUTPUT);
   digitalWrite(VS1053_CS, HIGH);
   pinMode(VS1053_RST, OUTPUT);
   digitalWrite(VS1053_RST, HIGH); 
   
   // Start SPI
   SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
 
   // init TFT
   #ifdef ESP32F    // in case you have an ESP-32F
      pinMode(TFT_DC , OUTPUT);
      pinMode(TFT_CS , OUTPUT);
      pinMode(TFT_RST , OUTPUT);
      pinMode(TFT_LED , OUTPUT);    // turns on the light
      digitalWrite(TFT_LED, HIGH);  // important for ESP-32F
   #endif
   tft.initR(INITR_144GREENTAB);
   tft.fillScreen(ST77XX_BLACK);
   // init SD
   SD.begin(SD_CS);
   
   // Start WiFi if needed
   
   // init VS1053
   mp3.begin();
   mp3.setVolume(volume);  

   .... etc.   
}

merdok, 2019-02-05:
@carolwi thanks for the info. But unfortunately that did not help me :(
I am using the 2.4 touch screen shield based on ILI9341 and i was already initializing it like that:
`Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC, TFT_RST);`
This shield has also the touch screen on CS 12, could maybe this be the issue?

michaelkleinhenz, 2019-02-06:
@carolwi that helps with some specific devices that are "gentle" with the SPI, but is not a general solution. The problem is in the SD card SPI interface design that requires the SD module to "own" the bus. There might be a deeper, electrical reason for doing it like this because the problem is not specific to Lolin or any SD-SPI module. I tried several models, even an Adafruit ESP32 Feather, all with the same problem.

As the MMC bus is working, I wonder why there is no SD-MMC-to-SPI bus conversion module available.

merdok, 2019-05-25:
The cause of issue was a faulty touch screen! After replacing the tft with a working touch screen this is fixed.

merdok, 2019-05-26:
Unfortunately this is still not resolved... It did somehow work with the new touch screen for a moment but now still the same issue. Any news about it? It would be great to be able to use SD card and TFT Touch screen simultaneously.

closed 2019-08-14 by stale[bot]