espressif/arduino-esp32 #2278: GPIO0 connected to GND makes the esp32 enter download mode

Grul-, 2019-01-05:
### Hardware:
Board: ESP32 Dev Module
IDE name: Platform.io
Flash Frequency: 40Mhz
PSRAM enabled: no
Upload Speed: 115200
Computer OS: Windows 10

### Description:
Hello, I have a custom made PCB developed for esp32-devkit-v4 form factory. This PCB basically has a 
a solid state relay controlled by the esp32 module.
Because of the PCB board design the relay is connected to the GPIO0 pin of the esp32 module. So during the boot the GPIO0 acts as it's _held low_.

I've been able to reproduce this behaviour by connecting, with a jumper cable, the GPIO0 pin with a GND pin. In this case the esp32 does not actually boot and the message printed in the serial monitor is below.
In this case, removing the connection between those two pins and rebooting the esp32 it will boot correctly.

So, as I can not re-design the PCB board at the moment and I can not change the esp32 device, is there a way (a workaround) to fix the GPIO0 behaviour in order to not be considered set on low during boot?

### Debug Messages:
```
rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
```

beegee-tokyo, 2019-01-05:
[Reserved GPIOs](https://desire.giesecke.tk/index.php/2018/07/06/reserved-gpios/)

Grul-, 2019-01-05:
Not even by changing the bootloader?

everslick, 2019-01-05:
the first stage bootloader is in ROM - I assume it is handling the GPIO0 boot mode. so no, there is no way around it AFAIK.

Grul-, 2019-01-05:
So the only solution atm is to fix somehow the PCB board, right?

stickbreaker, 2019-01-05:
Yep, hardware problem , hardware solution.

closed 2019-08-15 by stale[bot]