espressif/arduino-esp32 #2282: millis() overflows after 72 minutes

alxstpl, 2019-01-06:
In file esp32\cores\esp32\esp32-hal-misc.c
unsigned long IRAM_ATTR micros()
{
    return (unsigned long) (esp_timer_get_time()) * _sys_time_multiplier;
}

unsigned long IRAM_ATTR millis()
{
     return (unsigned long) (micros() / 1000);
}
method micros works correctly.
method millis works incorrectly, because it depends not from esp_timer_get_time()) , but from micros()
As result maximum value of millis is 2^32 / 1000 = 4294967 ms , 4294 secs or ~72 minutes. 
Instead of 2^32 ms or ~50 days. In some libraries millis() used for time calculations and if value overflows on maximum of uint it works normally with unsigned math calculations, but if value overflows not in maximum value it causes that result of calculated delay never ends (or very long delay). Application hangs at this point and mcu resetted by wdt. Tested one week. every 72 minutes cpu restarts.
I replace 
unsigned long IRAM_ATTR millis()
{
 return (unsigned long) (micros() / 1000);
 }
by 
return (unsigned long) ((esp_timer_get_time()) * _sys_time_multiplier /1000);
and after that app runs about one week without restarts.
Maybe need fix for millis() method?

lbernstone, 2019-01-06:
This is a known issue.  PR is in queue. https://github.com/espressif/arduino-esp32/pull/2250/files#diff-94536ff8ccf321f988ffa45f021f7137L148 shows the line that needs to be changed.

alxstpl, 2019-01-06:
Ok. Thanks!

closed 2019-01-06 by alxstpl