espressif/arduino-esp32 #2283: ESP32 guru meditation error while connecting to AP

vbshightime, 2019-01-06:
### Hardware:
Board:	                                       ESP 32 DEVKIT v1
Core Installation/update date:	       21/10/2018
IDE name:	                                       Arduino IDE
Flash Frequency:	                       80Mhz
Upload Speed:	                               9600			
Computer OS:                                 Windows 10/7

### Description:
I have set up an Acess point on ESP 32. It is successfully configuring. But, while trying to connect to access point I am getting Guru meditation exception

### Sketch:  
```cpp
#include <WiFi.h>
#include "SPIFFS.h"
#include <EEPROM.h>
#include <WebServer.h>
#include <DNSServer.h>
#define EEPROM_SIZE 512
WebServer server(80);
const byte DNS_PORT = 53;
DNSServer dnsServer;
//*********SSID and Pass for AP**************/
const char *ssidAP = "ESPuser";
//********** Variable for SSID and Pass*************/
String STATION_SSID; 
String STATION_PASSWORD;
String DEVICE_ID;
String HOST_ID;
//*********Static IP Config**************/
IPAddress ap_local_IP(192,168,1,4);
IPAddress ap_gateway(192,168,1,254);
IPAddress ap_subnet(255,255,255,0);
//*********AP Timer**************/
unsigned long apTimer =0;
unsigned long apInterval = 120000;

void setup()
{
  Serial.begin(9600); 
  while(!Serial);
  WiFi.mode(WIFI_AP);
  Serial.println(WiFi.softAPConfig(ap_local_IP, ap_gateway, ap_subnet) ? "access point onfigured...":"Not connected");
  Serial.print("Setting soft-AP ... ");
  Serial.println(WiFi.softAP(ssidAP) ? "SoftAp Setup": "Failed to connect");
  dnsServer.start(DNS_PORT, "*", ap_local_IP);
  if (!EEPROM.begin(EEPROM_SIZE))
  {
    Serial.println("failed to initialise EEPROM"); 
  }
  SPIFFS.begin();
  server.on("/", handleRoot);
  server.begin();
  apTimer = millis();
  while(millis()-apTimer<= apInterval) {
         dnsServer.processNextRequest();
        server.handleClient();  }
  InitWifi();
  }
void loop()
{    }
////**************Connect to local WiFi************/
void InitWifi(){
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
        char ssid[30];  char pass[30];char device[30];
        String string_Ssid= read_string(30,0); 
        String string_Password= read_string(30,60);
        String device_id =  read_string(30,120);
        string_Ssid.toCharArray(ssid,sizeof(string_Ssid)+1);
        string_Password.toCharArray(pass,sizeof(string_Ssid)+1);
        device_id.toCharArray(device,sizeof(string_Ssid)+1);
        Serial.println("ssid: "+ string_Ssid);
        Serial.println("Password: "+ string_Password);
  WiFi.begin(ssid,pass);
  while (WiFi.status() != WL_CONNECTED)
  {   delay(500);
      Serial.print(".");
  }
  Serial.println(WiFi.localIP());
 }
//****************************HANDLE ROOT***************************//
void handleRoot() {
   if (server.hasArg("ssid")&& server.hasArg("password") && server.hasArg("device") && server.hasArg("connect") ) {
    handleSubmit();
  }
  else {File  file = SPIFFS.open("/webform.html", "r");
      server.streamFile(file,"text/html");
      file.close();
   }
}

//**************************SUBMIT RESPONSE**************************//
void handleSubmit(){//dispaly values and write to memmory
 String response="<p>The ssid is ";
 response += server.arg("ssid");
 response +="<br>";
 response +="And the password is ";
 response +=server.arg("password");
 response +="<br>";
 response +="the Connection String is ";
 response +=server.arg("connect");
 response +="<br>";
 response +="the device Id is ";
 response +=server.arg("device");
 response +="</P><BR>";
 response +="<H2><a href=\"/\">go home</a></H2><br>";
 server.send(200, "text/html", response);
 
 ROMwrite(String(server.arg("ssid")),String(server.arg("password")), String(server.arg("device")));
 ROMconnectWrite(String(server.arg("connect")));
 }
//**************WRITE Connect String TO EEPROM******************//
void ROMconnectWrite(String conn){
 conn+="!";
 write_EEPROM(conn,200);
 EEPROM.commit();   
}
//**************WRITE RESPONSE TO EEPROM******************//
void ROMwrite(String s, String p, String d){
 s+=";";
 write_EEPROM(s,0);
 p+=";";
 write_EEPROM(p,60);
 d+=";";
 write_EEPROM(d,120);
}
//******************WRITE TO EEPROM**********************//
void write_EEPROM(String x,int pos){
  for(int n=pos;n<x.length()+pos;n++){
     EEPROM.write(n,x[n-pos]);
  }
   EEPROM.commit();   
}
//*****************READ STRING**********************//
String read_string(int l, int p){
  String temp;
  for (int n = p; n < l+p; ++n)
    {
     if(char(EEPROM.read(n))!=';'){
       temp += String(char(EEPROM.read(n)));
     }else n=l+p;
    }
  return temp;
}
//*****************READ CONN STRING**********************//
String read_conn_string(int l, int p){
  String temp;
  for (int n = p; n < l+p; ++n)
    {
     if(char(EEPROM.read(n))!='!'){
       temp += String(char(EEPROM.read(n)));
     }else n=l+p;
    }
  return temp;
}
```
### Debug Messages:
```
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x00000000  PS      : 0x00060430  A0      : 0x800fc409  A1      : 0x3ffb5670  
A2      : 0x3ffbbc90  A3      : 0x3ffbcf6c  A4      : 0x3ffbb9c8  A5      : 0x3ffbb954  
A6      : 0x0501a8c0  A7      : 0x0f01a8c0  A8      : 0x800fc2a8  A9      : 0x3ffb5630  
A10     : 0x3ffbbca0  A11     : 0x3ffbcf6c  A12     : 0x3ffb567c  A13     : 0x00000044  
A14     : 0x00000001  A15     : 0x00000006  SAR     : 0x00000010  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  

Backtrace: 0x00000000:0x3ffb5670 0x400fc406:0x3ffb56b0 0x40104389:0x3ffb56d0 0x40106f35:0x3ffb5710 0x4010cede:0x3ffb5730 0x400fb805:0x3ffb5750

```
### Decoded Messages:
```
PC: 0x00000000
EXCVADDR: 0x00000000

Decoding stack results
0x400fc406: handle_dhcp at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/apps/dhcpserver.c line 1035
0x40104389: udp_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/core/udp.c line 417
0x40106f35: ip4_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/core/ipv4/ip4.c line 705
0x4010cede: ethernet_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/netif/ethernet.c line 176
0x400fb805: tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/api/tcpip.c line 143

```
### Webform:
```
<!DOCTYPE HTML>
<html>
<head>
<meta content="text/html; charset=ISO-8859-1"
http-equiv="content-type">
<meta name = "viewport" content = "width = device-width, initial-scale = 1.0, maximum-scale = 1.0, user-scalable=0">
<title>Mesh</title>
<style>
body { background-color: #F7F9F9  ; font-family: Arial, Helvetica, Sans-Serif; Color: #000000; }
input[type=text], select {
    width: 100%;
    padding: 12px 20px;
    margin: 8px 0;
    display: inline-block;
    border: 1px solid #ccc;
    border-radius: 4px;
    box-sizing: border-box;
}
</style>
</head>
<body>
<center>
<h1 style="color:#3498DB; font-family:verdana">WebCzar</h1>
<h2 style="color:#3498DB; font-family:Times New Roman">Enter Your User credentials</h2>
<FORM action="/" method="post">
<P>
<label style="font-family:Times New Roman"><b>SSID</b></label><br>
<input maxlength="30" name="ssid"><br>
</P>
<P>
<label style="font-family:Times New Roman"><b>PASSWORD</b></label><br>
<input maxlength="30" name="password"><br>
</P>
<INPUT type="submit" value="Submit your form"> 
<style>
input[type="submit"]{
    background-color: #3498DB; /* Green */
    border: none;
    color: white;
    padding: 15px 32px;
    text-align: center;
    text-decoration: none;
    display: inline-block;
    font-size: 16px;
}
</style>
<INPUT type="reset">
<style>
input[type="reset"]{
    background-color: #3498DB; /* Green */
    border: none;
    color: white;
    padding: 15px 32px;
    text-align: center;
    text-decoration: none;
    display: inline-block;
    font-size: 16px;
}
</style>

</FORM>
</center>
</body>
```

vbshightime, 2019-01-10:
I am not frequently getting this exception. But sometimes when the esp32 is getting an inadequate power supply or it is getting warm the exception occurs. I have no clue why this is happening. One more thing, you can see it in code that I have configured a static IP but still, ESP32 connects to DHCP or it's default IP (192.168.4.1). I have resolved this issue by applying this fix to my code

      while(!(WiFi.softAPIP()== ap_local_IP)){
        WiFi.softAPConfig(ap_local_IP, ap_gateway, ap_subnet);    
    }

    Serial.println(WiFi.softAP(ssidAP) ? "soft-AP setup": "Failed to connect");

    Serial.println(WiFi.softAPIP());`

yeah it showed some effective results but I'm still getting this exception once a while

siochs, 2019-01-14:
I also ran into this problem. My solution was to call `WiFi.softAP(WIFI_SSID, WIFI_PASSWORD)` and then `WiFi.softAPConfig(local_IP, gateway, netmask)`. 
Changing the sequence will reproduce the exception on my system.

vbshightime, 2019-01-15:
Thanks for your suggestion @siochs. I did the same thing to resolve my problem. Now I am not getting this exception

treii28, 2019-03-28:
I'm running into the same issue on what appears to be a different version of the idf (line numbers are changed but similar). Changing the order of the softAP andn softAPConfig calls makes no difference.

Decoded backtrace:

```
0x4010c7a2: handle_dhcp at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/apps/dhcpserver/dhcpserver.c line 1031
0x40119281: udp_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/udp.c line 401
0x4011cc19: ip4_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/ipv4/ip4.c line 740
0x40121e4e: ethernet_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/netif/ethernet.c line 184
0x4010f77f: tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/tcpip.c line 135
0x4008f2ad: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

```

vbshightime, 2019-04-10:
Are you frequently getting this issue. Sometimes the device tries to connect to the DHCP IP or it's default IP(192.168.4.1) even when  you have configured the AP. S if changing the sequence doesn't works then try WiFi.disconnect() first and then insert following check  

        while(!(WiFi.softAPIP()== ap_local_IP)){
              WiFi.softAPConfig(ap_local_IP, ap_gateway, ap_subnet);    
              }
                  Serial.println(WiFi.softAP(ssidAP) ? "soft-AP setup": "Failed to connect");
                  Serial.println(WiFi.softAPIP());

closed 2019-04-10 by vbshightime

jayrawatrj, 2020-02-03:
> Are you frequently getting this issue. Sometimes the device tries to connect to the DHCP IP or it's default IP(192.168.4.1) even when you have configured the AP. S if changing the sequence doesn't [...]

i am still getting this error. i already applied thes lines of code but this didn't solve my problem