espressif/arduino-esp32 #2284: ADC 200ksps

folny, 2019-01-07:
Hi

In forums I read that it is possible to use ADC with a speed of 200ksps possible ?.

lbernstone, 2019-01-07:
See https://github.com/espressif/arduino-esp32/issues/2281#issuecomment-451767073

folny, 2019-01-07:
Thanks for your advice but it seems that it will not work for me I need to read 0-1023 at 80kHz timer frequency.

lbernstone, 2019-01-07:
Arduino runs at a lower priority than system functions, so you cannot guarantee reads to happen on a fixed frequency, and even if it could, you would not be able to process the results at that rate.  i2s can pull from adc at that rate and store the results in DMA.

lbernstone, 2019-01-09:
This has come up enough that I decided to make a sketch.  Note that the resolution will always be 12 bits.  I didn't test to see if the attenuation can be changed.
```
#include <driver/i2s.h>

#define I2S_SAMPLE_RATE 80000
#define ADC_INPUT ADC1_CHANNEL_4 //pin 32

void i2sInit()
{
   i2s_config_t i2s_config = {
    .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_ADC_BUILT_IN),
    .sample_rate =  I2S_SAMPLE_RATE,
    .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,
    .channel_format = I2S_CHANNEL_FMT_ONLY_LEFT,
    .communication_format = I2S_COMM_FORMAT_I2S_MSB,
    .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1,
    .dma_buf_count = 4,
    .dma_buf_len = 8,
    .use_apll = false,
    .tx_desc_auto_clear = false,
    .fixed_mclk = 0
   };
   i2s_driver_install(I2S_NUM_0, &i2s_config, 0, NULL);
   i2s_set_adc_mode(ADC_UNIT_1, ADC_INPUT);
   i2s_adc_enable(I2S_NUM_0);
}

void reader(void *pvParameters) {
  uint32_t read_counter = 0;
  uint64_t read_sum = 0;
  while(1){
    size_t bytes_read = 0;
    uint16_t buffer = 0;
    //char i2s_read_buff[I2S_BUFF_LEN];
    i2s_read(I2S_NUM_0, &buffer, sizeof(buffer), &bytes_read, portMAX_DELAY);
    read_sum += 0x4FFF - buffer; // Not sure why this is inverted
    read_counter++;
    if (bytes_read != sizeof(buffer)) Serial.println("buffer empty!");
    if (read_counter == I2S_SAMPLE_RATE) {
      Serial.printf("avg: %d\n", read_sum/I2S_SAMPLE_RATE);
      read_counter = 0;
      read_sum = 0;   
    }
  }
}

void setup() {
  Serial.begin(115200);
  xTaskCreatePinnedToCore(reader, "ADC_reader", 2048, NULL, 1, NULL, 1);
  i2sInit();
  uint32_t freq = ledcSetup(0, 80000, 10);
  ledcWrite(0, 256);
  ledcAttachPin(27, 0);
}

void loop() {}
```

everslick, 2019-01-11:
@folny well meaning advice: read the MARKDOWN documentation to post your code in a readable form. (hint: three ticks).

folny, 2019-01-11:
Yes, I will try to respect it :-)

closed 2019-08-15 by stale[bot]