espressif/arduino-esp32 #2285: Invalid Event when Client connect AP + debug output

luc-github, 2019-01-07:
Using latest Git
Using WiFiAccessPoint sample
Debug is disabled but message appear

When client connect to AP there is an unwished output:
```
E (28866) event: mismatch or invalid event, id=63
E (28867) event: default event handler failed!
dhcps: send_offer>>udp_sendto result 0
```
error messages seems come from : https://github.com/espressif/esp-idf/blob/master/components/esp32/event_default_handlers.c#L457

Debug message is from : https://github.com/espressif/esp-idf/blob/master/components/lwip/apps/dhcpserver/dhcpserver.c#L527
looks like `#if DHCPS_DEBUG` is missing

I know they looks like IDF issue but I only use arduino

me21, 2019-01-14:
I confirm, my Arduino version is 1.0.1.

markingle, 2019-01-14:
I am having the same problem.  I am unable to connect to the esp32 wifi from my Windows laptop and Mac mini.  I just pulled the latest version from the arduino-esp32 repo

me21, 2019-01-14:
This is not the same bug. I am able to connect, but I get error messages from ESP32 as a side effect.

In fact, I wasn't able to connect to ESP32 AP on the previous versions of Arduino framework (was getting authentication errors).

me21, 2019-01-14:
Related bugs in ESP-IDF: https://github.com/espressif/esp-idf/issues/2916, https://github.com/espressif/esp-idf/issues/2915.

They are going to fix those within 2 weeks.

me-no-dev, 2019-01-23:
There already is a "fix" for this in arduino master, but is available to github installs only.

luc-github, 2019-01-23:
@me-no-dev you mean esp-idf github ?

me-no-dev, 2019-01-23:
@luc-github https://github.com/espressif/arduino-esp32/commit/45fa0f829449419aee83d61f967ca32450e63527

luc-github, 2019-01-23:
@me-no-dev  Yes I saw this commit but it is not same issue actually 
the message is raised by `ESP_LOGE(TAG, "mismatch or invalid event, id=%d", event->event_id);`
  https://github.com/espressif/esp-idf/blob/master/components/esp32/event_default_handlers.c#L457

not https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiGeneric.cpp#L342

zhivko, 2019-01-23:
Same problem here with latest esp-idf

```
event: mismatch or invalid event, id=63
```

```
I (13955) wifi: new:<1,1>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
I (13955) wifi: bss=0x3ffbcfac, resp=16 join
I (13965) wifi: station: 78:0c:b8:b3:33:c3 join, AID=1, bgn, 40U
I (13965) wifi: join success, ap send assoc response
W (13975) event: unexpected system event 63!
E (13975) event: mismatch or invalid event, id=63
E (13985) event: default event handler failed!
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40099ad4  PS      : 0x00060330  A0      : 0x800dd0e6  A1      : 0x3ffc0740  
A2      : 0xb33fffff  A3      : 0xb33ffffb  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x3ffc0942  A12     : 0x00000000  A13     : 0x3ffbe27c  
A14     : 0x00000002  A15     : 0x3ffbe920  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xb33fffff  LBEG    : 0x40099af5  LEND    : 0x40099b05  LCOUNT  : 0xfffffffc
```
gives:
```
c:\msys32\home\klemen\esp\DoubleLifter>java -jar EspStackTraceDecoder.jar c:\Users\klemen\Dropbox\Voga\BleVogaLifter-esp32-DRV8703Q\xtensa-lx106-elf-addr2line.exe c:\msys32\home\klemen\esp\DoubleLifter\build\DoubleLifter.elf c:\msys32\home\klemen\esp\DoubleLifter\dump3.txt
Exception Cause: Not found

0x40099ad4: strlen at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/strlen.S:46
0x40099af5: strlen at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/strlen.S:84
0x40099b05: strlen at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/strlen.S:96
0x40099ad4: strlen at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/strlen.S:46
0x400dd0e3: _svfprintf_r at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:1529
0x40168489: _vsnprintf_r at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vsnprintf.c:72
0x401684d6: vsnprintf at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vsnprintf.c:41
0x4010007a: log_printf at C:/msys32/home/klemen/esp/DoubleLifter/components/arduino/cores/esp32/esp32-hal-uart.c:473
0x401040ce: WiFiGenericClass::_eventCallback(void*, system_event_t*) at C:/msys32/home/klemen/esp/DoubleLifter/components/arduino/libraries/WiFi/src/WiFiGeneric.cpp:321
0x401044f2: _network_event_task at C:/msys32/home/klemen/esp/DoubleLifter/components/arduino/libraries/WiFi/src/WiFiGeneric.cpp:321
0x40094725: vPortTaskWrapper at C:/msys32/home/klemen/esp/esp-idf/components/freertos/port.c:435
```

Aimlessisland, 2019-01-26:
Same problem here really annoying
IP address: 192.168.4.1
E (1113) event: mismatch or invalid event, id=63
E (1114) event: default event handler failed!

chegewara, 2019-01-26:
If i remember correct `event: mismatch or invalid event, id=63` is wifi mesh event and in normal wifi connection can be ignored. It was answered on esp-idf issue tracker or on forum, but im not 100% sure.

sbarrufaldi, 2019-01-29:
I have the same problem when I try to connect to AP.

```
E (4302) event: mismatch or invalid event, id=63
E (4303) event: default event handler failed!
dhcps: send_offer>>udp_sendto result 0
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x00000000  PS      : 0x00060e30  A0      : 0x80116eb1  A1      : 0x3ffb3d30  
A2      : 0x3ffc9314  A3      : 0x3ffc98c4  A4      : 0x3ffc8a60  A5      : 0x3ffbb95c  
A6      : 0x0204a8c0  A7      : 0x0c04a8c0  A8      : 0x80116d54  A9      : 0x3ffb3cf0  
A10     : 0x3ffc9324  A11     : 0x3ffc98c4  A12     : 0x3ffb3d3c  A13     : 0x00000044  
A14     : 0x00000001  A15     : 0x00000006  SAR     : 0x00000010  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  

Backtrace: 0x00000000:0x3ffb3d30 0x40116eae:0x3ffb3d70 0x40123055:0x3ffb3d90 0x401269ed:0x3ffb3dd0 0x4012bc22:0x3ffb3df0 0x4011a547:0x3ffb3e10 0x4008e9d1:0x3ffb3e40

Rebooting...
```
It keeps rebooting 4 or 5 times and then it connects perfectly. But for my program are 10 missed seconds...

b1gmans, 2019-02-14:
I also have this issue, it is serious as my project will not run as expected after the error messages are displayed:
E (27188) event: mismatch or invalid event, id=63
E (27189) event: default event handler failed!

Is there a fix for this please?
I have downloaded the latest esp32 system in my arduino ide - still no better? v1.0.1 is what Im running

Is there a way to get latest fixes that may not yet be in this official release?  how? Im using windows 10?

me-no-dev, 2019-02-14:
so, I can not fix the error in IDF, but with the commit I linked above this has been fixed to not cause exception.

b1gmans, 2019-02-14:
I have tried that and it still doesnt work - I have no debug on at all, and still get the error.

luc-github, 2019-02-15:
invalid event is solved with https://github.com/espressif/arduino-esp32/commit/010a7c60f7000eb17c6333309c87156769848a0e

but still Debug output even debug is disabled:
```
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
```

me-no-dev, 2019-02-15:
this is something that comes from IDF :) you should complain there

luc-github, 2019-02-16:
Yes I did , it is in first comment :D   (https://github.com/espressif/esp-idf/issues/2924) need to wait IDF 4.0 
I did not expect this change needed  to wait so long ...

themindfactory, 2019-02-17:
I have the same issue too....

GeorgeFlorian, 2019-02-18:
> invalid event is solved with [010a7c6](https://github.com/espressif/arduino-esp32/commit/010a7c60f7000eb17c6333309c87156769848a0e)
> 
> but still Debug output is present, even debug is disabled:
> 
> ```
> dhcps: send_offer>>udp_sendto result 0
> dhcps: send_offer>>udp_sendto result 0
> dhcps: send_offer>>udp_sendto result 0
> dhcps: send_offer>>udp_sendto result 0
> dhcps: send_offer>>udp_sendto result 0
> ```

How was it solved ? Do we need to reinstall the Arduino ESP32 core ? 
I reinstalled Arduino IDE and ESP32 Core for Arduino IDE last week (probably before the fix) and I still have both of these issues.

me-no-dev, 2019-02-18:
it's not in boards manager yet. it's fixed in master branch here on github. We can not push releases to package manager too often or the json descriptors will become huge. Look at installing Arduino from github instructions :)

GeorgeFlorian, 2019-02-18:
> it's not in boards manager yet. it's fixed in master branch here on github. We can not push releases to package manager too often or the json descriptors will become huge. Look at installing Arduino from github instructions :)

Thank you !

I can confirm that it worked this way. The only *strange* output I get now is this: 
```13:06:38.056 -> dhcps: send_offer>>udp_sendto result 0```

svicar, 2019-02-18:
I deleted ESP32 from board meneger and install from git 2 days ago. The errors are not showing only dhcps: send_offer>>udp_sendto result 0 is hown sometimes...but the problems are not solved. Still dropping wifi connection after few seconds (not always), then problem to get the IP. Maybe is working a little better but the problem are not totaly fixed.

vsn15497, 2019-02-22:
hi @svicar. Can you please provide me the link of the ESP32 in git.

williamesp2015, 2019-02-23:
I have the same problem since few weeks ago using PlatformIO and Arduino Platform using ESP32

Setting soft-AP configuration ... Ready..
Setting soft-AP ... Soft-AP IP address = 192.168.1.11
E (10844) event: mismatch or invalid event, id=63
E (10845) event: default event handler failed!
dhcps: send_offer>>udp_sendto result 0
Please help

svicar, 2019-02-24:
I downloaded git from:
https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md
here are video manual I folowed (but I didnt install python)
https://www.youtube.com/watch?v=4jfb_PL-FEA&t=390s

vsn15497, 2019-02-25:
> I downloaded git from:
> https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md
> here are video manual I folowed (but I didnt install python)
> https://www.youtube.com/watch?v=4jfb_PL-FEA&t=390s

Thank you very much :)

vsn15497, 2019-02-25:
@williamesp2015  I dont think so there is an fix for this issue yet.

williamesp2015, 2019-02-27:
Any help would be appreciated to update these changes in PlatformIo.ini
[env:esp32dev]
platform = espressif32
framework = arduino
board = esp32dev

DeepSkyDad, 2019-03-06:
Hello all,

For me the (temporary) solution was downgrading to espressif32 1.5.0

Best regards,
Pavle

treii28, 2019-03-27:
same problem

luc-github, 2019-04-13:
I cannot reproduce with latest git - thank you

closed 2019-04-13 by luc-github

DarvosBR, 2019-06-19:
Any solution yet? I'm using esp32-arduino version 1.0.2. I though it would be already solved by now!!

anfelbeghoura, 2019-12-10:
hey there;
I have the same problem here; I'm using ESP32, and I am using an android app as a TCPClient ; I think it may have problem with android 9 cause the same app in other phones (android version less than 9)  is working fine ...
So if anyone is working with an android app, could you please mention the android version ..? 
thank you

mike-s123, 2020-04-04:
@luc-github Why is this closed? The problem still exists, Sometimes getting

"dhcps: send_nak>>udp_sendto result 0" 

on serial output with no _#define DHCPS_DEBUG_ present. Using recent git pull of arduino-esp32. SDK version  v3.3.1-61-g367c3c09c Seems to be an issue upstream? (esp-idf,  LWIP)?

    $ grep -r "udp_sendto result" *
    Binary file tools/sdk/lib/liblwip.a matches

christophepersoz, 2020-04-18:
Agree, I also notice this on my dev, sometime the serial print with no apparent reasons.

```
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
```

Swap-File, 2020-05-14:
I may have noticed a related issue.  I See "dhcps: send_offer>>udp_sendto result 0" during AP mode when being a captive portal (DNSServer related UDP traffic), and the ESP32 reboots.

Downgrading to PlatformIO's Espressif IDF 1.9.0 seems to have fixed it, but the problem was intermittent so it's tricky to pin down.

michaelliubetter, 2020-05-31:
i still have the same issue for my esp32 board does anyone found a solution for that,
im trying fix this by using 'config' to auto connect but it still can not be connected every time 
please help

arslan437, 2020-09-02:
I also have same issue. I am trying esp32 Captive portal example but when I try to connect esp AP it reboots continuously.
```
 dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x00000000  PS      : 0x00060e30  A0      : 0x800fdf71  A1      : 0x3ffb4e20  
A2      : 0x3ffcede4  A3      : 0x3ffcf2a8  A4      : 0x3ffcebc8  A5      : 0x3ffceba8  
A6      : 0x020a0a0a  A7      : 0x0c0a0a0a  A8      : 0x800fde14  A9      : 0x3ffb4de0  
A10     : 0x3ffcedf4  A11     : 0x3ffcf2a8  A12     : 0x3ffb4e2c  A13     : 0x00000044  
A14     : 0x00000001  A15     : 0x00000006  SAR     : 0x00000010  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  

Backtrace: 0x00000000:0x3ffb4e20 0x400fdf6e:0x3ffb4e60 0x40109f59:0x3ffb4e80 0x4010d8f1:0x3ffb4ec0 0x40113286:0x3ffb4ee0 0x401013d3:0x3ffb4f00 0x400883cd:0x3ffb4f30
```

Exception decoder results

```
PC: 0x00000000
EXCVADDR: 0x00000000

Decoding stack results
0x400fdf6e: handle_dhcp at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/apps/dhcpserver/dhcpserver.c line 1031
0x40109f59: udp_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/udp.c line 401
0x4010d8f1: ip4_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/ipv4/ip4.c line 740
0x40113286: ethernet_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/netif/ethernet.c line 184
0x401013d3: tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/tcpip.c line 135
0x400883cd: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```

scubachristopher, 2020-09-22:
Same issue of unwanted messages.  SoftAP is up.  When I do NOT connect to it, none of these occur:

`dhcps: send_offer>>udp_sendto result 0`

If I DO hit my SoftAP, these messages seem to correlate to me serving my webpage.

No reboots on my end, but these unsolicited messages are a problem...

pasveer, 2021-11-09:
I still use version 1.0.5 but I like to share my fix anyway.

It's a bit of a hack, but it works like a charm: patch the string in the library file liblwip.a .

In my hex editor I searched for "udp_sendto result" to find the string in the binary. Then I replaced the first byte of the debug message (the letter 'd') by a null-byte, making it zero-length. The message is never seen again ;-)

brownrb, 2021-11-14:
I really cannot ask the users out there to hack the arduino core. And when we update again on the nect release, another patch?
I really should not have to do that. It would be better it the issue was fixed "in the core"

pasveer, 2021-11-14:
I can't agree with you more! 

But if something like this flaw (which exists for ~2 years) is messing up your communication protocol, you sometimes need take drastic/creative measures.

And yes, until this is finally solved in a future release... we patch ;-)

qlalfdu, 2022-01-28:
This message is displayed when a client connects to the same router.

brownrb, 2022-02-01:
Well why is there not a way to turn the message off. 
Who is meant to be in control deciding what to so, the app or the library.
There are much cleaner ways of handling errors.

In my view libraries should NOT under any circumstance be sending messaged out to the serial port.
Typically a library should
1. if there is an error, set an error flag
2. Provide an error routine where the last error and text can be obtained.

I think it is poor design when developers do things like that without thinking of all the downstream
problems that would arise.

Think about it. If developers did that stuff on a mobile phone then it would not last long and people would stop using/buying those phones.

regards
Robert

MikeyMoMo, 2022-05-12:
Getting this today (5/12/2022) on ESP32.  Surely wish someone knew how to make this stop.  Just pops up with no other error printed.

mrengineer7777, 2022-05-13:
> dhcps: send_nak>>udp_sendto result 0
> 
> Getting this today (5/12/2022) on ESP32. Surely wish someone knew how to make this stop. Just pops up with no other error printed.

What version of arduino-esp32 are you using?  Searching through the latest 2.0.3 version, udp_sendto() appears to be in the IDF core.  Searching in IDF core 4.4.0, it seems to come from dhcpserver.c:
```
static void send_nak(struct dhcps_msg *m, u16_t len)
{
...
#if DHCPS_DEBUG
    SendNak_err_t = udp_sendto(pcb_dhcps, p, &ip_temp, DHCPS_CLIENT_PORT);
    DHCPS_LOG("dhcps: send_nak>>udp_sendto result %x\n", SendNak_err_t);
#else
    udp_sendto(pcb_dhcps, p, &ip_temp, DHCPS_CLIENT_PORT);
#endif
...
}
```

It should only print if the arduino lib builder is compiled with DHCPS_DEBUG set.

lrfeliciano, 2022-08-10:
dhcps: send_nak>>udp_sendto result 0

Getting this today (8/11/2022) on ESP32.

VasylVladimirovich, 2022-09-29:
I think, "It's very hard to find a black cat in a dark room, especially if it's not there" ©
You should not show your code that no one wants to understand.
In my opinion the best answer is here:
https://www.anycodings.com/1questions/2079945/esp32-as-wifi-access-point-crashes-on-connection
`IPAddress apIP(192,168,1,1);

WiFi.mode(WIFI_AP);
WiFi.softAP("MyESP32");
delay(2000);
WiFi.softAPConfig(apIP,apIP,IPAddress(255,255,255,0));`

The core panic in the time of client connect can be solved by adding a 2 seconds delay after 'WiFi.softAP' and before 'WiFi.softAPConfig' like this:

more info can be found here: WiFi.softAPIP() causes Core 0 panic'ed (InstrFetchProhibited)

superfrobo, 2022-10-02:
> I think, "It's very hard to find a black cat in a dark room, especially if it's not there" © You should not show your code that no one wants to understand. In my opinion the best answer is here: https://www.anycodings.com/1questions/2079945/esp32-as-wifi-access-point-crashes-on-connection **The core panic in the time of client connect can be solved by adding a 2 seconds delay after 'WiFi.softAP and before'WiFi.softAPConfig** like this:
> 
> `IPAddress apIP(192,168,1,1); WiFi.mode(WIFI_AP); WiFi.softAP("MyESP32"); delay(2000); WiFi.softAPConfig(apIP,apIP,IPAddress(255,255,255,0));` more info can be found here: WiFi.softAPIP() causes Core 0 panic'ed (InstrFetchProhibited)

The problem is not the core panic but the unwanted message that appears even when there is no core panic.

brownrb, 2022-10-02:
Maybe there is confusion ?
We were talking and complaining about the unwanted dhcp error sent to the serial port when esp32 is configured in AP mode.
I did not get any core panics, so your suggestion is of no use to me, but it might be helpful to those that did.
At the very first post the discussion was around the unwanted dhcp error message written to the serial port.

Regards
R