espressif/arduino-esp32 #2286: MbedTLS error -17040 BIGNUM memory allocation failed, and -78 UNKNOWN ERROR CODE (004E)

Adwin-Jason, 2019-01-07:
## Hardware

- Board or module used:  **[ESP32-WROOM-32]**
- IDE: **[ArduinoIDE]**
- Operating System:     **[Windows]**
- Power Supply:         **[USB]**
- IDE Settings:
[image]

## Problem Description

I'm trying to send data to Firebase every 15 seconds through the use of a task set by FreeRTOS. I previously had two tasks running without issue (though not related to Firebase). If I remove the task, it works as expected, though without updated the keepalive value. re-enabling the third task causes the following failures to occur at random:

Error type 1 (90% of the time):
```
[W][HTTPClient.cpp:152] beginInternal(): unexpected protocol: https, expected http
[D][HTTPClient.cpp:181] beginInternal(): host: **<project name redacted>**.firebaseio.com port: 443 url: //status.json?auth=**<auth key redacted>**
[I][ssl_client.cpp:119] start_ssl_client(): WARNING: Use certificates for a more secure communication!
[E][ssl_client.cpp:28] handle_error(): RSA - The public key operation failed : BIGNUM - Memory allocation failed
[E][ssl_client.cpp:30] handle_error(): MbedTLS message code: -17040
[E][WiFiClientSecure.cpp:108] connect(): lwip_connect_r: 11
[D][HTTPClient.cpp:846] connect(): failed connect to **<project name redacted>**.firebaseio.com:443
[W][HTTPClient.cpp:1124] returnError(): error(-1): connection refused
[W][HTTPClient.cpp:1124] returnError(): error(-4): not connected
```
Error type 2 (10% of the time):
```
[W][HTTPClient.cpp:152] beginInternal(): unexpected protocol: https, expected http
[D][HTTPClient.cpp:181] beginInternal(): host: **<project name redacted>**.firebaseio.com port: 443 url: //status.json?auth=**<auth key redacted>**
[I][ssl_client.cpp:119] start_ssl_client(): WARNING: Use certificates for a more secure communication!
[E][ssl_client.cpp:28] handle_error(): UNKNOWN ERROR CODE (004E)
[E][ssl_client.cpp:30] handle_error(): MbedTLS message code: -78
[E][WiFiClientSecure.cpp:108] connect(): lwip_connect_r: 12
[D][HTTPClient.cpp:846] connect(): failed connect to **<project name redacted>**.firebaseio.com:443
[W][HTTPClient.cpp:1124] returnError(): error(-1): connection refused
[W][HTTPClient.cpp:1124] returnError(): error(-4): not connected
```

At first I tested `xTaskCreatePinnedToCore` with an extra-large stack size (just over 16k words) and used `uxTaskGetStackHighWaterMark( NULL );` to determine the exact stack size requirements. I later reduced the stack size to 6200 words.

### Expected Behavior
Firebase Real Time Database updated with keepalive timestamp
### Actual Behavior
Connection refused with errors as mentioned above.

### Steps to repropduce

The code we use is proprietary, so I cannot post all of it, only sections that concern the error. I hope that it can be reproduced with the following:

<steps may require the creation of a free Firebase account. >

1. Create two tasks using freeRTOS `xTaskCreatePinnedToCore`. These two tasks are not continuous loops with the exception of the LED blinking during operation. 
2. Confirm functionality
3. Create third task which will connect to Firebase and update a variable every xx seconds (i chose 15)
4. Operation ceases and error(s) occur.

//No wiring necessary apart from one or two simple LEDs to mimic the blink lights during "operation"

### Code to reproduce this issue
code was longer than 30 lines, so i moved it to GIST.

[GIST link to code sample](https://gist.github.com/Adwin-Jason/fa296e6c2ca78871a35c0c8ebc487dc1)

## Debug Logs

[GIST link to debug log (long)](https://gist.github.com/Adwin-Jason/26a301c4ce70671bbc520eea44c810b3)

Adwin-Jason, 2019-01-07:
Note:
I moved this issue from the IDF repository to here, since I am not working with the IDF on this project.
[https://github.com/espressif/esp-idf/issues/2920 CLOSED](https://github.com/espressif/esp-idf/issues/2920)

Adwin-Jason, 2019-01-08:
##Additional information
Compilation sizes:
````
Sketch uses 838050 bytes (63%) of program storage space. Maximum is 1310720 bytes.
Global variables use 43420 bytes (14%) of dynamic memory, leaving 251492 bytes for local variables. Maximum is 294912 bytes.
````

Jeroen88, 2019-01-08:
> [W][HTTPClient.cpp:152] beginInternal(): unexpected protocol: https, expected http

suggest that you are mixing up http and https. If you add port :443 you should use https (and may ommit the port.

Can you show the line for HTTPClient::begin() call? You may hide the host and auth key, but only these.

Adwin-Jason, 2019-01-11:
> suggest that you are mixing up http and https. If you add port :443 you should use https (and may ommit the port.
>
> Can you show the line for HTTPClient::begin() call? You may hide the host and auth [...]

We are using the default <IOXhop_FirebaseESP32.h> library. I believe that this is what is creating the HTTPClient call. I don't have it called within my program.

Adwin-Jason, 2019-01-15:
> 
> 
> Looking at the source (you can do that too!) at https://github.com/ioxhop/IOXhop_FirebaseESP32/blob/603d0e0ebf027036feb38fc164dd04942fe7a425/IOXhop_FirebaseESP32.cpp#L20 suggest that any "https://" in your url is stripped. So the communication is always http. I suspect you add :443 to the url, which you should't. Also your firebase server should accept plain http requests to make this work. Neither are problems of the ESP32 libraries.

The primary issue, however, isn't the HTTPClient call to http / https (which was primarily a warning not an error, and would work with just the two tasks), but rather the memory allocation. However I did find that the primary issue may be related to the heap memory allocation. I'm still investigating this, but it seems that Firebase might be eating up most of the heap... leaving almost nothing for MbedTLS to run on. My colleague is writing his own light-weight firebase library to test with.

Checking the heap with all 3 tasks, prior to attempting to send the firebase data, it shows as having 48k free heap. If I omit the firebase "keepalive" timestamp update task, the heap shows at nearly 90k free and the operation runs as expected. If I can confirm that this issue is regarding the firebase library, and not an issue of MbedTLS, I will close out the issue.

closed 2019-08-15 by stale[bot]