espressif/arduino-esp32 #2289: Strange loop behaviour on minimal sketch

nikuz, 2019-01-08:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			Mon Jan 7
IDE name:							Arduino IDE 1.8.6
Flash Frequency:					40Mhz
PSRAM enabled:                                          no
Upload Speed:						921600
Computer OS:                                            Mac OSX

### Description:
The below minimal sketch hangs the device at the 4294290 millisecond. I mean serial just stops to print, without any errors, and the last number always 4294290. My DebugLevel set to "error". What the magic?
I tried it many many times. Every time I have to wait more than hour to get this error. But this is stable reproduces on the ff18a211e4638784d7d892e500c9fe67e36eeae7 commit. And it doesn't reproduce on the 1.0.0 release.

Sketch just prints the uptime in milliseconds once in second without interrupts by `delay`.

### Sketch:
```cpp
#include <Arduino.h>

const unsigned long printUptimeTimer = 1L * 1000L;
unsigned long printUptimeLastTime = 0;

void setup() {
    Serial.begin(115200);
    while (!Serial) {
        ;
    }
}

void loop() {
    unsigned long now = millis();
    if (printUptimeTimer < now && now - printUptimeTimer > printUptimeLastTime) {
        Serial.println(millis());
        printUptimeLastTime = millis();
    }
}
```

### Debug Messages:
```
nothing
```

lbernstone, 2019-01-08:
millis is going to roll over to 0 at some point.  You need to have some mechanism to deal with that.

stickbreaker, 2019-01-08:
as @lbernstone says, `millis()` overflows. The current version of `millis()` has a maximum value of ((2^32)-1)/1000, or as you found 4294967. about 71 minutes.  The next version will regain (2^32)-1 or 4294967295ms or ~49days .

Your code does not handle the rollover.
```
   if (printUptimeTimer < now && now - printUptimeTimer > printUptimeLastTime) {
        Serial.println(millis());
        printUptimeLastTime = millis();
    }
// try this
uint32_t now = millis();
static uint32_t lastUpdate = 0;
static uint32_t overFlowCounter = 0;
if(now < lastUpdate){ // overflow
  overFlowCounter++;
  lastUpdate = now;
}
if(now - lastUpdate > updateInterval){
    lastUpdate = now;
   Serial.printf("uptime =%dms",(overFlowCounter*4294968)+now);
}  
```
Chuck.