espressif/arduino-esp32 #2292: ACS712 give strange current value

gabovich, 2019-01-09:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do.
Describe your system( Hardware, computer, O/S, core version, environment)
Describe what is failing
Show the shortest possible code that will duplicate the error
Show the EXACT error message(it doesn't work is not enough)
Then if someone is interested and knowledgeable you might get a answer.  All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:	                                        NodeMCU-32s
Core Installation/update date:	11/jul/2018
IDE name:	                                Arduino IDE 1.8.5  
Flash Frequency:	                        80Mhz
Upload Speed:	                       115200

### Description:
Trying to measure current via ACS712 (30A module) sensor current with esp32. From some reason I get all the time values about ~2750. Try to measure with different resistance and still get the same values.  I know that I need to take into account output sensitivity of the and all, but first I want to see that I get reasonable values that depends on the resistance connected. 
See the code attached.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

const int analogIn = 36;

void setup(){ 
Serial.begin(115200);
pinMode(analogIn, INPUT);
}

void loop(){

  Serial.println(analogRead(analogIn));
 delay(250);
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

lbernstone, 2019-01-09:
Have you put a multimeter on the pin?  2750 => 2750/4096*3.3 = 2216 mV at default settings.

gabovich, 2019-01-11:
2750 is default value i get from analog pin. To centralize it I need to subtract 702 to get 4096/2=2048.
And then I get 2048/4096*5 = 2.5v at default settings.

lbernstone, 2019-01-11:
google "capacitor stabilize analog reading".  ACS7XX meters are inherently noisy. See https://github.com/espressif/arduino-esp32/issues/1804 for calibration.  This forum is for errors in the code base, please use www.esp32.com for general issues.

a-c-sreedhar-reddy, 2019-03-13:
@gabovich  did placing a capacitor solve your issue?

gabovich, 2019-03-14:
@a-c-sreedhar-reddy  no, i still get the same wide range of values.

a-c-sreedhar-reddy, 2019-03-14:
> @a-c-sreedhar-reddy no, i still get the same wide range of values.

are you measuring AC or DC

gabovich, 2019-03-14:
for start i'm trying to measure 0A n

> > @a-c-sreedhar-reddy no, i still get the same wide range of values.
> 
> are you measuring AC or DC

gabovich, 2019-03-14:
> for start i'm trying to measure 0A n
> 
> > > @a-c-sreedhar-reddy no, i still get the same wide range of values.
> > 
> > 
> > are you measuring AC or DC

for start i'm trying to measure 0A current but i'm getting wrong and unstable values.

gabovich, 2019-03-26:
@a-c-sreedhar-reddy The capacitor improved the noise immunity, but the main problem is the non-linearity of the ADC in ESP32. Tried to test it with different resolutions but still I get inaccuracy results.
Is anyone knows how to fix the non-linearity ADC issue in ESP32?

lbernstone, 2019-03-26:
https://github.com/espressif/arduino-esp32/issues/1804
Note, the ACS is also quite non-linear near 0A.  What @a-c-sreedhar-reddy is getting at with his question, is that the value is going to be sinusoidal when measuring AC values.

FrKiko, 2019-04-10:
@gabovich are you being able to connect AC712 sensor (5V needed) with ESP32? How are you doing it?

closed 2019-08-15 by stale[bot]