espressif/arduino-esp32 #2293: Error compilling ESP32\examples\Camera\CameraWebServer

baoshuai9527, 2019-01-10:
hi,Today, using the latest version of git，Error compilling ESP32\examples\Camera\CameraWebServer
Board ：ESP32 Wrover Module
arduino ide 1.8.7

### code
`#define CAMERA_MODEL_WROVER_KIT`

### error compilling 
f:/xxx/arduino-ide/arduino-1.8.7/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:F:\xxx\arduino-ide\arduino-1.8.7\hardware\espressif\esp32/tools/sdk/lib\liblib.a: file format not recognized; treating as linker script

f:/xxx/arduino-ide/arduino-1.8.7/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:F:\xxx\arduino-ide\arduino-1.8.7\hardware\espressif\esp32/tools/sdk/lib\liblib.a:1: syntax error

collect2.exe: error: ld returned 1 exit status

me-no-dev, 2019-01-10:
But I ran this same example. It compiled fine. WTF? The CI also had no issues building this example (apart from needing to select the proper partition scheme).

baoshuai9527, 2019-01-10:
@me-no-dev  git checkout 1.0.1-rc3 ,compile any example error

me-no-dev, 2019-01-10:
can you try to delete all and reinstall esp32-arduino and all of it's tools (run get.exe in tools)? Maybe some files did not get pulled properly... no clue really. The lib that gives the error is not build at Arduino prep time. It comes directly from ESP-WHO.

baoshuai9527, 2019-01-10:
yes，
### LOG
```
F:\bao\arduino-ide\arduino-1.8.7\hardware\espressif
$ rm -rf esp32\

F:\bao\arduino-ide\arduino-1.8.7\hardware\espressif
$ ls
esp8266/

F:\bao\arduino-ide\arduino-1.8.7\hardware\espressif
$ git clone --recursive https://github.com/espressif/arduino-esp32.git esp32
Cloning into 'esp32'...
remote: Enumerating objects: 9, done.
remote: Counting objects: 100% (9/9), done.
remote: Compressing objects: 100% (7/7), done.
remote: Total 11464 (delta 0), reused 3 (delta 0), pack-reused 11455
Receiving objects: 100% (11464/11464), 161.12 MiB | 4.46 MiB/s, done.
Resolving deltas: 100% (6779/6779), done.
Checking out files: 100% (1916/1916), done.
Submodule 'libraries/AzureIoT' (https://github.com/VSChina/ESP32_AzureIoT_Arduino) registered for path 'libraries/AzureIoT'
Submodule 'libraries/BLE' (https://github.com/nkolban/ESP32_BLE_Arduino.git) registered for path 'libraries/BLE'
Cloning into 'F:/bao/arduino-ide/arduino-1.8.7/hardware/espressif/esp32/libraries/AzureIoT'...
remote: Enumerating objects: 201, done.
remote: Total 201 (delta 0), reused 0 (delta 0), pack-reused 201
Receiving objects: 100% (201/201), 314.48 KiB | 310.00 KiB/s, done.
Resolving deltas: 100% (34/34), done.
Cloning into 'F:/bao/arduino-ide/arduino-1.8.7/hardware/espressif/esp32/libraries/BLE'...
remote: Enumerating objects: 431, done.
remote: Total 431 (delta 0), reused 0 (delta 0), pack-reused 431
Receiving objects: 100% (431/431), 225.58 KiB | 307.00 KiB/s, done.
Resolving deltas: 100% (265/265), done.
Submodule path 'libraries/AzureIoT': checked out '67dfa4f31ef88b0938dd87d955612100dea5562e'
Submodule path 'libraries/BLE': checked out 'b232e7f5f0e87f36afbc2f4e03a2c49c48dd47bc'

F:\bao\arduino-ide\arduino-1.8.7\hardware\espressif
$ cd esp32\tools\

F:\bao\arduino-ide\arduino-1.8.7\hardware\espressif\esp32\tools (master -> origin)
$ python get.py
System: Windows, Info: Windows-7-6.1.7601-SP1
Platform: i686-mingw32
Downloading xtensa-esp32-elf-win32-1.22.0-80-g6c4433a-5.2.0.zip
Done
Extracting xtensa-esp32-elf-win32-1.22.0-80-g6c4433a-5.2.0.zip
Downloading esptool-2.6.0-windows.zip
Done
Extracting esptool-2.6.0-windows.zip
Downloading mkspiffs-0.2.3-arduino-esp32-win32.zip
Done
Extracting mkspiffs-0.2.3-arduino-esp32-win32.zip
Renaming mkspiffs-0.2.3-arduino-esp32-win32/ to mkspiffs
Done

F:\bao\arduino-ide\arduino-1.8.7\hardware\espressif\esp32\tools (master -> origin)
$ ls
build-release.sh*  build.py*  check_cmakelists.sh*  deploy.sh*  espota.exe*  esptool/     gen_esp32part.exe*  get.exe*  mkspiffs/    platformio-build.py  xtensa-esp32-elf/
build-tests.sh*    build.sh*  common.sh*            dist/       espota.py*   esptool.py*  gen_esp32part.py*   get.py*   partitions/  sdk/
```
### arduino ide
open arduino ide 1.8.7，Select board ESP32 Wrover Module，Partition Scheme default，minimal ，no ota，minimal spiffs ，I've tried all，But still failed to compile
### arduino ide log
```
f:/bao/arduino-ide/arduino-1.8.7/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:F:\bao\arduino-ide\arduino-1.8.7\hardware\espressif\esp32/tools/sdk/lib\liblib.a: file format not recognized; treating as linker script

f:/bao/arduino-ide/arduino-1.8.7/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:F:\bao\arduino-ide\arduino-1.8.7\hardware\espressif\esp32/tools/sdk/lib\liblib.a:1: syntax error

collect2.exe: error: ld returned 1 exit status
```

me-no-dev, 2019-01-10:
OK, I have one idea :)
Edit this line: https://github.com/espressif/arduino-esp32/blob/master/platform.txt#L38
And remove `-llib` then try to compile again :)

baoshuai9527, 2019-01-10:
> OK, I have one idea :)
> Edit this line: https://github.com/espressif/arduino-esp32/blob/master/platform.txt#L38
> And remove `-llib` then try to compile again :)

@me-no-dev Thank you, sir.It compiled fine

me-no-dev, 2019-01-10:
Awesome! I'm pushing the update and new RC now :)

baoshuai9527, 2019-01-10:
> [b0582e1](https://github.com/espressif/arduino-esp32/commit/b0582e1ec885a595c6191b026c088e7f28a52bfd)
> Also adding a new partition scheme for Wrover that you should use with the Camera demo :) (Huge App)

Awesome!6666666

me-no-dev, 2019-01-10:
rc4 is released :)

closed 2019-01-10 by me-no-dev