espressif/arduino-esp32 #2297: Compiling of mbedtls failes with '#include expects "FILENAME" or <FILENAME>'

plcengineer, 2019-01-10:
**In platform.txt must be fixed on line 25, from:**

"compiler.cpreprocessor.flags=-DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -[..]"

**into:**

"compiler.cpreprocessor.flags=-DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="<mbedtls/esp_config.h>" -[..]"

me-no-dev, 2019-01-10:
Can you please not skip the proper form to fill an issue? OS, Arduino version, etc... how should we know all of that? Minimal sketch to reproduce?

closed 2019-01-10 by me-no-dev

plcengineer, 2019-01-10:
Hello, it is an issue in the preprocessor command line argument.

- Platform is: ESP32, but that is the common context here. So this information would be redundant.
- OS is not important (my system is Win10 x86_64)
**- The misstake is shown above in the issue description: "[..] -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" [..]" is wrong and must be changed into: "[..] -DMBEDTLS_CONFIG_FILE="<mbedtls/esp_config.h" [..]>" in the file "platform.txt" at the root of this repository.**

A sample sketch would be:

```
#include <WiFi.h>
#include <WiFiClientSecure.h>

const char ssid[] = "ssid";
const char pass[] = "pass";

WiFiClientSecure net;

unsigned long lastMillis = 0;

void setup() {

}

void loop() {

}
```

**results in:**

"
c:\bin\sloeber\arduinoplugin\packages\esp32\hardware\esp32\1.0.0\tools\sdk\include\mbedtls\mbedtls\platform.h:39:10: error: #include expects "FILENAME" or <FILENAME>
 #include MBEDTLS_CONFIG_FILE
          ^
"

BR,
Mathias

plcengineer, 2019-01-10:
@me-no-dev, please reopen it. I hope there are enought information now.

me-no-dev, 2019-01-10:
It is definitely not enough :) what version of this esp32-arduino core you are running? And OS does matter, since we compile those fine on Linux and Mac for sure.
Just to chime in some more on this issue... many people use WiFiClientSecure and nobody has reported this issue since we last "fixed" the define. I just compiled your exact code also and have no problem at all.

mouridis, 2019-01-10:
[image]

Sample sketch compiles fine in Win10 x86_64

mouridis, 2019-01-10:
Oh... I just saw "c:\bin\sloeber\" on your error... You are using Eclipse with Arduino plugin? I would open an issue there...

plcengineer, 2019-01-16:
Ok, thanks for your great work :-) I will set a user specific compile line in my eclipse project to fix the issue for now.. And I will also try to work with sloeber under linux. Working with Arduino IDE isn't the right way for me. Eclipse has better smart coding functionality.

me-no-dev, 2019-01-16:
Agreed! :)

mouridis, 2019-02-02:
@plcengineer May I suggest VSCode with Arduino Plugin? You get rid of Arduino IDE and it's so much better than Sloeber IMHO.

arnold-b, 2019-04-27:
I´m using VSCode and have the same problem. but the solution don´t help