espressif/arduino-esp32 #2303: Missing TTGO boards

HakanP, 2019-01-10:
Missing board, possible to get the TTGO Lora32 V2.1 PCB V1.6 added?
https://www.aliexpress.com/item/TTGO-LoRa32-V2-1-1-6-version-433-868-915Mhz-ESP32-LoRa-OLED-0-96-Inch/32872078587.html

Also the model without OLED would be nice: https://www.aliexpress.com/item/TTGO-2Pcs-SX1276-LoRa-ESP32-868-915MHz-Bluetooth-WI-FI-Internet-Antenna-Development-Board-for-Arduino/32848733448.html

Jeroen88, 2019-01-11:
For the model _without_ OLED I am almost shure that you can use the TTGO LoRa32-OLED V1 board, just do not use the OLED.

nae9, 2019-01-11:
Board spec in arduino sets pinouts and another chip config. If you boards use same chip, then you can use this preset, just reassign pins in manner SDA = 4 SCL = 5, or simple input pins as numbers.
OLED - it is i2c slave - use this pin as I2C pins. If you whant - dont send i2c data for OLED to I2C bus ;)

closed 2019-08-15 by stale[bot]