espressif/arduino-esp32 #2304: After update Arduino-esp32 to 101 error 	esp_restart_noos();

SailorJos, 2019-01-11:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			updated form 1.0.0 to 1.0.1
IDE name:						Arduino IDE
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    no
Upload Speed:					NA
Computer OS:                                        Windows 10

### Description:
After updating from version Arduino-ESP32 Espressif Systems 1.0.0 to 1.0.1 I get an error at: 	esp_restart_noos(); when compiling. 
After reinstalling version 1.0.0 again program compiles without error. 
Is there an error in version 1.0.1 or an error in my program which was not encounterd by old version ?

-------------------------------
#include <Arduino.h>

void setup() {
.....
watchdogTimer = timerBegin(0, 80, true); // timer0, divisor 80
timerAlarmWrite(watchdogTimer, 4000000, false); //set time in uSec. must be fed within this time else reboot. 4 seconds
timerAttachInterrupt(watchdogTimer, &interruptReboot, true);
timerAlarmEnable(watchdogTimer); //enable interrupt
....
}

void loop() {
```
timerWrite(watchdogTimer, 0); // reset timer (feed watchdog)
.....
}

void interruptReboot(){
	esp_restart_noos();
}

-------------------------------

### Debug Messages:

In function 'void interruptReboot()':

RE_UDP_BROADCASTINGvvvv:328:19: error: 'esp_restart_noos' was not declared in this scope

  esp_restart_noos();

                   ^

Thanks for your advice

me-no-dev, 2019-01-11:
yes, this was old api that got deprecated and removed. Why do you not use the Arduino API for reboot though? `ESP.restart();` or `esp_restart();`

closed 2019-01-11 by me-no-dev

SailorJos, 2019-01-11:
This solved the problem. I copied an example from somewhere on the internet. Apparently an old example.

@mee-no-dev: thanks!!

Netoperz, 2019-02-18:
i have faced the same problem, the example should be updated. thx me-no-dev fo info

UCIS, 2020-04-09:
> yes, this was old api that got deprecated and removed. Why do you not use the Arduino API for reboot though? `ESP.restart();` or `esp_restart();`

The `ESP.restart()` / `esp_restart()` functions call some clean-up code, which may not run correctly from an ISR context. I want to reboot in an ISR to provide a custom reset button, which should work even if the `loop()` function is not functioning. Now the chip does reboot via `ESP.restart()`, but it does so because of a panic, not because of the reboot command.