espressif/arduino-esp32 #2305: watchdog not work...

spa-sam, 2019-01-11:
### Hardware:
Board:						      ESP32 Dev Module
Core Installation version:		      1.0.1
IDE name:					       Arduino IDE
Flash Frequency:				       80Mhz
PSRAM enabled:                                    no
Upload Speed:				       115200
Computer OS:                                        Windows 10

### Description:
I use the standard watchdogtimer.ino example. My project works from the power supply and from the battery as a reserve. In the event of a power failure, everything continues to work perfectly on the battery. But as soon as it goes back to power from the power supply unit, esp32 freezes, and the serial port removes garbage. watchdog does not work. Only the reset button helps.

### Sketch: 
```cpp
#include "esp_system.h"

const int button = 21;         //gpio to use to trigger delay
const int wdtTimeout = 3000;  //time in ms to trigger the watchdog
hw_timer_t *timer = NULL;

void IRAM_ATTR resetModule() {
  ets_printf("reboot\n");
  esp_restart();
}

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println("running setup");

  pinMode(button, INPUT_PULLUP);                    //init control pin
  timer = timerBegin(0, 80, true);                  //timer 0, div 80
  timerAttachInterrupt(timer, &resetModule, true);  //attach callback
  timerAlarmWrite(timer, wdtTimeout * 1000, false); //set time in us
  timerAlarmEnable(timer);                          //enable interrupt
}

void loop() {
  Serial.println("running main loop");

  timerWrite(timer, 0); //reset timer (feed watchdog)
  long loopTime = millis();
  //while button is pressed, delay up to 3 seconds to trigger the timer
  while (!digitalRead(button)) {
    Serial.println("button pressed");
    delay(500);
  }
  delay(1000); //simulate work
  loopTime = millis() - loopTime;
  
  Serial.print("loop time is = ");
  Serial.println(loopTime); //should be under 3000
}
```

### Debug Messages:
```
~rn��n���nln|�ll�l~rn��n���nln|�l��l~rn��n���nln|�l�l~rn��n���nln|�ll`�n�n����n�np|�b���쒜��|�r�nnprrnbl�l���|�bl�lb�����ll�lll�����l��n�n����n�np|�b�l���쒜��|�r�nnprrnbl�
```

spa-sam, 2019-01-11:
I noticed one thing: if you select the ESP32 Dev Module, then everything works fine. With what it can be connected?

me-no-dev, 2019-01-11:
This is  rather strange issue! ESP32 dev module should not behave in any way different than other configs (except maybe flash size and speed). Do you have a scope to look at the power to the esp when switching the source? And another note: Arduino now supports the RTOS WDT and you can enable it for the whole core1 or just the loop task (like in your case). API is here: https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal.h#L78

spa-sam, 2019-01-11:
How will the latest changes look like the example above? Is it possible to add an example to a repository?

me-no-dev, 2019-01-11:
```cpp
const int button = 21;         //gpio to use to trigger delay

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println("running setup");
  enableLoopWDT();
  pinMode(button, INPUT_PULLUP);                    //init control pin
}

void loop() {
  Serial.println("running main loop");

  long loopTime = millis();
  //while button is pressed, delay up to 5 seconds to trigger the timer
  while (!digitalRead(button)) {
    Serial.println("button pressed");
    delay(500);
  }
  delay(1000); //simulate work
  loopTime = millis() - loopTime;
  
  Serial.print("loop time is = ");
  Serial.println(loopTime); //should be under 5000
}
```

me-no-dev, 2019-01-12:
the time depends on the IDF configuration. For Arduino it uses the default value. You can use Arduino as IDF component if you want to play with that time. 
To feed the WDT you can call `esp_task_wdt_reset();` after including `esp_task_wdt.h`. I will add api also

me-no-dev, 2019-01-12:
you can see the changes here: https://github.com/espressif/arduino-esp32/commit/81844f53600c3035fda6b9affb79c19573ee8f1d

spa-sam, 2019-01-13:
Thank you very much for the work done!!!

ArTyGill, 2019-01-22:
@me-no-dev  Сторожевой таймер, работает отлично в моих тестах. Но у меня происходит такая проблема что контроллер зависает без причины, и не один сторожевой таймер так и не срабатывает.

ArTyGill, 2019-01-23:
@me-no-dev It happens that the controller may hang after a week, for no reason at all, the power is stable, the watchdog is set up.

me-no-dev, 2019-01-23:
@ArTyGill even though I understood everything in Russian (I am Bulgarian and not Russian :) ) I am not sure how to help. If WDT is not triggered, there is no way to reset the CPU

ArTyGill, 2019-01-25:
@me-no-dev oh sorry, I forgot to translate) I solved the problem with a " crutch" - I rolled back to the first stable version v1.0.0 and everything works well, but I want to understand why wdt does not work.
On the v1.0.0 often repeat error assert param (0, 32), on release 1.0.1 freezes.  esp32

closed 2019-08-15 by stale[bot]