espressif/arduino-esp32 #2308: BLE Error  assertion "heap != NULL && "free() target pointer is outside heap areas""

czuvich, 2019-01-11:
Board: ESP32 WROVER Dev Kit 4.1
Arduino ESP32 Core: 1.0.1
PSRAM Enabled : Yes

I'm getting the following exception in my BLE code in the new build. The 1.0.0 build did not have this issue.  

### Debug Messages:
```
assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_caps.c", line 268, function: heap_caps_free

```

I'm running a BLEClient that is connecting to another ESP32.  I'm reading the characteristics, and that's where it seems to crash.  I've included Verbose output messages and backtrace.  For now, I'm going to revert back to 1.0.0.  Let me know if you want to see code; however, this code worked fine in 1.0.0 (it's pretty straight forward BLE interfacing).

### Backtrace:
```
0x4009520c: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
0x4009543d: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
0x4016e62b: __assert_func at ../../../.././newlib/libc/stdlib/assert.c line 63
0x40087dfb: heap_caps_free at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_caps.c line 268
0x4008e77d: _free_r at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/syscalls.c line 42
0x400e677a: BLERemoteCharacteristic::gattClientEventHandler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at /Users/Clay/Library/Arduino15/packages/esp32/hardware/esp32/1.0.1/libraries/BLE/src/BLERemoteCharacteristic.cpp line 192
0x400e7261: BLERemoteService::gattClientEventHandler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at /Users/Clay/Library/Arduino15/packages/esp32/hardware/esp32/1.0.1/libraries/BLE/src/BLERemoteService.cpp line 120
0x400e54c9: BLEClient::gattClientEventHandler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at /Users/Clay/Library/Arduino15/packages/esp32/hardware/esp32/1.0.1/libraries/BLE/src/BLEClient.cpp line 314
0x400e62b9: BLEDevice::gattClientEventHandler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at /Users/Clay/Library/Arduino15/packages/esp32/hardware/esp32/1.0.1/libraries/BLE/src/BLEDevice.cpp line 173
0x401045d9: btc_gattc_cb_handler at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/btc/profile/std/gatt/btc_gattc.c line 31
0x4010057a: btc_task at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/btc/core/btc_task.c line 110
0x400980cd: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

```

### Debug Messages:
```

12:04:49.173 -> [D][FreeRTOS.cpp:165] take(): Semaphore taking: name: ReadCharEvt (0x3ffe1568), owner: <N/A> for readValue
12:04:49.173 -> [D][FreeRTOS.cpp:174] take(): Semaphore taken:  name: ReadCharEvt (0x3ffe1568), owner: readValue
12:04:49.207 -> [V][FreeRTOS.cpp:70] wait(): >> wait: Semaphore waiting: name: ReadCharEvt (0x3ffe1568), owner: readValue for readValue
12:04:49.207 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 3
12:04:49.207 -> [D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
12:04:49.207 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 3
12:04:49.207 -> [V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
12:04:49.241 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 3
12:04:49.241 -> [D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
12:04:49.241 -> assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_caps.c", line 268, function: heap_caps_free
12:04:49.275 -> abort() was called at PC 0x4016e62b on core 0

```

czuvich, 2019-01-11:
I changed verbosity to INFO, and here's the output.  I noticed the exception changed.  I may have been getting heap issues earlier due to the huge amount of logs being generated in Verbose.  Either way, it looks like an issue in the BLERemoveService.cpp.

### Debug Messages
```
12:19:28.466 -> [E][BLERemoteCharacteristic.cpp:280] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
12:19:28.466 -> [E][BLERemoteCharacteristic.cpp:280] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
12:19:28.466 -> [E][BLERemoteCharacteristic.cpp:280] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
12:19:28.466 -> [E][BLERemoteCharacteristic.cpp:280] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
12:19:28.569 -> 
12:19:28.569 -> Stack smashing protect failure!
12:19:28.569 -> 
12:19:28.569 -> abort() was called at PC 0x40173a57 on core 1

```

### Backtrace
```
0x400951e0: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
0x40095411: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
0x40173a57: __stack_chk_fail at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/stack_check.c line 36
0x400e5f3e: BLERemoteService::retrieveCharacteristics() at /Users/Clay/Library/Arduino15/packages/esp32/hardware/esp32/1.0.1/libraries/BLE/src/BLERemoteService.cpp line 216
0x400e5f7e: BLERemoteService::getCharacteristic(BLEUUID) at /Users/Clay/Library/Arduino15/packages/esp32/hardware/esp32/1.0.1/libraries/BLE/src/BLERemoteService.cpp line 149
0x400d4b33: BluetoothDevice::executeReadCommand() at /var/folders/_d/ksp5fdyx5h1flr8s6l6rvgj00000gn/T/arduino_build_800126/sketch/BluetoothDevice.cpp line 202

```

wakwak-koba, 2019-01-14:
I found some bugs in 1.0.1, and fixed it.
https://github.com/nkolban/ESP32_BLE_Arduino/compare/master...wakwak-koba:master

me-no-dev, 2019-01-14:
@chegewara this is for you buddy :)

czuvich, 2019-01-15:
Any takers?  I'm still seeing some issues.

wakwak-koba, 2019-01-16:
Would you please try using mine for testing

elotn, 2019-01-16:
I have the same problem. I tested with https://github.com/wakwak-koba/ESP32_BLE_Arduino and 1.0.1
I get the same error:
**DEBUG  Info:**
```
Starting Arduino BLE Client application...
BLE Loop Task created
 - Created client
HEAP SIZE LEFT =118008
HEAP SIZE LEFT =118008
Forming a connection to 24:0a:c4:09:7d:0a
[I][BLEDevice.cpp:613] addPeerDevice(): add conn_id: 0, GATT role: client
 - Connected to server
HEAP SIZE LEFT =113752
 - Found our service
 - Found our characteristic
assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_caps.c", line 268, function: heap_caps_free
abort() was called at PC 0x4014b643 on core 0

Backtrace: 0x4008d928:0x3ffd28d0 0x4008db59:0x3ffd28f0 0x4014b643:0x3ffd2910 0x4008475f:0x3ffd2940 0x4008a2a9:0x3ffd2960 0x4000bec7:0x3ffd2980 0x400d4e62:0x3ffd29a0 0x400d5729:0x3ffd29e0 0x400d40f1:0x3ffd2a00 0x400d4cdd:0x3ffd2ab0 0x400e410d:0x3ffd2b00 0x400e01ce:0x3ffd2b50 0x4008fd31:0x3ffd2b80
```
**DEBUG  Verbose:**
```
Starting Arduino BLE Client application...

BLE Loop Task created
[D][BLEDevice.cpp:66] createClient(): >> createClient
[D][BLEDevice.cpp:72] createClient(): << createClient
 - Created client
HEAP SIZE LEFT =118000
HEAP SIZE LEFT =118000
Forming a connection to 24:0a:c4:09:7d:0a
[D][BLEClient.cpp:103] connect(): >> connect(24:0a:c4:09:7d:0a)
[I][BLEDevice.cpp:613] addPeerDevice(): add conn_id: 0, GATT role: client
[D][FreeRTOS.cpp:165] take(): Semaphore taking: name: RegEvt (0x3ffdff38), owner: <N/A> for connect
[D][FreeRTOS.cpp:174] take(): Semaphore taken:  name: RegEvt (0x3ffdff38), owner: connect
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_REG_EVT
[V][FreeRTOS.cpp:70] wait(): >> wait: Semaphore waiting: name: RegEvt (0x3ffdff38), owner: connect for connect
[V][BLEUtils.cpp:1272] dumpGattClientEvent(): GATT Event: ESP_GATTC_REG_EVT
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_REG_EVT
[V][FreeRTOS.cpp:120] give(): Semaphore giving: name: RegEvt (0x3ffdff38), owner: connect
[V][FreeRTOS.cpp:86] wait(): << wait: Semaphore released: name: RegEvt (0x3ffdff38), owner: connect
[D][FreeRTOS.cpp:165] take(): Semaphore taking: name: OpenEvt (0x3ffe21ac), owner: <N/A> for connect
[D][FreeRTOS.cpp:174] take(): Semaphore taken:  name: OpenEvt (0x3ffe21ac), owner: connect
[V][FreeRTOS.cpp:70] wait(): >> wait: Semaphore waiting: name: OpenEvt (0x3ffe21ac), owner: connect for connect
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_CONNECT_EVT
[V][BLEUtils.cpp:1272] dumpGattClientEvent(): GATT Event: ESP_GATTC_CONNECT_EVT
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_CONNECT_EVT
[D][BLEDevice.cpp:597] updatePeerDevice(): update conn_id: 4, GATT role: client
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_OPEN_EVT
[V][BLEUtils.cpp:1272] dumpGattClientEvent(): GATT Event: ESP_GATTC_OPEN_EVT
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_OPEN_EVT
[V][FreeRTOS.cpp:120] give(): Semaphore giving: name: OpenEvt (0x3ffe21ac), owner: connect
[V][FreeRTOS.cpp:86] wait(): << wait: Semaphore released: name: OpenEvt (0x3ffe21ac), owner: connect
[D][BLEClient.cpp:136] connect(): << connect(), rc=1
[D][BLEClient.cpp:383] getService(): >> getService: uuid: 4fafc201-1fb5-459e-8fcc-c5c9c331914b
[D][BLEClient.cpp:419] getServices(): >> getServices
[D][BLEClient.cpp:78] clearServices(): >> clearServices
[D][BLEClient.cpp:85] clearServices(): << clearServices
[D][FreeRTOS.cpp:165] take(): Semaphore taking: name: SearchCmplEvt (0x3ffe220c), owner: <N/A> for getServices
[D][FreeRTOS.cpp:174] take(): Semaphore taken:  name: SearchCmplEvt (0x3ffe220c), owner: getServices
[V][FreeRTOS.cpp:70] wait(): >> wait: Semaphore waiting: name: SearchCmplEvt (0x3ffe220c), owner: getServices for getServices
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_SEARCH_RES_EVT
[V][BLEUtils.cpp:1272] dumpGattClientEvent(): GATT Event: ESP_GATTC_SEARCH_RES_EVT
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_SEARCH_RES_EVT
[D][BLERemoteService.cpp:32] BLERemoteService(): >> BLERemoteService()
[D][BLERemoteService.cpp:40] BLERemoteService(): << BLERemoteService()
HEAP SIZE LEFT =110364
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_SEARCH_RES_EVT
[V][BLEUtils.cpp:1272] dumpGattClientEvent(): GATT Event: ESP_GATTC_SEARCH_RES_EVT
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_SEARCH_RES_EVT
[... 31 lines cut ...]
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_READ_CHAR_EVT
assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_caps.c", line 268, function: heap_caps_free
abort() was called at PC 0x4014d1b7 on core 0

Backtrace: 0x4008d928:0x3ffd23f0 0x4008db59:0x3ffd2410 0x4014d1b7:0x3ffd2430 0x4008475f:0x3ffd2460 0x4008a2a9:0x3ffd2480 0x4000bec7:0x3ffd24a0 0x400d5f3e:0x3ffd24c0 0x400d6825:0x3ffd2500 0x400d4cf5:0x3ffd2520 [... cut ...]
0x400e410d: btc_gattc_cb_to_app at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/btc/profile/std/gatt/btc_gattc.c line 31
:  (inlined by) btc_gattc_cb_handler at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/btc/profile/std/gatt/btc_gattc.c line 795
0x400e01ce: btc_task at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/btc/core/btc_task.c line 110
0x4008fd31: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 355 (discriminator 1)
```

elotn, 2019-01-16:
Hi @chegewara 
Thanks. For now it looks like it helped. After this change I can't replicate this problem. I tested with modified https://github.com/nkolban/ESP32_BLE_Arduino/commit/b232e7f5f0e87f36afbc2f4e03a2c49c48dd47bc

czuvich, 2019-01-16:
So I tried a few variations, and I'm seeing the same error.  I tried the following:

1) @chegewara suggestion on commenting out a few lines in `BLERemoteCharacteristic`
2) @wakwak-koba branch
3) @wakwak-koba branch with @chegewara suggestion

I also added an esp_log to see if `ESP_GATTC_READ_CHAR_EVT` was even being called in the commented out section; however, I'm not seeing that log.  So, the error is occurring elsewhere.  I've attached my backtrace.

```

Decoding stack results
0x400951e0: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
0x40095411: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
0x4016c407: __assert_func at ../../../.././newlib/libc/stdlib/assert.c line 63
0x40087dcf: heap_caps_free at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_caps.c line 268
0x4008e751: _free_r at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/syscalls.c line 42
0x400e53f2: BLERemoteCharacteristic::gattClientEventHandler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at /Users/Clay/Library/Arduino15/packages/esp32/hardware/esp32/1.0.1/libraries/BLE/src/BLERemoteCharacteristic.cpp line 192
0x400e5e59: BLERemoteService::gattClientEventHandler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at /Users/Clay/Library/Arduino15/packages/esp32/hardware/esp32/1.0.1/libraries/BLE/src/BLERemoteService.cpp line 120
0x400e45dd: BLEClient::gattClientEventHandler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at /Users/Clay/Library/Arduino15/packages/esp32/hardware/esp32/1.0.1/libraries/BLE/src/BLEClient.cpp line 314
0x400e525d: BLEDevice::gattClientEventHandler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at /Users/Clay/Library/Arduino15/packages/esp32/hardware/esp32/1.0.1/libraries/BLE/src/BLEDevice.cpp line 187
0x401023b9: btc_gattc_cb_handler at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/btc/profile/std/gatt/btc_gattc.c line 31
0x400fe35a: btc_task at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/btc/core/btc_task.c line 110
0x400980a1: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

```

czuvich, 2019-01-16:
Oh wait.. my apologies.  I had to turn down my logs.  So... it looks like commenting out those lines worked.  I am also using @wakwak-koba branch as well.  Having said that, I tried testing my code with 2 BLE Servers (both ESP32), and I am now getting a different exception.  I'll create another issue for it.

chegewara, 2019-01-16:
Hi,
your stack trace suggests the same problem, with free() in `BLERemoteCharacteristic::gattClientEventHandler`. There is only 2 calls to free() in BLERemoteCharacteristic and one is in destructor. 

EDIT thanks

czuvich, 2019-01-16:
Thanks for the help!  I'll leave the ticket open until a fix is released.  I also opened up a different issue for the other problem I'm seeing with 2 BLE servers and 1 client.

ayandas4, 2019-03-20:
I'm facing the same issue in Arduino. Writing an Eddystone beacon, along with Wi-Fi, EEPROM and few GPIO activities. After I added BLEDevice, I faced Sketch too big issue, and after I changed the default partition to increase the program memory area. I also tried selecting partition scheme "No Ota (Large APP)", but the issue remains same.
BLE initialization code:

`/*****************************************************************************
Initialize Bluetooth Low energy for Eddystone
*****************************************************************************/
void BLE_Init() {
  char beacon_data[22];
  uint16_t beaconUUID = 0xFFAA;   // UUID for Eddystone Service
  int url_length;
  int count;
  
  // Create BLE device
  BLEDevice::init("UniShelf");

  // Create BLE Server
  BLEServer *pServer = BLEDevice::createServer();
  
  pAdvertising = pServer->getAdvertising();

  //setBeacon();
  BLEAdvertisementData oAdvertisementData = BLEAdvertisementData();
  oAdvertisementData.setFlags(0x06);    // GENERAL_DISK_MODE 0x02 | BR_EDR_NOT_SUPPORTED 0x04
  oAdvertisementData.setCompleteServices(BLEUUID(beaconUUID));

  //beacon_data[0] = 0x20;    // Eddystone Frame Type (Unencrypted Eddystone - TLM)
  beacon_data[0] = 0x10;      // Eddystone Frame Type (URL)
  beacon_data[1] = 0x06;      // 
  beacon_data[2] = 0x03;      // Length
  beacon_data[3] = 0x03;      // Flag - Complete list of 16-bit Service UUIDs data type value
  beacon_data[4] = 0xAA;      // 16bit Eddystone UUID
  beacon_data[5] = 0xFE;      // ...

  url_length = product_url.length();
  beacon_data[6] = url_length+3;      // Length
  beacon_data[7] = 0x10;      // Frame Type - URL
  beacon_data[8] = 0x04;      // Tx power 4dBm?
  beacon_data[9] = 0x03;      // URL Scheme Prefix - https://
  for(count=0; count<url_length; count++) {
    beacon_data[9+count] = product_url.charAt(count);
  }

  oAdvertisementData.setServiceData(BLEUUID(beaconUUID), std::string(beacon_data, url_length+10));

  pAdvertising->setScanResponseData(oAdvertisementData);

  // Start advertising
  pAdvertising->start();
#ifdef DEBUG
  Serial.println("Advertising started...");
#endif
}`

Here is the backtrace with core debug level "verbose" 

`[D][BLEDevice.cpp:82] createServer(): >> createServer
[D][BLEServer.cpp:290] registerApp(): >> registerApp - 0
[D][FreeRTOS.cpp:165] take(): Semaphore taking: name: RegisterAppEvt (0x3ffe461c), owner: <N/A> for registerApp
[D][FreeRTOS.cpp:174] take(): Semaphore taken:  name: RegisterAppEvt (0x3ffe461c), owner: registerApp
[V][FreeRTOS.cpp:70] wait(): >> wait: Semaphore waiting: name: RegisterAppEvt (0x3ffe461c), owner: registerApp for registerApp
[D][BLEDevice.cpp:108] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_REG_EVT
[V][BLEUtils.cpp:1519] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_REG_EVT
[V][BLEUtils.cpp:1709] dumpGattServerEvent(): dumpGattServerEvent: *** NOT CODED ***
[D][BLEServer.cpp:153] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_REG_EVT
[V][FreeRTOS.cpp:120] give(): Semaphore giving: name: RegisterAppEvt (0x3ffe461c), owner: registerApp
[D][BLEServer.cpp:280] handleGATTServerEvent(): << handleGATTServerEvent
[V][FreeRTOS.cpp:86] wait(): << wait: Semaphore released: name: RegisterAppEvt (0x3ffe461c), owner: registerApp
[D][BLEServer.cpp:294] registerApp(): << registerApp
[D][BLEDevice.cpp:89] createServer(): << createServer
[I][BLEDevice.cpp:561] getAdvertising(): create advertising
[D][BLEDevice.cpp:563] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:169] setScanResponseData(): >> setScanResponseData
[D][BLEAdvertising.cpp:177] setScanResponseData(): << setScanResponseData
[V][BLEUtils.cpp:1038] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_SCAN_RSP_DATA_RAW_SET_COMPLETE_EVT
[D][BLEAdvertising.cpp:186] start(): >> start: customAdvData: 0, customScanResponseData: 1
[V][BLEUtils.cpp:1253] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEDevice.cpp:563] getAdvertising(): get advertising
`

elettro1, 2019-05-21:
Try to see if you solve with this fix.

https://github.com/espressif/arduino-esp32/pull/2800

closed 2019-08-14 by stale[bot]