espressif/arduino-esp32 #2310: template classes do not work in multiple file project

TopiRinkinen, 2019-01-11:
Hi,

Multiple file project (one *.ino, one *.h and one *.cpp), where h-file defines a template class, and the cpp file has function body for the template class member function.

If the template class object is instantiated inside the cpp-file, it links correctly.

If the template class object is instantiated inside the ino-file, the linker claims that the template member-function cannot be found.

If the class is a non-template class, it works fine.

bug, feature, or as intended?

Br, -Topi

czuvich, 2019-01-12:
Templates need their definition in the .h file.

https://stackoverflow.com/questions/115703/storing-c-template-function-definitions-in-a-cpp-file

czuvich, 2019-01-12:
Can you post your code?

TopiRinkinen, 2019-01-12:
Yup,

Note also that if the class c is changed to non-template class, the compilation succeeds.

template_multifile_tester.ino:
```
#include "file.h"

c<int> object; // If this is commented, the compilation succeeds.

void setup() 
{
}

void loop() 
{
}
```

file.h:
```
#ifndef FILE_H_INCLUDED
#define FILE_H_INCLUDED

template<class T>
class c
{
public:
  int a;
  int x;
  c();
  ~c()
  {
    
  }
};

int file_function(void);

#endif
```

file.cpp:
```
#include "file.h"

template<class T>
c<T>::c()
{
  a=5;
}

int file_function(void)
{
  c<int> aaa;
}
```

TopiRinkinen, 2019-01-12:
I also noted that the behaviour is identical if the board is changed to ESP8266.

czuvich, 2019-01-12:
Move your implementation of the template class to the .h file and get rid of your .cpp.  If you take a look at the stackoverflow article I referenced earlier, you'll see how it's done.

TopiRinkinen, 2019-01-12:
I noticed also that inline coding of template class methods does work. But:
- It is not per C++ standrad. It really should work as I wrote it. 
- Static methods and static variables (of the class) can not be written as inline.

TopiRinkinen, 2019-01-12:
Ah, sorry, my fault.

So long when I last used templates, that I forgot how to code them properly.

As template class and the functions inside are generated on compile-time (no on link-time), the have to be implemented in the header file...

closed 2019-01-12 by TopiRinkinen

TopiRinkinen, 2019-01-12:
This is the code I needed to write.

IRQ based watt-hour meter interface (x pulses per each kWh consumed).
The template parameter generates a new isr function for each new instance (pin) used, as I need to read several meters within one board.

usage: 
```
#define METER_1_PIN GPIO_NUM_34
electricity_meter<METER_1_PIN> ele_1;
...
float watts = ele_1.watts(800); // 800 pulses per kwh.
```
 

The header file electricity_meter.h (no cpp needed):
```
#ifndef ELECTRICITY_METER_H_INCLUDED
#define ELECTRICITY_METER_H_INCLUDED

#include <Arduino.h>

template<uint8_t pin>
class electricity_meter
{
  volatile static uint64_t isr_cnt;
  static int used_count;
  static portMUX_TYPE isr_mux;
  static const int buff_size = 10;
  volatile static bool buff_ready;
  volatile static uint32_t buff[buff_size];
  static int buff_cnt;

  static void IRAM_ATTR ele_isr();
public:
  electricity_meter();
  ~electricity_meter();
  uint64_t get_pulses(void);
  float sample_rate(void);
  float watts(float pulses_per_kwh);
};

template<uint8_t pin>
void IRAM_ATTR electricity_meter<pin>::ele_isr()
{
  uint32_t isr_us;
  int n;

  isr_us = micros();
  
  portENTER_CRITICAL_ISR(&isr_mux);
  isr_cnt++;
  
  for(n=0; n<buff_size-1; n++)
    buff[n+1] = buff[n];
  buff[0] = isr_us;

  if(buff_cnt < buff_size)
    buff_cnt++;
  else
    buff_ready = true;
  
  portEXIT_CRITICAL_ISR(&isr_mux);  
}

template<uint8_t pin>
volatile uint64_t electricity_meter<pin>::isr_cnt = 0;

template<uint8_t pin>
int electricity_meter<pin>::used_count = 0;

template<uint8_t pin>
portMUX_TYPE electricity_meter<pin>::isr_mux = portMUX_INITIALIZER_UNLOCKED;

template<uint8_t pin>
volatile uint32_t electricity_meter<pin>::buff[buff_size];

template<uint8_t pin>
int electricity_meter<pin>::buff_cnt = 0;

template<uint8_t pin>
volatile bool electricity_meter<pin>::buff_ready = false;

template<uint8_t pin>
electricity_meter<pin>::electricity_meter()
{
  if(used_count++ > 0)
    return;
  isr_cnt = 0;
  buff_cnt = 0;
  buff_ready = false;
  pinMode(pin, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(pin), ele_isr, RISING);
}

template<uint8_t pin>
electricity_meter<pin>::~electricity_meter()
{
  if(--used_count == 0)
  {
    portENTER_CRITICAL(&isr_mux);
    detachInterrupt(digitalPinToInterrupt(pin));
    portEXIT_CRITICAL(&isr_mux);
  }
}

template<uint8_t pin>
uint64_t electricity_meter<pin>::get_pulses(void)
{
  uint64_t r;
  portENTER_CRITICAL(&isr_mux);
  r = isr_cnt;
  portEXIT_CRITICAL(&isr_mux);
  return r;
}

template<uint8_t pin>
float electricity_meter<pin>::sample_rate(void)
{
  uint32_t buff_diff_us;
  float pulses_per_sec_1;
  float pulses_per_sec_2;
  uint32_t now_us;
  uint32_t diff_from_isr_us;
  bool rdy;

  now_us = micros();

  portENTER_CRITICAL(&isr_mux);
  buff_diff_us = buff[0] - buff[buff_size-1];
  diff_from_isr_us = now_us - buff[0];
  rdy = buff_ready;
  portEXIT_CRITICAL(&isr_mux);
  
  if(!rdy)
    return -1;

  pulses_per_sec_1 = (buff_size-1)/(float)buff_diff_us * 1.0e6;
  pulses_per_sec_2 = 1.0e6/diff_from_isr_us;

  return min(pulses_per_sec_1, pulses_per_sec_2);
}

template<uint8_t pin>
float electricity_meter<pin>::watts(float pulses_per_kwh)
{
  float rate;
  float watts;

  rate = sample_rate; // pulses per second;
  if(rate<0)
    return -1; // not ready or error.

  watts = rate*1000.0*3600.0/pulses_per_kwh;
  return watts;
}

#endif

```

bertmelis, 2019-01-12:
Why don't you use the pulse counter for that purpose? https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/pcnt.html

TopiRinkinen, 2019-01-13:
> Why don't you use the pulse counter for that purpose? https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/pcnt.html

Because the kWh meter gives pulses in range of few pulses per second downto one in a minute. 
This code takes the duration from last detected pulse into an account while calculating the current power in the situation when the power had dropped. 
If pulse rate is tens or more pulses per measuring interval, then the pulse counter does work.