espressif/arduino-esp32 #2313: Arduino OTA as Firmware resets while uploading

wgbeckmann, 2019-01-13:
### Hardware:
Board:							ODROID-GO
Core Installation version:			Master-Branch
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
Partition Schema:                                   Standard
Computer OS:                                        Ubuntu

### Description:
Hi,
I want to update my ODUID-GO via OTA, but the OTA Sketch should be a firmware.
I got the ArduinoOTA Basic sketch and replaced al Serial.prints to GO.lcd.print...
After compile I make a Firmware with mkfw out of it. Put it on the Flash Card an install it on the ODROID-GO. And the Sketch is ready for getting an OTA Update. I can see my ODROID in the Arduino IDE.
So far so good.

But if I make an update only a few bytes (4096) are received then suddenly the ODROID-GO resets.

I flashed the same sketch via USB to the ODROID-GO and everything works fine.  

The RESET reason (look at the Sketch) is for both CPUs "SW_CPU_RESET"

### Sketch: 

Here my mkfw line:
./mkfw test tile.raw 0 16 2097152 app /tmp/arduino_build_586302/Test1.ino.bin

Here my sketch:
```cpp
#include <WiFi.h>
#include <ESPmDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>
#include <odroid_go.h>
#include <rom/rtc.h>

const char* ssid = ".....";
const char* password = ".....";

void setup() {
  GO.begin(); 
  GO.lcd.println("Hello, ODROID-GO V3");
  delay(1000);
  GO.lcd.println("Booting");
  GO.lcd.println("CPU0 reset reason:");
  print_reset_reason(rtc_get_reset_reason(0));
  verbose_print_reset_reason(rtc_get_reset_reason(0));

  GO.lcd.println("CPU1 reset reason:");
  print_reset_reason(rtc_get_reset_reason(1));
  verbose_print_reset_reason(rtc_get_reset_reason(1));

  GO.lcd.println("Connecting to WIFI");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    GO.lcd.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
  }

  // Port defaults to 3232
  // ArduinoOTA.setPort(3232);

  // Hostname defaults to esp3232-[MAC]
  // ArduinoOTA.setHostname("myesp32");

  // No authentication by default
  // ArduinoOTA.setPassword("admin");

  // Password can be set with it's md5 value as well
  // MD5(admin) = 21232f297a57a5a743894a0e4a801fc3
  // ArduinoOTA.setPasswordHash("21232f297a57a5a743894a0e4a801fc3");

  ArduinoOTA
    .onStart([]() {
      String type;
      if (ArduinoOTA.getCommand() == U_FLASH)
        type = "sketch";
      else // U_SPIFFS
        type = "filesystem";

      // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
      GO.lcd.println("Start updating " + type);
    })
    .onEnd([]() {
      GO.lcd.println("\nEnd");
      delay(10000);
    })
    .onProgress([](unsigned int progress, unsigned int total) {
      GO.lcd.printf("Progress: %u von %u\r\n", progress,total);
    })
    .onError([](ota_error_t error) {
      GO.lcd.printf("Error[%u]: ", error);
      if (error == OTA_AUTH_ERROR) GO.lcd.println("Auth Failed");
      else if (error == OTA_BEGIN_ERROR) GO.lcd.println("Begin Failed");
      else if (error == OTA_CONNECT_ERROR) GO.lcd.println("Connect Failed");
      else if (error == OTA_RECEIVE_ERROR) GO.lcd.println("Receive Failed");
      else if (error == OTA_END_ERROR) GO.lcd.println("End Failed");
      delay(50000);
    });

  ArduinoOTA.begin();

  GO.lcd.println("Ready");
  GO.lcd.print("IP address: ");
  GO.lcd.println(WiFi.localIP());
  delay(2000);
  GO.lcd.print("Waiting on OTA ");
}

void print_reset_reason(RESET_REASON reason)
{
  switch ( reason)
  {
    case 1 : GO.lcd.println ("POWERON_RESET");break;          /**<1,  Vbat power on reset*/
    case 3 : GO.lcd.println ("SW_RESET");break;               /**<3,  Software reset digital core*/
    case 4 : GO.lcd.println ("OWDT_RESET");break;             /**<4,  Legacy watch dog reset digital core*/
    case 5 : GO.lcd.println ("DEEPSLEEP_RESET");break;        /**<5,  Deep Sleep reset digital core*/
    case 6 : GO.lcd.println ("SDIO_RESET");break;             /**<6,  Reset by SLC module, reset digital core*/
    case 7 : GO.lcd.println ("TG0WDT_SYS_RESET");break;       /**<7,  Timer Group0 Watch dog reset digital core*/
    case 8 : GO.lcd.println ("TG1WDT_SYS_RESET");break;       /**<8,  Timer Group1 Watch dog reset digital core*/
    case 9 : GO.lcd.println ("RTCWDT_SYS_RESET");break;       /**<9,  RTC Watch dog Reset digital core*/
    case 10 : GO.lcd.println ("INTRUSION_RESET");break;       /**<10, Instrusion tested to reset CPU*/
    case 11 : GO.lcd.println ("TGWDT_CPU_RESET");break;       /**<11, Time Group reset CPU*/
    case 12 : GO.lcd.println ("SW_CPU_RESET");break;          /**<12, Software reset CPU*/
    case 13 : GO.lcd.println ("RTCWDT_CPU_RESET");break;      /**<13, RTC Watch dog Reset CPU*/
    case 14 : GO.lcd.println ("EXT_CPU_RESET");break;         /**<14, for APP CPU, reseted by PRO CPU*/
    case 15 : GO.lcd.println ("RTCWDT_BROWN_OUT_RESET");break;/**<15, Reset when the vdd voltage is not stable*/
    case 16 : GO.lcd.println ("RTCWDT_RTC_RESET");break;      /**<16, RTC Watch dog reset digital core and rtc module*/
    default : GO.lcd.println ("NO_MEAN");
  }
}

void verbose_print_reset_reason(RESET_REASON reason)
{
  switch ( reason)
  {
    case 1  : GO.lcd.println ("Vbat power on reset");break;
    case 3  : GO.lcd.println ("Software reset digital core");break;
    case 4  : GO.lcd.println ("Legacy watch dog reset digital core");break;
    case 5  : GO.lcd.println ("Deep Sleep reset digital core");break;
    case 6  : GO.lcd.println ("Reset by SLC module, reset digital core");break;
    case 7  : GO.lcd.println ("Timer Group0 Watch dog reset digital core");break;
    case 8  : GO.lcd.println ("Timer Group1 Watch dog reset digital core");break;
    case 9  : GO.lcd.println ("RTC Watch dog Reset digital core");break;
    case 10 : GO.lcd.println ("Instrusion tested to reset CPU");break;
    case 11 : GO.lcd.println ("Time Group reset CPU");break;
    case 12 : GO.lcd.println ("Software reset CPU");break;
    case 13 : GO.lcd.println ("RTC Watch dog Reset CPU");break;
    case 14 : GO.lcd.println ("for APP CPU, reseted by PRO CPU");break;
    case 15 : GO.lcd.println ("Reset when the vdd voltage is not stable");break;
    case 16 : GO.lcd.println ("RTC Watch dog reset digital core and rtc module");break;
    default : GO.lcd.println ("NO_MEAN");
  }
}

void loop() {
  
  GO.lcd.print(".");
  for (int i=0;i<10;i++) {
     delay(100);
     ArduinoOTA.handle();
  }
  
}
```

bertmelis, 2019-01-13:
take ArduinoOTA.handle() out of the for loop.

wgbeckmann, 2019-01-13:
Hi,
without ArduinoOTA.handle() I cannot contact the ODROID-GO.

```
12:09:14 [ERROR]: No response from the ESP
```

lbernstone, 2019-01-13:
The extra for loop in loop() is not necessary, but shouldn't be causing a reboot.  Connect your device via USB, turn on verbose debugging and watch the serial monitor while you do an OTA update.  
Does OTA complete if you use the BasicOTA example with no changes for the OLED?

wgbeckmann, 2019-01-15:
Ok, I changed the loop() to:
```cpp
unsigned long lasttime = millis();
void loop() {
  if ((millis() - lasttime) > 1000) {
    Serial.print(".");
    lasttime = millis();  
  }
  ArduinoOTA.handle();
}
```

And added Core Debug:
```
[20:33:07:217] ODROID_GO initializing...OK␍␊
[20:33:07:586] Hello, ODROID-GO V4␍␊
[20:33:09:886] OTA server at: esp32-30aea4c7d634.local:3232␊
[20:33:09:886] ␍␊
[20:33:11:886] ...........................................Start updating sketch␍␊
[20:33:54:284] Progress: 0 von 835136␍Progress: 1024 von 835136␍Progress: 2048 von 835136␍Progress: 3072 von 835136␍Progress: 4096 von 835136␍␍␊
[20:33:54:322] Backtrace: 0x4008c934:0x3ffb1df0 0x4008cb65:0x3ffb1e10 0x400e82d1:0x3ffb1e30 0x400823d3:0x3ffb1e50 0x400d98dd:0x3ffb1e70 0x400d517c:0x3ffb1e90 0x400d53a9:0x3ffb1eb0 0x400d4daa:0x3ffb1ed0 0x400d4f4e:0x3ffb1f70 0x400d242b:0x3ffb1f90 0x400da31d:0x3ffb1fb0 0x4008ebb5:0x3ffb1fd0␍␊
[20:33:54:346] ␍␊
[20:33:54:346] Rebooting...␍␊
[20:33:54:346] ets Jun  8 2016 00:22:57␍␊
[20:33:54:380] ␍␊
[20:33:54:380] rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)␍␊
[20:33:54:380] configsip: 0, SPIWP:0xee␍␊
[20:33:54:380] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00␍␊
[20:33:54:380] mode:DIO, clock div:1␍␊
[20:33:54:380] load:0x3fff0018,len:4␍␊
[20:33:54:380] load:0x3fff001c,len:6724␍␊
[20:33:54:380] load:0x40078000,len:8900␍␊
[20:33:54:380] ho 0 tail 12 room 4␍␊
[20:33:54:380] load:0x40080000,len:7388␍␊
[20:33:54:380] entry 0x4008039c␍␊
[20:33:54:380] <0x1b>[0;32mI (30) boot: ESP-IDF v3.1-odroid-1-g08f2c8f26 2nd stage bootloader<0x1b>[0m␍␊
[20:33:54:424] <0x1b>[0;32mI (31) boot: compile time 09:24:57<0x1b>[0m␍␊
[20:33:54:424] <0x1b>[0;32mI (31) boot: Enabling RNG early entropy source...<0x1b>[0m␍␊
[20:33:54:424] <0x1b>[0;32mI (37) qio_mode: Enabling default flash chip QIO<0x1b>[0m␍␊
[20:33:54:424] <0x1b>[0;32mI (42) boot: SPI Speed      : 80MHz<0x1b>[0m␍␊
[20:33:54:424] <0x1b>[0;32mI (46) boot: SPI Mode       : QIO<0x1b>[0m␍␊
[20:33:54:424] <0x1b>[0;32mI (50) boot: SPI Flash Size : 16MB<0x1b>[0m␍␊
[20:33:54:424] <0x1b>[0;32mI (54) boot: Partition Table:<0x1b>[0m␍␊
[20:33:54:424] <0x1b>[0;32mI (58) boot: ## Label            Usage          Type ST Offset   Length<0x1b>[0m␍␊
[20:33:54:457] <0x1b>[0;32mI (65) boot:  0 nvs              WiFi data        01 02 00009000 00004000<0x1b>[0m␍␊
[20:33:54:457] <0x1b>[0;32mI (73) boot:  1 otadata          OTA data         01 00 0000d000 00002000<0x1b>[0m␍␊
[20:33:54:457] <0x1b>[0;32mI (80) boot:  2 phy_init         RF data          01 01 0000f000 00001000<0x1b>[0m␍␊
[20:33:54:457] <0x1b>[0;32mI (88) boot:  3 firmware         factory app      00 00 00010000 000f0000<0x1b>[0m␍␊
[20:33:54:457] <0x1b>[0;32mI (95) boot:  4 app              OTA app          00 10 00100000 00200000<0x1b>[0m␍␊
[20:33:54:457] <0x1b>[0;32mI (103) boot: End of partition table<0x1b>[0m␍␊
[20:33:54:492] <0x1b>[0;32mI (107) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x32d14 (208148) map<0x1b>[0m␍␊
[20:33:54:519] <0x1b>[0;32mI (171) esp_image: segment 1: paddr=0x00132d3c vaddr=0x3ff80000 size=0x00000 (     0) load<0x1b>[0m␍␊
[20:33:54:524] <0x1b>[0;32mI (171) esp_image: segment 2: paddr=0x00132d44 vaddr=0x3ff80000 size=0x00000 (     0) load<0x1b>[0m␍␊
[20:33:54:542] <0x1b>[0;32mI (178) esp_image: segment 3: paddr=0x00132d4c vaddr=0x3ffbdb60 size=0x03068 ( 12392) load<0x1b>[0m␍␊
```

Hmmm,
ot booting I saw the Partition table. Perhaps thats the thing.

lbernstone, 2019-01-15:
You are getting a backtrace.  Please decode it: https://github.com/me-no-dev/EspExceptionDecoder

wgbeckmann, 2019-01-15:
Yes, it is the Partition table. I create manualy a new one:
```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x4000,
otadata,  data, ota,     0xd000,  0x2000,
phy_init, data, 0x01,    0xf000,  0x1000,
firmware, app,  factory, 0x10000, 0xf0000,
app0,     app,  ota_0,   0x100000,0x150000,
app1,     app,  ota_1,   0x250000,0x150000,
eeprom,   data, 0x99,    0x400000,0x1000,
spiffs,   data, spiffs,  0x401000,0x200000
```

Compiled it:
```
gen_esp32part.py firmware.csv firmware.bin
```

And flashed it:
```
esptool.py write_flash 0x8000 firmware.bin 
```

And it works.
But when I write it with the Firmeware again. The old partition table is back again...

wgbeckmann, 2019-01-15:
Ok, I found the reason.

In main.c of ~/Arduino/hardware/espressif/esp32/tools/odroid-go-firmware (The Firmware of the ODROID-GO) is a new partition Table created after flashing the firmware file.

wgbeckmann, 2019-01-18:
Here is the Backtrace:
```
0x4008c934: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 707
0x4008cb65: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 707
0x400e829d: is_safe_write_address at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/flash_ops.c line 125
0x400823d3: spi_flash_erase_sector at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/flash_ops.c line 203 (discriminator 2)
0x400d98a9: EspClass::flashEraseSector(unsigned int) at /home/wolf/Arduino/hardware/espressif/esp32/cores/esp32/Esp.cpp line 210
0x400d5148: UpdateClass::_writeBuffer() at /home/wolf/Arduino/hardware/espressif/esp32/libraries/Update/src/Updater.cpp line 363
0x400d5375: UpdateClass::write(unsigned char*, unsigned int) at /home/wolf/Arduino/hardware/espressif/esp32/libraries/Update/src/Updater.cpp line 363
0x400d4d76: ArduinoOTAClass::_runUpdate() at /home/wolf/Arduino/hardware/espressif/esp32/libraries/ArduinoOTA/src/ArduinoOTA.cpp line 105
0x400d4f1a: ArduinoOTAClass::handle() at /home/wolf/Arduino/hardware/espressif/esp32/libraries/ArduinoOTA/src/ArduinoOTA.cpp line 105
0x400d23f7: loop() at /home/wolf/Arduino/ESP32/Test1/Test1.ino line 158
0x400da2e9: loopTask(void*) at /home/wolf/Arduino/hardware/espressif/esp32/cores/esp32/main.cpp line 25
0x4008ebb5: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 355 (discriminator 1)
```

wgbeckmann, 2019-01-18:
Ok, I looked a little bit deeper after I saw the Stacktrace.
And the Idea with the Partition Table above was not right. I flashed my App at Firmware App position so the OTA runns with two partitions, but a Reset to the Original Firmware was not posible.
-----
The Error is now clear for me:
I have only one APP Partition. Because the Firmware whites a Partition Table after flashing from the Flash-Card with only one APP Partition.
When the Updater wants to flash, it gets the active APP partition and when the Updater flashes into that Partition the is_safe_write_adress function runs this code:
```cpp
    const esp_partition_t *p = esp_ota_get_running_partition();
    if (addr >= p->address && addr < p->address + p->size) {
        UNSAFE_WRITE_ADDRESS;
    }
```
So the write_adress is unsafe and the Updater aborts....

The only Idea I have is to update the partition Table and add another APP Partition before starting the Update. 
Hmm, tomorrow the Weekend starts the Weekend ;-)

wgbeckmann, 2019-01-18:
Yes, I thougt it too, but esp_ota_get_next_update_partition() gives also the only existing Partition back if it findes nothing. See here: 
[https://github.com/espressif/esp-idf/blob/master/components/app_update/esp_ota_ops.c#L522](https://github.com/espressif/esp-idf/blob/master/components/app_update/esp_ota_ops.c#L522)

wgbeckmann, 2019-01-19:
I go another way:
My goal was to put new Arduino Sketches to the ODROID-GO without loosing the original Firmware. To write e new Partition Table is a too big step for me, so i created a new "Board" called "ODROID-GO with Firmware" If I use this board only the Partition Table and the APP0 is flashed to the ODROID-GO.

So only actual the running App is overwritten. The only thing is, that you must have an actual running App (eg. the Go-Play or MicroPython).

For somone who wants to do the same:
Put in  **~/Arduino/hardware/espressif/esp32/tools/partitions** the File **firmware.csv**  with the content:
```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x4000,
otadata,  data, ota,     0xd000,  0x2000,
phy_init, data, 0x01,    0xf000,  0x1000,
firmware, app,  factory, 0x10000, 0xf0000,
app,      app,  ota_0,   0x100000,0x640000,
eeprom,   data, 0x99,    0x740000,0x1000,
spiffs,   data, spiffs,  0x741000,0x200000,
```
**Add** to **~/Arduino/hardware/espressif/esp32/boards.txt** the following lines:
```
##############################################################

odroid_esp32_fw.name=ODROID ESP32 with Firmware

odroid_esp32_fw.upload.tool=esptool_py_fw
odroid_esp32_fw.upload.maximum_size=1310720
odroid_esp32_fw.upload.maximum_data_size=327680
odroid_esp32_fw.upload.wait_for_upload_port=true

odroid_esp32_fw.serial.disableDTR=true
odroid_esp32_fw.serial.disableRTS=true

odroid_esp32_fw.build.mcu=esp32
odroid_esp32_fw.build.core=esp32
odroid_esp32_fw.build.variant=odroid_esp32
odroid_esp32_fw.build.board=ODROID_ESP32

odroid_esp32_fw.build.f_cpu=240000000L
odroid_esp32_fw.build.flash_size=16MB
odroid_esp32_fw.build.flash_mode=dio
odroid_esp32_fw.build.boot=dio
odroid_esp32_fw.build.partitions=default
odroid_esp32_fw.build.defines=

odroid_esp32_fw.menu.FlashMode.qio=QIO
odroid_esp32_fw.menu.FlashMode.qio.build.flash_mode=dio
odroid_esp32_fw.menu.FlashMode.qio.build.boot=qio
odroid_esp32_fw.menu.FlashMode.dio=DIO
odroid_esp32_fw.menu.FlashMode.dio.build.flash_mode=dio
odroid_esp32_fw.menu.FlashMode.dio.build.boot=dio
odroid_esp32_fw.menu.FlashMode.qout=QOUT
odroid_esp32_fw.menu.FlashMode.qout.build.flash_mode=dout
odroid_esp32_fw.menu.FlashMode.qout.build.boot=qout
odroid_esp32_fw.menu.FlashMode.dout=DOUT
odroid_esp32_fw.menu.FlashMode.dout.build.flash_mode=dout
odroid_esp32_fw.menu.FlashMode.dout.build.boot=dout

odroid_esp32_fw.menu.FlashFreq.80=80MHz
odroid_esp32_fw.menu.FlashFreq.80.build.flash_freq=80m
odroid_esp32_fw.menu.FlashFreq.40=40MHz
odroid_esp32_fw.menu.FlashFreq.40.build.flash_freq=40m

odroid_esp32_fw.menu.PartitionScheme.firmware=Firmware
odroid_esp32_fw.menu.PartitionScheme.firmware.build.partitions=firmware

odroid_esp32_fw.menu.UploadSpeed.921600=921600
odroid_esp32_fw.menu.UploadSpeed.921600.upload.speed=921600
odroid_esp32_fw.menu.UploadSpeed.115200=115200
odroid_esp32_fw.menu.UploadSpeed.115200.upload.speed=115200
odroid_esp32_fw.menu.UploadSpeed.256000.windows=256000
odroid_esp32_fw.menu.UploadSpeed.256000.upload.speed=256000
odroid_esp32_fw.menu.UploadSpeed.230400.windows.upload.speed=256000
odroid_esp32_fw.menu.UploadSpeed.230400=230400
odroid_esp32_fw.menu.UploadSpeed.230400.upload.speed=230400
odroid_esp32_fw.menu.UploadSpeed.460800.linux=460800
odroid_esp32_fw.menu.UploadSpeed.460800.macosx=460800
odroid_esp32_fw.menu.UploadSpeed.460800.upload.speed=460800
odroid_esp32_fw.menu.UploadSpeed.512000.windows=512000
odroid_esp32_fw.menu.UploadSpeed.512000.upload.speed=512000

odroid_esp32_fw.menu.DebugLevel.none=None
odroid_esp32_fw.menu.DebugLevel.none.build.code_debug=0
odroid_esp32_fw.menu.DebugLevel.error=Error
odroid_esp32_fw.menu.DebugLevel.error.build.code_debug=1
odroid_esp32_fw.menu.DebugLevel.warn=Warn
odroid_esp32_fw.menu.DebugLevel.warn.build.code_debug=2
odroid_esp32_fw.menu.DebugLevel.info=Info
odroid_esp32_fw.menu.DebugLevel.info.build.code_debug=3
odroid_esp32_fw.menu.DebugLevel.debug=Debug
odroid_esp32_fw.menu.DebugLevel.debug.build.code_debug=4
odroid_esp32_fw.menu.DebugLevel.verbose=Verbose
odroid_esp32_fw.menu.DebugLevel.verbose.build.code_debug=5
```

And finally **add** to **~/Arduino/hardware/espressif/esp32/platform.txt** the following lines:
```
# ------- Upload with Firmware ---------
tools.esptool_py_fw.path={runtime.platform.path}/tools/esptool
tools.esptool_py_fw.cmd=esptool
tools.esptool_py_fw.cmd.linux=esptool.py
tools.esptool_py_fw.cmd.windows=esptool.exe

tools.esptool_py_fw.upload.protocol=esp32
tools.esptool_py_fw.upload.params.verbose=
tools.esptool_py_fw.upload.params.quiet=
tools.esptool_py_fw.upload.pattern="{path}/{cmd}" --chip esp32 --port "{serial.port}" --baud {upload.speed}  --before default_reset --after hard_reset write_flash -z --flash_mode {build.flash_mode} --flash_freq {build.flash_freq} --flash_size detect  0x100000 "{build.path}/{build.project_name}.bin" 0x8000 "{build.path}/{build.project_name}.partitions.bin"
tools.esptool_py_fw.upload.pattern.linux=python "{path}/{cmd}" --chip esp32 --port "{serial.port}" --baud {upload.speed}  --before default_reset --after hard_reset write_flash -z --flash_mode {build.flash_mode} --flash_freq {build.flash_freq} --flash_size detect  0x100000 "{build.path}/{build.project_name}.bin" 0x8000 "{build.path}/{build.project_name}.partitions.bin"
tools.esptool_py_fw.upload.network_pattern={network_cmd} -i "{serial.port}" -p "{network.port}" "--auth={network.password}" -f "{build.path}/{build.project_name}.bin"
```

If you want to make your on .fw file if you are done with develpment follow this guide:
[Arduino Applications](https://wiki.odroid.com/odroid_go/arduino_app)

closed 2019-01-19 by wgbeckmann