espressif/arduino-esp32 #2317: Adafruit ESP32 Feather not compiling with 1.0.1, but works with 1.0.0

gamingoutbrexit, 2019-01-14:
### Hardware:
Board:							Adafruit ESP32 Feather (aka Huzzah32)
Core Installation version:			1.0.1
IDE name:						Arduino 1.8.8
Flash Frequency:					80Mhz
PSRAM enabled:                                    don't know
Upload Speed:					921600
Computer OS:                                        Mac OSX

### Description:
Describe your problem here
I want to run simple Arduino examples like Blink and WiFiScan. The programs worked with 1.0.0 of esp but not once I upgraded to 1.0.1 in the Arduino Boards Manager.  I can fix the problem by using the Boards Manager to remove esp32 and then use it to install version 1.0.0 again. I have replicated it on two different Macs.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>
void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
}
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   
  delay(2000);                       
  digitalWrite(LED_BUILTIN, LOW);    
  delay(500);                       
}
```

### Debug Messages:
Arduino: 1.8.8 (Mac OS X), Board: "Adafruit ESP32 Feather, 80MHz, 921600, None"
Board featheresp32 (platform esp32, package esp32) is unknown
Error compiling for board Adafruit ESP32 Feather.
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

me-no-dev, 2019-01-23:
Try installing 1.0.1 again. The Adafruit Feather works on my end. (Did you restart IDE after update?)

hasalasri, 2019-05-16:
Try deleting the esp32 folder in the following path and install again.
 C:\Users\userName\AppData\Local\Arduino15\packages

closed 2019-08-14 by stale[bot]