espressif/arduino-esp32 #2320: BluetoothSerial Status && API discussion for passkey/PIN

Craftplorer, 2019-01-14:
Hi,
i have a few question and topic for possible discussion.
My first question would be what the status on the Branch: bt-serial-overhaul is. Is it already merged? Should future commits/pull-request affecting the BluetoothSerial library be done on this branch?

On #1458 it was already ask for a way to use passkey/PIN with this library. How should a API look like for something like this in a arduino-ish style? Any suggestion?

Currently the BluetoothSerial library is limited to one device connected at a time. #2061 
Is there any plan on allowing more devices to recieve serial data via bluetooth?

I would like to contribute by coding some part but need my questions answered before starting to spend time coding.

mouridis, 2019-01-14:
Regarding your multiple connections question, actually, accepting multiple connections was actually the default behavior of the BluetoothSerial library before commit 0d564d7b1d7ed2a85dcbd80e4b4e6f6e393e486d which accepted the PR you mentioned and limited incoming connections to a maximum of 1.

You can revert the changes of that commit and you will disable the limit, but keep in mind:
1. ESP will receive messages from ALL connected devices and you will not have a way to distinguish which message comes from which device because all messages are on the same stream. (you can easily solve this by adding some kind of client IDs in the each of the messages that the clients sent)
2. When ESP sends messages, ONLY the last (chronologically) client connected receives the messages from the ESP. So, if you want to broadcast messages to all clients, or control the recipient for each message send from ESP, it's not possible with the solution I suggest, (and the library tweaking required to fix that is not that simple).

Also check #2055 to why the limit was placed.

Craftplorer, 2019-01-14:
Yeah i was aware of these problems. My question was a bit unclear.
It should be more like this:
Is there any plan to make the library send serial data to all connected devices instead of only the last conected device? Is it even possbile? Im not really sure how the whole bluetooth stuff works in detail and how the underlying software (ESP-IDF) is handling it.

mouridis, 2019-01-14:
Yes, generally speaking, of course it's possible.

Looking at the current BluetoothSerial code (which I suggest that you do - I'm a very novice C coder but I could understand the mechanics easily), I understand that the correct way to implement support for multiple individual and independent clients would not be an easy task. (it would require handling of multiple RX, TX buffers and many other things)

But if you need a dirty fix for just simultaneous broadcast to all connected clients, I think it could be implemented with not much effort. 

The library essentially transmits data in this line:
https://github.com/espressif/arduino-esp32/blob/0d564d7b1d7ed2a85dcbd80e4b4e6f6e393e486d/libraries/BluetoothSerial/src/BluetoothSerial.cpp#L90

That _spp_client parameter of esp_spp_write contains the handle of the recipient connected client.  

The variable receives value when a client is connected here:
https://github.com/espressif/arduino-esp32/blob/0d564d7b1d7ed2a85dcbd80e4b4e6f6e393e486d/libraries/BluetoothSerial/src/BluetoothSerial.cpp#L165-L174

Now, as you can see, currently, the library only stores the handle of the most recently connected client.

You could easily modify _spp_client to be a data structure as an array or a queue, and then modify this snippes so that this structure holds the handles of all connected devices and not just the most recent one. 

Then, you should modify the first part so that you loop between all values of your connected clients handles and use esp_spp_write not just once, but once for each connected client.

Of course, you would need a couple more small changes here and there (i.e. handle the disconnect of a client and update your data structure) and it would still be a very dirty fix, but it should work.

copercini, 2019-01-19:
> My first question would be what the status on the Branch: bt-serial-overhaul is. Is it already merged? 

yes, it fixes a packet loss problem and is already merged

> Should future commits/pull-request affecting the BluetoothSerial library be done on this branch?

no, you can use master if you want

> On #1458 it was already ask for a way to use passkey/PIN with this library. How should a API look like for something like this in a arduino-ish style? Any suggestion?

I planned to implement this, initially it would be a fixed PIN passing on begin, like  `SerialBT.begin("ESP32name", 123456);`, but it's a legacy BT method, and hard to enable on current ESP32 BT stack, that currently uses Secure Simple Pairing (SSP), the discussion is here: https://github.com/espressif/esp-idf/issues/2774

> Currently the BluetoothSerial library is limited to one device connected at a time. #2061
Is there any plan on allowing more devices to recieve serial data via bluetooth?

It's possible, but will they be independent clients? or all the received data will be mixed? how to handle it?

> I would like to contribute by coding some part but need my questions answered before starting to spend time coding.

Yes, please, any contributions are welcome !!!

Craftplorer, 2019-01-27:
> 1. ESP will receive messages from ALL connected devices and you will not have a way to distinguish which message comes from which device because all messages are on the same stream. (you can easily solve this by adding some kind of client IDs in the each of the messages that the clients sent)

Can the handle in ESP_SPP_DATA_IND_EVT be used to differentiate between differant clients sending data to the esp? I also get the same handle on  ESP_SPP_SRV_OPEN_EVT and ESP_SPP_CLOSE_EVT

pacifickashyap, 2019-02-28:
@copercini Now we can turn SSP on/off so can you please try to implement it in BluetoothSerial

pacifickashyap, 2019-02-28:
https://github.com/espressif/esp-idf/issues/2774

copercini, 2019-03-01:
@pacifickashyap this lead us to the 2nd problem: 
SSP is already compiled as enabled in arduino core, [here](https://github.com/espressif/arduino-esp32/blob/010a7c60f7000eb17c6333309c87156769848a0e/tools/sdk/sdkconfig#L215)

And seems not possible to change it in runtime, maybe @me-no-dev can help us with this =)

copercini, 2019-03-01:
The most significant change is this one:
https://github.com/espressif/esp-idf/blob/65142bc59edb9fbf97b98b7aa867c68c32e15cb1/components/bt/bluedroid/device/controller.c#L152-L160

but if we disable `CONFIG_BT_SSP_ENABLED` this will not work without a PIN for BT classic in all arduino core

pacifickashyap, 2019-03-02:
@copercini  can we disable it from sdkconfig.h. If yes how?

copercini, 2019-03-02:
@pacifickashyap arduino core use a aldeady compiled IDF version by @me-no-dev , so we can't change sdkconfig, unless he update it in the next version

You can use arduino as an IDF component, then you will compile everything from zero, including the sdkconfig changes

xiongyumail, 2019-03-27:
@Craftplorer 
You can try this compiled library.
[disable_ssp.zip](https://github.com/espressif/arduino-esp32/files/3014703/disable_ssp.zip)

pacifickashyap, 2019-05-20:
Thanks,
it works perfectly.

xiongyumail, 2019-05-20:
@pacifickashyap 
HI , 
You can try this `PR`. 
https://github.com/espressif/arduino-esp32/pull/2765

pacifickashyap, 2019-06-20:
Tried it is working fine. Need to give pin only when new device is paired.
But if we use IRemote.h for Infrared remote, it gives us a error(cache disable but cache memory accesed) while pin is verified.

luc-github, 2019-08-19:
Stale keep it pls ^_^

luc-github, 2019-10-29:
@me-no-dev I see  you work in v3.3 idf branch - does the BT pin protection would be enabled using this v3.3 idf version ? or will be in 4.0 only ?

HarpreetHeera, 2019-12-20:
> @Craftplorer
> You can try this compiled library.
> [disable_ssp.zip](https://github.com/espressif/arduino-esp32/files/3014703/disable_ssp.zip)

Thanks it works...

AK5005, 2020-01-29:
> @Craftplorer
> You can try this compiled library.
> [disable_ssp.zip](https://github.com/espressif/arduino-esp32/files/3014703/disable_ssp.zip)

I Tried to compile the ino file in zip after changing the file in lib, but the following error shows to me. Could somebody help with this
[image]

closed 2020-04-12 by stale[bot]

Sule-2002, 2020-05-28:
Hello I am new on ESP32,
I am working with nrf52832 bluetooth modules . I prepared static passkey "123456" on nrf modules      I can connect   on nrf connect app  after  I write static passkey .Now I am trying to connect with ESP32 DEV Module .Is there any example Arduino code for ESP32 for pairing my nrf module with passkey?   Can you advise me any example?

RdlP, 2020-07-19:
> > @Craftplorer
> > You can try this compiled library.
> > [disable_ssp.zip](https://github.com/espressif/arduino-esp32/files/3014703/disable_ssp.zip)
> 
> Thanks it works...

@HarpreetHeera using this library how can I set the PIN? (what function should I call?)

Thanks

HamzaHajeir, 2020-09-28:
> > @Craftplorer
> > You can try this compiled library.
> > [disable_ssp.zip](https://github.com/espressif/arduino-esp32/files/3014703/disable_ssp.zip)
> 
> I Tried to compile the ino file in zip after changing the file in lib, but the following error shows to me. Could somebody help with this
> [image]

It seems out-dated, It's meant for core version 1.01, Now we're on 1.04

hulktech, 2021-04-01:
Hello, a lot of people also me, we are trying to find an option to set static PIN on Bluetooth classic serial connection when pairing. A way to input custom PIN not the default 123456.An old way was created for 1.0.1 version adding a custom libbt.a file that was disabling ssp as far as i know .We want to be able to have this option on newer versions. I'll appreciate anyone who really want to help
_

SaiEmbedlover, 2021-06-17:
Hi, 
I want to transfer an audio file from esp32-a1s-audio kit to an android mobile. For this, i have taken pipeline_a2dp_source_stream example. 
I am using the following hardware and software:
1. ESP32-A1s-Audio kit 
2. Ai-Thinker ESP ADF Frame work and internal IDF within ADF Frame work folder.
3. Bluetooth Classic

When i run this example using idf command prompt,
1.  esp32 module is paired with mobile but no data is transferred and given an error in monitor with btc_a2d_src_connect abort() called presented and BT_BTC: BTA_AV_OPEN_EVT::FAILED status: 2 and module is rebooting....and roams in this loop only.
2. Also , No prompt for PIN displayed on mobile while pairing and mobile itself paired. 

    My question is what are the considerations in mobile side to connect to eps32 and to receive an audio file?
  Why system is rebooting if  the example was already tested?
Why the above errors are arised?

Please help me!  Thank you All!!

italocjs, 2021-12-09:
Nearly 2 years and no official solution from expressif?

HamzaHajeir, 2021-12-10:
> Nearly 2 years and no official solution from expressif?

Can you open a new issue referring to this?