espressif/arduino-esp32 #2322: Add AiThinker Cam board

just-jason, 2019-01-14:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git?
IDE name:						?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					?115200?
Computer OS:                                        ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

Niek, 2019-01-15:
I have tried to flash to this board, but I keep getting errors when booting. I understand that the ``esp32wrover`` board should work since it has PSRAM:
```bash
$ make -f ~/makeEspArduino/makeEspArduino.mk ESP_ROOT=~/arduino-esp32 CHIP=esp32 BOARD=esp32wrover UPLOAD_PORT=/dev/tty.usbserial erase_flash flash
```
Flashing is OK, but while booting I get a bootloop like this:
```bash
rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:9280
load:0x40080400,len:5860
entry 0x40080698
ets Jun  8 2016 00:22:57
```
This message repeats over and over.

me-no-dev, 2019-01-23:
Yes, what you are getting is not normal. Try to set pins 13 and 14 as `INPUT_PULLUP` as first lines on your code. I do not guarantee that it will work, but you can try.

Niek, 2019-01-23:
Adding this to the top of the ``setup()`` function doesn't do anything for me:
```c
pinMode(13, INPUT_PULLUP);
pinMode(14, INPUT_PULLUP);
```

Niek, 2019-02-04:
I figured out the issue, the partition size was incorrect. Here is a boards.txt entry that works best for me:
```ruby
esp32cam.name=AI Thinker ESP32-CAM

esp32cam.upload.tool=esptool_py
esp32cam.upload.maximum_size=3145728
esp32cam.upload.maximum_data_size=327680
esp32cam.upload.wait_for_upload_port=true
esp32cam.upload.speed=460800

esp32cam.serial.disableDTR=true
esp32cam.serial.disableRTS=true

esp32cam.build.mcu=esp32
esp32cam.build.core=esp32
esp32cam.build.variant=esp32
esp32cam.build.board=ESP32_DEV
esp32cam.build.f_cpu=240000000L
esp32cam.build.flash_size=4MB
esp32cam.build.flash_freq=80m
esp32cam.build.flash_mode=dio
esp32cam.build.boot=qio
esp32cam.build.partitions=huge_app
esp32cam.build.defines=-DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue
esp32cam.build.code_debug=0
```

Niek, 2019-03-11:
Created a PR: https://github.com/espressif/arduino-esp32/pull/2572

closed 2019-08-15 by stale[bot]