espressif/arduino-esp32 #2326: Classic Bluetooth lecture problem

totovr, 2019-01-15:
What you are trying to do?

I'm sending lectures from different sensor via classic serial Bluetooth, I have done it previosly but in this case when I read the lecture of a Heart Sensor once I activate the library is always turning the PIN lecture to 1023, if I read it just with normal serial is working, I have checked different pins and also putting to ground and read 0.

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 Dev Module
Core Installation version:		        1.6
IDE name:						Platform.io
Upload Speed:					115200
Computer OS:                                         Windows 10

##### This sketch is working correctly using just Serial

```c++
#include <Arduino.h>
#include <SparkFunTMP102.h>

// For GSR
// Declaration of functions
void GSRCalculation();
// Input PIN
const int GSRInput = 35;
// Variables
int sensorValue = 0;
int gsr_average = 0;
long sum = 0;
int userResistence = 0;

// For BPM
// Declaration of functions
int myTimer1(long delayTime, long currentMillis);
int myTimer2(long delayTime2, long currentMillis);
void BPMCalculation();

// Input PIN
const int BPMInput = 26;

// Variables
int UpperThreshold = 518;
int LowerThreshold = 490;
int reading = 0;
float BPM = 0.0;
int bpmInt;
bool IgnoreReading = false;
bool FirstPulseDetected = false;
unsigned long FirstPulseTime = 0;
unsigned long SecondPulseTime = 0;
unsigned long PulseInterval = 0;

// Measure every 500 seconds
const unsigned long delayTime = 10;
const unsigned long delayTime2 = 3000;
unsigned long previousMillis = 0;
unsigned long previousMillis2 = 0;

// Save previous state
int bpmPrevious = 60;
int bpmIntCurrent;
int bpmIntPrevious;

// For Calories Calculator
int CaloriesBurned(unsigned long _exerciseTime);
int age = 26;
int weight = 72;
int metValue = 5; // https://sites.google.com/site/compendiumofphysicalactivities/Activity-Categories/conditioning-exercise
unsigned long exerciseTime;
int totalCalories;

[... 427 lines cut ...]
int CaloriesBurned(unsigned long _exerciseTime)
{
  // https://www.cmsfitnesscourses.co.uk/blog/53/using-mets-to-calculate-calories
  unsigned long exerciseTimeMinutes = _exerciseTime / 60000;
  int calories = metValue * weight / 200 * exerciseTimeMinutes;
  return calories;
}

void CalculateTemperature()
{
  // Turn sensor on to start temperature measurement.
  sensor0.wakeup();

  // read temperature data
  temperature = sensor0.readTempC();

  // Place sensor in sleep mode to save power.
  // Current consumtion typically <0.5uA.
  sensor0.sleep();
}
```

This is the output of the pin that I'm reading

[image]

me-no-dev, 2019-01-16:
adc does not work on those pins when Radio is on (WiFi or BT). try to read a pin above 33 :)

totovr, 2019-01-16:
Thanks I will prove it!, one more question, if possible could you tell me which PIN can be used?

beegee-tokyo, 2019-01-16:
Here is a list of [reserved GPIO's](https://desire.giesecke.tk/index.php/2018/07/06/reserved-gpios/)

stickbreaker, 2019-01-16:
ADC2 functions are unavailable while WiFi,BlueTooth are in use.  The Pins can be uses for other functions like GPIO.

If you want to use Analog to Digital Conversion while Wifi or BlueTooth is in use, you have to use ADC1 pins : 
* GPIO32 input/output, ADC4
* GPIO33 input/output, ADC5
* GPI 34 input only,  ADC6 
* GPI 35 input only, ADC7
* GPI 36 input only, ADC0
* GPI 39 input only, ADC3

Chuck.

closed 2019-01-16 by totovr