espressif/arduino-esp32 #2327: BLE libraries in 1.0.1 output garbage to GATT client

FLamparski, 2019-01-15:
### Hardware:
Board:							ESP32-WROOM-32
Core Installation version:			1.0.1
IDE name:						Arduino IDE (v1.8.8) / VS Code
Flash Frequency:					80MHz
PSRAM enabled:                                    no
Upload Speed:					921600
Computer OS:                                        Windows 10

### Description:

I have a BLE application which outputs sizeable amounts of data to the GATT client. To facilitate this, I chunk up the message into 512-byte segments and return each subsequent segment on a new read:

```c++
    // in my BLECharacteristicCallbacks implementation:
    void onRead(BLECharacteristic *pCharacteristic) {
        // This is called by the app. On each invocation it will return
        // the next chunk of the current message. If the current message has been
        // fully transmitted, moves on to the next one. If the messages are
        // exhausted, sends empty string.
        if (curMsg == "") {
            if (mService->hasQueuedMessages()) {
                curMsg = mService->popMessage();
                pos = 0;
                // Note, this could technically notify too but notifying
                // waits on a semaphore that's already taken by this method,
                // causing a deadlock.
                // TODO: Set an event flag so that the notification can be sent
                // on the next tick.
                mService->updateNumMessages();
            }
            else {
                pCharacteristic->setValue("");
                return;
            }
        }

        auto chunk = curMsg.substr(pos, ATT_MTU);
        if (chunk.size() < ATT_MTU) {
            curMsg = "";
        }
        else {
            pos += ATT_MTU;
        }

        pCharacteristic->setValue(chunk);
    }
```

In version 1.0.0 of the library, this works fine. In 1.0.1, the function seems to be dumping garbage* to the GATT client (easily verifiable with the excellent nRF Connect app). Am I using the library wrong, or is this a bug in the new implementation?

\* Either overlapping message segments, or what looks like random memory contents; which might be pretty bad if the application is doing anything _ahem_ secure

Full sketch is at https://github.com/FLamparski/MobilityGATTDemo

me-no-dev, 2019-01-16:
@chegewara please have a look!

chegewara, 2019-01-16:
I would suggest to decrease chunk size from 512 to 510.
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLECharacteristic.cpp#L362

FLamparski, 2019-01-16:
I can try that (but otoh if the library is already doing the chunking for the MAC/PHY, should I not be able to pass arbitrary length strings to it?)

chegewara, 2019-01-16:
You are right, it should work anyway. I didnt try to use read to send big files, but i was able to send it with indicate/notify the way you are doing it.

Without testing your code or logs its hard to say why its not working.

closed 2019-08-15 by stale[bot]