espressif/arduino-esp32 #2328: Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)

rugal0123, 2019-01-15:
### Hardware:
Board:							ESP32 Dev Module1 
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git?
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                      no
Upload Speed:					921600
Computer OS:                                        Windows 10
### Description:
Describe your problem here

Hello, can anyone please help me with this issue. I'm using touch screen library xpt2046 and sometimes ran into this problem where the device crashed and restarted itself.

Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x400eb62c  PS      : 0x00060034  A0      : 0x40081708  A1      : 0x3ffc0bf0  
A2      : 0x0000000d  A3      : 0x3ffc472c  A4      : 0x00000000  A5      : 0x00002000  
A6      : 0x00000000  A7      : 0x1300005c  A8      : 0x80080f34  A9      : 0x3ffc0bd0  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3ffc569c  A13     : 0x00000000  
A14     : 0x3ffc5698  A15     : 0xffffffff  SAR     : 0x00000016  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x40082624  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400eb62c

Backtrace: 0x400eb62c:0x3ffc0bf0 0x40081705:0x3ffc0c10 0x40082621:0x00000000

Decoder result: 

[image]

Traced to line  #58:

[image]

i4things, 2019-01-15:
does the library has any interrupt handlers ?

bertmelis, 2019-01-15:
You tracked down the source of the crash tot the library. So I'd advise you to seek for help on the library's support system.

We don't even know which library you're using.

CelliesProjects, 2019-01-15:
Could be down to the interrupt handler not being in IRAM.
See (https://github.com/PaulStoffregen/XPT2046_Touchscreen/commit/1a273183a426bb71e947ab6653daaec4d41d8c7e) this PR.
Pull master from the XPT2046_Touchscreen to have the IRAM fix.

closed 2019-08-15 by stale[bot]