espressif/arduino-esp32 #2329: issue with values read analog pin - e.g. voltmeter

vtomanov, 2019-01-15:
implementing  simple voltmeter give quite a wrong readings : 

e.g. I'm comparing same setup for a simple voltmeter using Heltec ESP32  with Feather32u4 with AVR - both 3.3v logic ESP32 with 12bit ADC and AVR with 10bit ADC - the values in the end from ESP32 differs  about 0.2v sometime compared to AVR and quality multi-meter - AVR and multi-meter always the same.

I do use : esp_adc_cal.h and esp_adc_cal_characteristics_t initialized with : 

adc1_config_width(ADC_WIDTH_12Bit);
  adc1_config_channel_atten(HOUSE_VOLTMETER_PIN, ADC_ATTEN_11db);
  esp_adc_cal_value_t val_type = esp_adc_cal_characterize(ADC_UNIT_1, ADC_ATTEN_DB_11, ADC_WIDTH_BIT_12, ESP32_DEFAULT_REF, &HOUSE_VOLTMETER_adc_chars);
  if (val_type == ESP_ADC_CAL_VAL_EFUSE_TP)
  {
    LOG64_SET(F("HOUSE_V: Characterized using Two Point Value."));
    LOG64_NEW_LINE;
  }
  else if (val_type == ESP_ADC_CAL_VAL_EFUSE_VREF)
  {
    LOG64_SET(F("HOUSE_V: Characterized using eFuse Vref."));
    LOG64_NEW_LINE;
  }
  else
  {
    LOG64_SET(F("HOUSE_V: Characterized using Default Vref."));
    LOG64_NEW_LINE;
  }

I read the raw value with : 

uint16_t value = (uint16_t) adc1_get_raw(HOUSE_VOLTMETER_PIN);

and the final result is adjusted using  : 

from_sensor = (uint16_t) esp_adc_cal_raw_to_voltage(from_sensor, &HOUSE_VOLTMETER_adc_chars);

still the rejust in some values differs in 0.2v which in 12v battery can account for 30% of battery capacity...

any suggestion very welcome...

stickbreaker, 2019-01-15:
@vtomanov read this article on [ESP32 Analog input](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/adc.html) 

Chuck.

vtomanov, 2019-01-15:
unfortunately this is exactly what I'm doing :(  and the results still differs a lot from the correct value  - ( without the characterising the analog read is useless to be honest ) 

the only thing that I do not do is to check if it collide with wifi  , but I also get the middle from 10 readings e.g. if value is very wrong the it will be discarded)

I will read again in detail the doc  to see if I have missed something...

me-no-dev, 2019-01-16:
which pin is HOUSE_VOLTMETER_PIN? do you have wifi or BT on?

RottenVanSuti, 2019-01-22:
The ADC of esp32 is not so good, because it doesn´t have a linear behavior. There's a "wrap around" using a polynomial function.   I provide the link[](https://github.com/G6EJD/ESP32-ADC-Accuracy-Improvement-function/blob/master/ESP32_ADC_Read_Voltage_Accurate.ino) credits are for @G6EJD . 
Also, take into account that only adc1 channel is usable while WiFi is on.

vtomanov, 2019-01-22:
I  do use ADC1 but ADC1 channel 0 and 1 give different reading on same voltage input ( small difference but still a difference)

closed 2019-08-15 by stale[bot]