espressif/arduino-esp32 #2335: Build fails: esp_camera.h not found

kroegerama, 2019-01-16:
The current release (1.0.1) fails to build as component with any current esp-idf version (tested stable 3.1.2 and master), because `esp-camera` is not a submodule of esp-idf.

```
libraries/ESP32/examples/Camera/CameraWebServer/app_httpd.cpp:16:24: fatal error: esp_camera.h: No such file or directory
compilation terminated.
```

atanisoft, 2019-01-16:
are you on the latest esp-idf, did you use the sdkconfig from arduino-esp32 as a basis for generating yours?

why did you not use the issue template that is present by default?

kroegerama, 2019-01-16:
I tested it with the latest esp-idf and with 3.1.2
> with any current esp-idf version (tested stable 3.1.2 and master)

I don't think, the sdkconfig is relevant here. You can see that the no esp-idf contains the esp-camera component: https://github.com/espressif/esp-idf/tree/release/v3.2/components

> why did you not use the issue template that is present by default?
Because it is too verbose for this kind of issue. My two lines of text are enough to describe the issue.

atanisoft, 2019-01-16:
Hmm, it was there previously. @me-no-dev any ideas what happened here?

> Because it is too verbose for this kind of issue. My two lines of text are enough to describe the issue.
That may be the case, but be prepared for the issue to get rejected for failure to adhere to the template.

me-no-dev, 2019-01-16:
I need to look at this :) it was reported a couple of days ago, but I am currently in Shanghai and have my schedule overloaded with meetings and such in the company HQ so I am a bit preoccupied :D 
@atanisoft is correct about using the issue template, but in this case I'll let it slide :P :D 
Reality is that IDF should not try to build the examples and their c/cpp files and that is causing the problem... will fix though!

me-no-dev, 2019-01-16:
you can just delete the camera web server cpp file and that should make your sketches compile ;) temporary fix though!

nevercast, 2019-01-18:
Stumbled in to this when using Arduino-ESP32 as a component. Will delete the Camera example and await further instruction when you return from Shanghai!

Aiouan, 2019-07-23:
Did you ever fix this?
I am having the same issue compiling from the command line in IDF.

rikki78, 2019-07-27:
@Aiouan You also will get this error if you selected the wrong board type

podericof, 2019-09-29:
Hi there, I'm having the same issue... any progress here?

me-no-dev, 2019-09-29:
this was fixed as far as I remember long time ago. how are you having the same issue?

Harriethijssen, 2019-12-10:
Same/Similar issue here: fatal error: esp_camera.h: No such file or directory.

Not too familiar with PlatformIO dev en yet, be patient...

PLATFORMIO
=========
[env:esp32cam]
platform = espressif32
board = esp32cam
framework = espidf ; espidf arduino
monitor_speed = 115200
upload_speed = 921600
build_flags = -DLOG_LOCAL_LEVEL=ESP_LOG_VERBOSE 
;ESP_LOG_NONE ESP_LOG_DEBUG ESP_LOG_VERBOSE 
board_build.partitions = partitions_singleapp.csv
build_type = debug
;lib_deps = https://github.com/espressif/esp32-camera

OUTPUT
=====
Processing esp32cam (platform: espressif32; board: esp32cam; framework: espidf; monitor_speed: 115200; upload_speed: 921600; build_flags: -DLOG_LOCAL_LEVEL=ESP_LOG_VERBOSE; board_build.partitions: partitions_singleapp.csv; build_type: debug)
--------------------------------------------------------------------------------------------------------------------------------

CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32cam.html
PLATFORM: Espressif 32 1.11.1 > AI Thinker ESP32-CAM
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-espidf 3.30300.190916 (3.3.0), tool-esptoolpy 1.20600.0 (2.6.0)      
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in debug mode
xtensa-esp32-elf-g++ -o .pio\build\esp32cam\src\main.o -c -fno-rtti -fno-exceptions -std=gnu++11 -Werror=all -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-error=unused-function -mfix-esp32-psram-cache-issue -Wall -nostdlib -Wpointer-arith -Wno-error=unused-but-set-variable -Wno-error=unused-variable -mlongcalls -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Og -g2 -ggdb2 -DWITH_POSIX -DUNITY_INCLUDE_CONFIG_H -DIDF_VER=\"3.30300.190916\" -DPLATFORMIO=40100 -DARDUINO_ESP32_DEV -DBOARD_HAS_PSRAM -DLOG_LOCAL_LEVEL=ESP_LOG_VERBOSE -DESP32 -DESP_PLATFORM -DF_CPU=240000000L -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DGCC_NOT_5_2_0=0 -DPROJECT_NAME=\"ESP32Cam-v1\" -DPROJECT_VER=\"1.0.0\" -D__PLATFORMIO_BUILD_DEBUG__ @"D:\OneDrive\ESP32\PlatformIO\Projects\ESP32Cam-v1\.pio\build\esp32cam\longcmd-10d17b7a52ce5b1fdeb77f6f9b55aefc" src\main.cpp
src\main.cpp:34:24: fatal error: esp_camera.h: No such file or directory

********************************************************************
* Looking for esp_camera.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_camera.h"
* Web  > https://platformio.org/lib/search?query=header:esp_camera.h
*
********************************************************************

compilation terminated.

closed 2020-02-22 by stale[bot]

dimitre, 2020-08-11:
@Harriethijssen  try to change your framework to arduino. ; is a comment