espressif/arduino-esp32 #2337: L2CA_CheckIsCongest BLERemoteCharacteristic

czuvich, 2019-01-16:
ESP32 WROVER Dev 4.1
Arduino Core 1.0.1

I'm running an ESP32 WROVER Dev kit as a Client connecting to 2 BLE Servers (both ESP32 chips).  The client execution flows as follows:

1) Scan
2) Connects to a server, reads the data, closes the connection
3) Runs analysis, re-opens the connection and then issues a write to the server

This execution flow works fine when it's 1 server (the write command actually puts the server to sleep); however, if I execute that flow in a loop with 2 servers, I notice the BLE stack slows down on the second write and throws an exception.  Any suggestions?  Again, thanks for the help!

```
Decoding stack results
0x40123928: L2CA_CheckIsCongest at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/l2cap/l2c_api.c line 1849
0x400fa3f5: esp_ble_gattc_write_char at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/api/esp_gattc_api.c line 420
0x400e565a: BLERemoteCharacteristic::writeValue(unsigned char*, unsigned int, bool) at /Users/Clay/Library/Arduino15/packages/esp32/hardware/esp32/1.0.1/libraries/BLE/src/BLERemoteCharacteristic.cpp line 571
0x400e56f7: BLERemoteCharacteristic::writeValue(std::__cxx11::basic_string  , std::allocator  >, bool) at /Users/Clay/Library/Arduino15/packages/esp32/hardware/esp32/1.0.1/libraries/BLE/src/BLERemoteCharacteristic.cpp line 528
0x400d4546: BluetoothDevice::executeHibernateCommand() at /var/folders/_d/ksp5fdyx5h1flr8s6l6rvgj00000gn/T/arduino_build_385074/sketch/BluetoothDevice.cpp line 292
0x400d41a9: BluetoothDevice::onExecuteCommand(int, void*) at /var/folders/_d/ksp5fdyx5h1flr8s6l6rvgj00000gn/T/arduino_build_385074/sketch/BluetoothDevice.cpp line 103
0x400e217c: WaterValveBluetoothDevice::onExecuteCommand(int, void*) at /var/folders/_d/ksp5fdyx5h1flr8s6l6rvgj00000gn/T/arduino_build_385074/sketch/WaterValveBluetoothDevice.cpp line 33
0x400d460f: BluetoothDevice::executeCommand(int, void*) at /var/folders/_d/ksp5fdyx5h1flr8s6l6rvgj00000gn/T/arduino_build_385074/sketch/BluetoothDevice.cpp line 40
0x400d6ee7: FarmAnalyzer::handleWaterValveUpdate(WaterValveBluetoothDevice*, WaterValveRegisteredDevice*) at /var/folders/_d/ksp5fdyx5h1flr8s6l6rvgj00000gn/T/arduino_build_385074/sketch/FarmAnalyzer.cpp line 264
0x400d7133: FarmAnalyzer::execute(BluetoothScanResults*) at /var/folders/_d/ksp5fdyx5h1flr8s6l6rvgj00000gn/T/arduino_build_385074/sketch/FarmAnalyzer.cpp line 50
0x400d54c1: BluetoothScanTask::run(void*) at /var/folders/_d/ksp5fdyx5h1flr8s6l6rvgj00000gn/T/arduino_build_385074/sketch/BluetoothScanTask.cpp line 25
0x400e266e: loop() at /Users/Clay/Media/work/personal/svn/gardenhub/trunk/arduino/src/gateway/gateway.ino line 161
0x400f2589: loopTask(void*) at /Users/Clay/Library/Arduino15/packages/esp32/hardware/esp32/1.0.1/cores/esp32/main.cpp line 25
0x400980a1: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

```

czuvich, 2019-01-16:
I added a 15ms delay, and it seems to help a little bit; however, I still see the issue on the second write.  During the read operation, I am reading a handful of characteristics in case that helps.  Would this be related to the client or the server?

chegewara, 2019-01-16:
It is congestion, which means you are trying to perform too many ble read/write operations in short time.
This line is from esp-idf bt stack `L2CA_CheckIsCongest`.

czuvich, 2019-01-16:
Got it.  Would you suggest I try putting a delay between each characteristic read as well?

czuvich, 2019-01-16:
Ok... So I tried adding a delay in between each read/write operation ranging from 5ms - 30ms.  I still get a LoadProhibited exception.  I saw another thread post on a similar issue (https://www.esp32.com/viewtopic.php?t=7769).  

It almost feels like a core issue and not BLE Arduino.  Anyways, I'll keep the delay in there since it does seem to help a little bit (the exception occurs less frequently).  

```
10:07:47.759 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
10:07:47.759 -> Core 1 register dump:
10:07:47.759 -> PC      : 0x40123950  PS      : 0x00060430  A0      : 0x800fa420  A1      : 0x3ffcf380  
10:07:47.759 -> A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3f81726a  A5      : 0x3f817260  
10:07:47.796 -> A6      : 0x00000000  A7      : 0xff000000  A8      : 0x80123944  A9      : 0x3ffcf360  
10:07:47.796 -> A10     : 0x3f806530  A11     : 0x3ffcf41c  A12     : 0x0000000a  A13     : 0x00000001  
10:07:47.796 -> A14     : 0x00060420  A15     : 0x00000000  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
10:07:47.796 -> EXCVADDR: 0x000000e8  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
```

chegewara, 2019-01-16:
> Got it. Would you suggest I try putting a delay between each characteristic read as well?

Try to not read/write too frequent, just dont ask me what mean not too frequent, i would suggest 3-5ms between consecutive operation as a start point.

beegee-tokyo, 2019-01-16:
@chegewara    
I run into congestion problems as well in an application that transfers data between an ESP32 and an Android app with 5ms between each data packet sent. 
The problem was on both sides, the ESP and Android. On the Android side I could solve it using callbacks when a transfer is finished, unfortunately on ESP side I couldn't find a similar solution. 
Just a suggestion, not sure if it is possible on the ESP32, but a callback when a transfer is finished (e.g. a notification has been acknowledged by the other side) could help to avoid these congestion. 

Just as a side note, for the sensor side I moved from ESP32 to a Nordic nRF52 chip to get the data transferred fast enough while I could keep the Android app basically unchanged.

czuvich, 2019-01-16:
@beegee-tokyo Just so I understand, your Android app is the client and the sensors are servers correct?  I'm not transferring large data sets (just a single packet of data - sensor data).  Could you elaborate a little more on what you did to solve your congestion issue as far as workflow between the client and server?

I've also considered looking at the nRF52 chip for sensors since I don't need WiFi, but for my proof-of-concept it was easier to use the same chip for both gateway and sensors.  But it's definitely interesting you didn't have to change your client code at all.... Were you using Arduino BLE or IDF directly for your sensor code (prior to changing chips)?

chegewara, 2019-01-16:
@czuvich I dont see reason to change android app, since ble is standard and it doesnt matter what device you use as peripheral, if your central app on android is wrote good there is no need to change it.

beegee-tokyo, 2019-01-16:
@czuvich   
a) correct, the sensors (ESP32) are the servers and Android app is the client
b) in my case a data packet is only 20 bytes. The problem was the that when sensor data was updated (notified) too fast (3 to 5ms) I  got a congestion error on the ESP side
c) I started with Arduino-ESP32 framework and then moved to ESP-IDF, but could not solve the problem.
d) As @chegewara just commented, the client side didn't need a change in the way it received the data. What I changed during testing was the way to process the data. In the beginning I tried to process each data packet on Android side immediately on application level but that didn't work well, the Android app slowed down the communication.
What solved the problem on Android side was to separate receiving and processing data completely. A service is receiving data and notifying the main app. The main app is reading data from the service, caching it and processing it whenever possible.

What helped as well was using a custom characteristic. In the beginning I tried to use BLE UART for the raw sensor data, heart rate characteristic, SpO2 characteristic and battery level characteristic. At the end to achieve best performance, I use one custom characteristic with a 20 byte data packet that holds raw data, heart rate, SpO2 level and battery level, where the raw sensor data is changing in ~5ms frequency and the other data is changing only between 500ms to 1 minute. But using only 1 characteristics and 1 notification made a big difference.

czuvich, 2019-01-16:
@beegee-tokyo Thank you for the great explanation!  I'll have to consider moving my data to a single characteristic.  On my side is I don't use notification, but I do have about 10 characteristics I'm reading.  The processing is done after I disconnect from the bluetooth sensor (sensor wakes up, client connects, grabs the data, disconnects, then client analyzes... and decides whether it needs to reconnect or not to do something else).

It really does sound like there's something going on at the core of IDF BLE stack regardless.  Anyways.. thanks to both of you for the excellent tips and advice.

closed 2019-08-15 by stale[bot]