espressif/arduino-esp32 #2338: WROVER Dev Kit 4.1 LCD and SD Not Working

czuvich, 2019-01-17:
Board: WROVER Dev Kit v4.1
Arduino 1.0.1

I am trying to get the WROVER Dev Kit LCD and SD to work together.  I removed R167, and I can now show results in the LCD even if an SD Card is inserted; however, when I try to initialize the SD card it fails to mount (see error below).  If I initialize ONLY the SD, then the SD works fine.   Is there any way to use the SD Card and LCD?  

I basically the LCD code on github (https://github.com/espressif/WROVER_KIT_LCD), and I added SPI and SD initialization after.

Error
```
19:49:03.595 -> [W][sd_diskio.cpp:471] ff_sd_initialize(): GO_IDLE_STATE failed
19:49:03.595 -> [E][sd_diskio.cpp:739] sdcard_mount(): f_mount failed 0x(3)
```

Code
```C++
SPI.begin(14, 2, 15, 13);
  if (!SD.begin(13)) {
    ESP_LOGE(LOG_TAG, "<< Failed. LocalStorage is not available.");
  }
```

me-no-dev, 2019-01-23:
On WROVER-KIT you should use SDMMC and not SD to start the SD Card :)
```cpp
if (!SDMMC.begin()) {
  log_e("<< Failed. LocalStorage is not available."); //do not use ESP_LOGx in Arduino code
}
```

czuvich, 2019-01-23:
Perfect!  Thanks @me-no-dev!  I wasn't aware to not use ESP_LOGX functions in Arduino code.

closed 2019-08-15 by stale[bot]