espressif/arduino-esp32 #2345: ESP32 strange WiFi behavior at standard example codes:

nae9, 2019-01-18:
Release example codes for arduino esp32 have some problem...
If i compile for example SimpleWiFiServer.ino and try to connect by web-browser then serial terminal shows that after right connection led wrong client with free rx buffer and it led to close this client by timeout (freezes main loop). (client.stop() not work properly)

It seemed that wifi go in to sleep mode after beginning and disable sleep mode after Wifi.begin() fix this problem. And all examples work fine with hi speed http response, as at very old libraries (before rx buffer introduce into wifi lib).

void setup()
{
...
WiFi.begin(ssid, password);
WiFi.setSleep(false);// this code solves my problem
...
}

But may be some body pull this parameter as default into release libs? Or change ALL example codes in such manner?

closed 2019-08-15 by stale[bot]

mcuadros, 2019-08-30:
@nae9 is your router a Linksys EA?

nae9, 2019-08-30:
> 
> 
> @nae9 is your router a Linksys EA?

EA4500 , openwrt git-18.228.31946-f64b152

mcuadros, 2019-08-30:
https://github.com/me-no-dev/ESPAsyncWebServer/issues/437

I have a EA6300 with stock firmware, and I am having the same issues, the problem is that for me setSleep() doesn't solve my problems for output connections. 

The only way to fix the problem in my case is doing a ping to the IP until wakeups

nae9, 2019-08-30:
Not linksys EA router can help issue?

mcuadros, 2019-08-30:
Here, I only have this router here, but I will try with another routers. Based on the issue referred and the fact that you have a very similar issue and the same brand and series of mine, I can imagine that the problem is only related with the routers.

mcuadros, 2019-08-31:
I can confirm that in my case, with outer routers works flawless, using same sever and board