espressif/arduino-esp32 #2346: USBHost to USBDevice bridge using ESP-NOW

ozarchie, 2019-01-18:
I am investigating a transparent USB bridge using two ESP32s communicating using ESPNOW.

**For my existing serial bridge:**

I am using ESPNOW as a transparent serial comms link using two ESP32s (ESP8266 also work).
Each device simply receives serial data, places it into an ESPNOW packet, and transmits via WiFi.
At the same time, it receives any ESPNOW packet destined for it, puts it into a serial buffer and transmits it to the serial data pin.
The Host serial plugs into the USB port and is recognised as a COMn device
The Device serial plugs into the ESP32 TTL pins via logic/level convertors.

**For the proposed USB bridge:**

1. The USBDevice is connected to DeviceESP32 via GPIO04/GPIO05.
2. The USBHost is connected to HostESP32 via GPIO04/GPIO05.
3. USBDevice serial data is received by DeviceESP32 (e.g. using cnlohr dma method) into a buffer.
4. Data is then transmitted using ESPNOW packets to the HostESP32.
5. ESPNOW packets are received into a HostESP32 buffer.
6. HostESP32 then sends the data to the HostUSB port (e.g.  cnlohr code).
7. When the Host computer responds to the USBDevice (via the HostESP32 pins), HostESP32 receives the data into a buffer.
8. Data is then sent via ESPNOW packets to the DeviceESP32.
9. DeviceESP32 receives the data into a buffer.
10. Data is then sent to the USBDevice pins.

As the communication is half duplex, no collisions should occur.

My question is:
Do you thing the ESP32/ESPNOW combination is fast enough to handle this?

### Hardware:
Board:							ESP32 Doit
Core Installation/update date:			Current
IDE name:							Platform.io
Flash Frequency:					
PSRAM enabled:                                          
Upload Speed:						
Computer OS:                                            Windows 10

copercini, 2019-01-18:
cnlohr USB code only works on ESP8266

ozarchie, 2019-01-19:
OK, but I was proposing a new way to do it by monitoring and echoing the bit levels.
cnlohr reads and interprets data, thus limiting the speed.
My suggestion is to echo the data pins and let the HostDevice do the reading and interpreting.

The esp32 is a serializer/de-serializer over WiFi.
The start/end conditions (D+/D- states) are monitored, but the data is then just echoed from the pins to WiFi, using UART/SPI/whatever and DMA.
I am just not sure whether ESPNOW is fast enough.

closed 2019-01-28 by ozarchie