espressif/arduino-esp32 #2351: uint8_t math results in incorrect mod(%)

stickbreaker, 2019-01-19:
### Hardware:
Board:							WeMos BlueTooth and Battery
Core Installation version:			1.0.1-rc4 git
IDE name:						Arduino IDE 1.8.8
Flash Frequency:					80Mhz
PSRAM enabled:                                     none
Upload Speed:					921600
Computer OS:                                        Windows 7 pro 32

### Description:
a multiple variable `uint8_t` equation evaluates incorrectly.
```c++
   uint8_t _topLine  = 0,
        _tailLine = 0,
        _headLine = 3,
        _maxLine  = 4,
        _rows     = 4,
        row       = 0;

  _topLine = (((_tailLine + row) -_rows) + (uint8_t)1)% _maxLine;
```
evaluates `_topLine` as 253, but it cannot be?  the MOD function must limit it to 0..(_maxLine-1) or 0..3

But, if I expand this equation to single operation statements it works.
```c++
   uint8_t _topLine  = 0,
        _tailLine = 0,
        _headLine = 3,
        _maxLine  = 4,
        _rows     = 4,
        row       = 0;

    _topLine = _tailLine;
    _topLine = _topLine + row;
    _topLine = _topLine - _rows;
    _topLine = _topLine + 1;
    _topLine = _topLine % _maxLine;
```
### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
void setup(){
    Serial.begin(115200);
    uint8_t _topLine  = 0,
        _tailLine = 0,
        _headLine = 3,
        _maxLine  = 4,
        _rows     = 4,
        row       = 0;
 
    Serial.printf(" before                 tail=%d, top=%d, head=%d, max=%d, _rows=%d, row=%d\n",_tailLine,_topLine, _headLine,_maxLine,_rows, row);

    _topLine = (((_tailLine + row) -_rows) + (uint8_t)1)% _maxLine;

    Serial.printf(" after single statement tail=%d, top=%d, head=%d, max=%d, _rows=%d, row=%d\n",_tailLine,_topLine, _headLine,_maxLine,_rows, row);

    _topLine = _tailLine;
    Serial.printf("\n top=tail =%d, tail=%d\n",_topLine,_tailLine);
    _topLine = _topLine + row;
    Serial.printf(" top+row=%d, row=%d\n",_topLine,row);
    _topLine = _topLine - _rows;
    Serial.printf(" top-_rows=%d, _rows=%d\n",_topLine,_rows);
    _topLine = _topLine + 1;
    Serial.printf(" top+1=%d\n",_topLine);
    _topLine = _topLine % _maxLine;
    Serial.printf(" top %% maxLine=%d, maxline=%d\n",_topLine,_maxLine);

    Serial.printf("\n after multiple statement tail=%d, top=%d, head=%d, max=%d, _rows=%d, row=%d\n",_tailLine,_topLine, _headLine,_maxLine,_rows, row);
 }
 
 void loop(){}
```

### Debug Messages:
```
before                 tail=0, top=0, head=3, max=4, _rows=4, row=0
 after single statement tail=0, top=253, head=3, max=4, _rows=4, row=0

 top=tail =0, tail=0
 top+row=0, row=0
 top-_rows=252, _rows=4
 top+1=253
 top % maxLine=1, maxline=4

 after multiple statement tail=0, top=1, head=3, max=4, _rows=4, row=0
```

Chuck.

stickbreaker, 2019-01-19:
Changing the data type to `uint32_t` results in correct calculations.  So, does this mean that the ESP32 cannot be relied on when using 8bit data type?

Chuck.

me21, 2019-01-19:
In C/C++, all arithmetic operations are working with data types not less than int. If the data type is less than int, it is promoted to int before calculation.
So in your one-liner all calculations are using signed ints, whereas in line-by-line every intermediate result is converted to uint8_t again.
In uint32_t variant, no such promotion to (signed) int takes place, because uint32_t not smaller than int.

https://stackoverflow.com/q/46073295/2404492
https://en.cppreference.com/w/cpp/language/implicit_conversion
See "Integral promotion" paragraph.

To sum up, it can be confusing, but it is conforming to the standard.

stickbreaker, 2019-01-19:
@me21 Integer promotion does not explain why the Modulus operation was not preformed.  253 is the twos complement representation of -3.  Modulus 4 of -3 is 1, not -3 or 253. Integer promotion issues result in intermediate values that are not consistent with the final data types.  

I will produce a test case for all integer types and post the results.
Chuck.

me21, 2019-01-19:
In i64:

`
_topLine = (((_tailLine + row) -_rows) + (uint64_t)1)% _maxLine;
`

You are adding `(uint64_t)1`. It causes conversion of the entire expression to uint64_t type, see my link on arithmetic operators, it is explained there.

Looking at your other functions, I think it's just copy&paste error and you meant `(int64_t)1`. Right?

stickbreaker, 2019-01-19:
@me21 yep, missed that one.

closed 2019-01-19 by stickbreaker