espressif/arduino-esp32 #2352: Error compiling for board Adafruit ESP32 Feather.

hackerspace09, 2019-01-19:
I am trying to build an IR camera and am having trouble compiling for board Adafruit ESP32 Feather.

Arduino: 1.8.8 (Windows 7), Board: "Adafruit ESP32 Feather, 80MHz, 921600, Debug", esp32 1.0.0
Error compiling for board Adafruit ESP32 Feather.

Hello, I'm completely new to this but trying to make progress.  I installed SiLabs CP2104 Driver.  Then loaded ESP32  1.0.1 in board manager.  I was unable to find Adafruit ESP32 Feather.  I found post "Compile error on latest commit #866" with the same problem.  I uninstalled ESP32  1.0.1 then loaded ESP32  1.0.0 in board manager.  I restarted IDE and I was able to find Adafruit ESP32 Feather.  :)  However, I'm still having trouble compiling for board Adafruit ESP32 Feather.  Any help is greatly appreciated.

Arduino: 1.8.8 (Windows 7), Board: "Adafruit ESP32 Feather, 80MHz, 921600, Debug"

C:\Program Files\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\Arduino\hardware -hardware C:\Users\M1\AppData\Local\Arduino15\packages -tools C:\Program Files\Arduino\tools-builder -tools C:\Program Files\Arduino\hardware\tools\avr -tools C:\Users\M1\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files\Arduino\libraries -libraries C:\Users\M1\Documents\Arduino\libraries -fqbn=esp32:esp32:featheresp32:FlashFreq=80,UploadSpeed=921600,DebugLevel=debug -ide-version=10808 -build-path C:\Users\M1\AppData\Local\Temp\arduino_build_21702 -warnings=none -build-cache C:\Users\M1\AppData\Local\Temp\arduino_cache_916091 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\M1\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\M1\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esptool.path=C:\Users\M1\AppData\Local\Arduino15\packages\esp32\tools\esptool\2.3.1 -prefs=runtime.tools.esptool-2.3.1.path=C:\Users\M1\AppData\Local\Arduino15\packages\esp32\tools\esptool\2.3.1 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\M1\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\M1\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -verbose C:\Users\M1\Documents\Arduino\sketch_jan19a\sketch_jan19a.ino
C:\Program Files\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files\Arduino\hardware -hardware C:\Users\M1\AppData\Local\Arduino15\packages -tools C:\Program Files\Arduino\tools-builder -tools C:\Program Files\Arduino\hardware\tools\avr -tools C:\Users\M1\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files\Arduino\libraries -libraries C:\Users\M1\Documents\Arduino\libraries -fqbn=esp32:esp32:featheresp32:FlashFreq=80,UploadSpeed=921600,DebugLevel=debug -ide-version=10808 -build-path C:\Users\M1\AppData\Local\Temp\arduino_build_21702 -warnings=none -build-cache C:\Users\M1\AppData\Local\Temp\arduino_cache_916091 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\M1\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\M1\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esptool.path=C:\Users\M1\AppData\Local\Arduino15\packages\esp32\tools\esptool\2.3.1 -prefs=runtime.tools.esptool-2.3.1.path=C:\Users\M1\AppData\Local\Arduino15\packages\esp32\tools\esptool\2.3.1 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\M1\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\M1\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -verbose C:\Users\M1\Documents\Arduino\sketch_jan19a\sketch_jan19a.ino
Using board 'featheresp32' from platform in folder: C:\Users\M1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0
Using core 'esp32' from platform in folder: C:\Users\M1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0
Build options changed, rebuilding all
Detecting libraries used...
"C:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/config" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/bluedroid" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/bluedroid/api" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/app_trace" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/app_update" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/bootloader_support" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/bt" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/driver" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/esp32" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/esp_adc_cal" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/esp_http_client" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/esp-tls" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/ethernet" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/fatfs" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/freertos" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/heap" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/jsmn" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/log" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/mdns" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/mbedtls" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/mbedtls_port" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/newlib" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/nvs_flash" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/openssl" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/spi_flash" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/sdmmc" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/smartconfig_ack" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/spiffs" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/tcpip_adapter" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/ulp" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/vfs" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/wear_levelling" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/xtensa-debug-module" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/coap" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/console" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include [... cut ...] \1.0.0/tools/sdk/include/config" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/bluedroid" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/bluedroid/api" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/app_trace" "-IC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/include/app_update" "-IC:\

bertmelis, 2019-01-19:
So after 5 minutes scrolling down, I still didn't see the failing sketch.

You could also just paste in the error message itself: the linker doesn't find the void setup() and void loop() functions.

hackerspace09, 2019-01-19:
thank you I will try that next time, I didn't want to leave anything out.
Total noob error, didn't add a sketch.  Got it now but a new error.  

Unable to cache built core, please tell esp32 maintainers to follow http://goo.gl/QdCUjo
Linking everything together...
"C:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-gcc" -nostdlib "-LC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/lib" "-LC:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/sdk/ld" -T esp32_out.ld -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.spiram_incompatible_fns.ld -u ld_include_panic_highint_hdl -u call_user_start_cpu0 -Wl,--gc-sections -Wl,-static -Wl,--undefined=uxTopUsedPriority -u __cxa_guard_dummy -u __cxx_fatal_exception -Wl,--start-group "C:\\Users\\M1\\AppData\\Local\\Temp\\arduino_build_167130\\sketch\\interpolation.cpp.o" "C:\\Users\\M1\\AppData\\Local\\Temp\\arduino_build_167130\\sketch\\thermal_cam_interpolate.ino.cpp.o" "C:\\Users\\M1\\AppData\\Local\\Temp\\arduino_build_167130\\libraries\\Adafruit_GFX_Library\\Adafruit_GFX.cpp.o" "C:\\Users\\M1\\AppData\\Local\\Temp\\arduino_build_167130\\libraries\\Adafruit_GFX_Library\\Adafruit_SPITFT.cpp.o" "C:\\Users\\M1\\AppData\\Local\\Temp\\arduino_build_167130\\libraries\\Adafruit_GFX_Library\\glcdfont.c.o" "C:\\Users\\M1\\AppData\\Local\\Temp\\arduino_build_167130\\libraries\\Adafruit_ILI9341\\Adafruit_ILI9341.cpp.o" "C:\\Users\\M1\\AppData\\Local\\Temp\\arduino_build_167130\\libraries\\SPI\\SPI.cpp.o" "C:\\Users\\M1\\AppData\\Local\\Temp\\arduino_build_167130\\libraries\\Wire\\Wire.cpp.o" "C:\\Users\\M1\\AppData\\Local\\Temp\\arduino_build_167130\\libraries\\Adafruit_AMG88xx_Library\\Adafruit_AMG88xx.cpp.o" "C:\\Users\\M1\\AppData\\Local\\Temp\\arduino_build_167130/arduino.ar" -lgcc -lopenssl -lbtdm_app -lfatfs -lwps -lcoexist -lwear_levelling -lesp_http_client -lhal -lnewlib -ldriver -lbootloader_support -lpp -lmesh -lsmartconfig -ljsmn -lwpa -lethernet -lphy -lapp_trace -lconsole -lulp -lwpa_supplicant -lfreertos -lbt -lmicro-ecc -lcxx -lxtensa-debug-module -lmdns -lvfs -lsoc -lcore -lsdmmc -lcoap -ltcpip_adapter -lc_nano -lesp-tls -lrtc -lspi_flash -lwpa2 -lesp32 -lapp_update -lnghttp -lspiffs -lespnow -lnvs_flash -lesp_adc_cal -llog -lsmartconfig_ack -lexpat -lm -lc -lheap -lmbedtls -llwip -lnet80211 -lpthread -ljson -lstdc++ -Wl,--end-group -Wl,-EL -o "C:\\Users\\M1\\AppData\\Local\\Temp\\arduino_build_167130/thermal_cam_interpolate.ino.elf"
"C:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/gen_esp32part.exe" -q "C:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.0/tools/partitions/default.csv" "C:\\Users\\M1\\AppData\\Local\\Temp\\arduino_build_167130/thermal_cam_interpolate.ino.partitions.bin"
"C:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esptool\\2.3.1/esptool.exe" --chip esp32 elf2image --flash_mode dio --flash_freq 40m --flash_size 4MB -o "C:\\Users\\M1\\AppData\\Local\\Temp\\arduino_build_167130/thermal_cam_interpolate.ino.bin" "C:\\Users\\M1\\AppData\\Local\\Temp\\arduino_build_167130/thermal_cam_interpolate.ino.elf"
esptool.py v2.3.1

Using library Adafruit_GFX_Library at version 1.3.6 in folder: C:\Users\M1\Documents\Arduino\libraries\Adafruit_GFX_Library 
Using library Adafruit_ILI9341 at version 1.1.0 in folder: C:\Users\M1\Documents\Arduino\libraries\Adafruit_ILI9341 
Using library SPI at version 1.0 in folder: C:\Users\M1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\SPI 
Using library Wire at version 1.0 in folder: C:\Users\M1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\Wire 
Using library Adafruit_AMG88xx_Library at version 1.0.4 in folder: C:\Users\M1\Documents\Arduino\libraries\Adafruit_AMG88xx_Library 
"C:\\Users\\M1\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-size" -A "C:\\Users\\M1\\AppData\\Local\\Temp\\arduino_build_167130/thermal_cam_interpolate.ino.elf"
Sketch uses 206996 bytes (15%) of program storage space. Maximum is 1310720 bytes.
Global variables use 15236 bytes (4%) of dynamic memory, leaving 312444 bytes for local variables. Maximum is 327680 bytes.
esptool.py v2.3.1
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header

bertmelis, 2019-01-19:
There are plenty of issues with failed connections. Install the latest driver for the USB chip of your board.

What could help: attach the board before starting Windows.

1TekPro, 2019-01-28:
[http://www.meeedstudio.com/article-66.html](url)

This should work with all 3 revisions of the board.

closed 2019-08-15 by stale[bot]