espressif/arduino-esp32 #2361: Connect HC-05 to esp32 Classic BT

mimansamaheshwari, 2019-01-21:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:	        11/jul/2017
IDE name:					        Arduino IDE
Flash Frequency:					80MHz
PSRAM enabled:                                         no
Upload Speed:					115200
Computer OS:                                          Windows 10

### Description:
I want to perform code upload through OTA on ESP32 with help of HC-05 using Arduino IDE. Is this possible? Or can I just connect ESP32 with HC-05 and perform basic data transfer? Currently I tried pairing HC-05 with esp32 Classic BT with following set of commands on HC-05:
AT+ROLE=1
AT+INIT
AT+INQ  (//IN THIS COMMAND I WAS ABLE TO SEE ADDRESS OF MY ESP32 MODULE)
AT+BIND= esp32device address ( // FOR THIS ALSO I GOT AN OK)
But still I am unable to pair both of them. Is it because of password parameters present in HC-05? How can I set this parameters for esp32?

1TekPro, 2019-01-28:
Without looking at any of your code.... Are both devices spp modes configured properly? Is one in discovery mode? Is one scanning for devices? Was a connection requested? These are just a few questions.

> How can I set this parameters for esp32?
AT+BTSPPCONN=<conn_index>,<sec_mode>,<remote_address>

NANDEESH-KM, 2019-03-06:
how to pair esp32 with HC-05 ?...I have configured HC-05 to slave mode using AT commands. please reply with the sollutions

closed 2019-08-15 by stale[bot]